MichaIng/DietPi #5166: unknown filesystem type 'exfat'

RunningUtes, 2022-01-14:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`

bullseye 0

- Kernel version | `uname -a`

Linux DietPi 5.10.52-v8+ #1441 SMP PREEMPT Tue Aug 3 18:14:03 BST 2021 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

RPi 4 Model B (aarch64)

- Power supply used | (EG: 5V 1A RAVpower)
5V
- SDcard used | (EG: SanDisk ultra)
SanDisk ultra

#### Additional Information (if applicable)

unable to mount exfat despite install exfat-utils
```
root@DietPi:/tmp/DietPi-Drive_Manager# apt-get install exfat-fuse exfat-utils
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
exfat-utils is already the newest version (1.3.0-2).
exfat-fuse is already the newest version (1.3.0-2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@DietPi:/tmp/DietPi-Drive_Manager# mount -t exfat /dev/sda1 /mnt/61A8-A2D6
mount: /mnt/61A8-A2D6: unknown filesystem type 'exfat'.
root@DietPi:/tmp/DietPi-Drive_Manager#  
```

- Bug report ID | `echo $G_HW_UUID`

15697a17-13e4-4865-b927-e7d2a812820d

#### Steps to reproduce

1. plug in 512 Gb USB
2. run dietpi-drive_manager
3. mount USB
4. error. occurs

#### Expected behaviour

- should mount the USB

#### Actual behaviour

- error that exfat filesystem is unknown

#### Extra details

- ...

MichaIng, 2022-01-14:
Many thanks for your report.

Please reboot and try again. The kernel got upgraded, so kernel modules for the currently running one cannot be loaded. Another case where #4732 makes sense.

BetaTester41, 2022-01-15:
I too have this problem, even after reboot and upgrades. DietPi 7.X works fine for me though.

MichaIng, 2022-01-16:
After a reboot (when not done yet), can you show the output of the following command:
```sh
modinfo exfat
```

BetaTester41, 2022-01-19:
When typing the command, I'm left with:
`modinfo: ERROR: Module exfat not found.`
While tinkering with commands, I found out that you can mount it using the `mount.exfat-fuse` command. Also changing the corresponding entry type in the fstab to `exfat-fuse` mounts it properly on startup. I don't know, but it works!

andbroe, 2022-03-21:
Have the same problem:

login as: root
root@192.168.31.252's password:
Linux DietPiHC2 4.14.241+ #1 SMP PREEMPT Wed Jul 28 16:55:16 UTC 2021 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
 ─────────────────────────────────────────────────────
 DietPi v8.2.2 : 13:05 - Sun 03/20/22
 ─────────────────────────────────────────────────────
 - Device model : Odroid XU3/XU4/MC1/HC1/HC2 (armv7l)
 - CPU temp : 30 °C / 86 °F : Cool runnings
 - LAN IP : 192.168.31.252 (eth0)
 - MOTD : Upgrade your kernel now to fix the "Dirty Pipe" vulnerability:
          https://dietpi.com/blog/?p=1379
 ─────────────────────────────────────────────────────

 DietPi Team     : https://github.com/MichaIng/DietPi#the-dietpi-project-team
 Image by        : DietPi Core Team (pre-image: Meveric)
 Patreon Legends : Camry2731, Chris Gelatt
 Website         : https://dietpi.com/ | https://twitter.com/DietPi_
 Contribute      : https://dietpi.com/contribute.html
 Web Hosting by  : https://myvirtualserver.com

 dietpi-launcher : All the DietPi programs in one place
 dietpi-config   : Feature rich configuration tool for your device
 dietpi-software : Select optimised software for installation
 htop            : Resource monitor
 cpu             : Shows CPU information and stats

root@DietPiHC2:~# apt update
Reading package lists... Done
E: Could not get lock /var/lib/apt/lists/lock. It is held by process 733 (apt-get)
N: Be aware that removing the lock file is not a solution and may break your system.
E: Unable to lock directory /var/lib/apt/lists/
root@DietPiHC2:~# apt update
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://dietpi.com/meveric all InRelease
Hit:4 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:5 https://dietpi.com/meveric bullseye InRelease
Hit:6 https://deb.debian.org/debian bullseye-backports InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
root@DietPiHC2:~# apt-get install exfat-fuse
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
exfat-fuse is already the newest version (1.3.0-2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@DietPiHC2:~# apt-get install exfat-utils
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages will be REMOVED:
  exfatprogs
The following NEW packages will be installed:
  exfat-utils
0 upgraded, 1 newly installed, 1 to remove and 0 not upgraded.
Need to get 42.8 kB of archives.
After this operation, 49.2 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 https://deb.debian.org/debian bullseye/main armhf exfat-utils armhf 1.3.0-2 [42.8 kB]
[... 123 lines cut ...]
 - MOTD : Upgrade your kernel now to fix the "Dirty Pipe" vulnerability:
          https://dietpi.com/blog/?p=1379
 ─────────────────────────────────────────────────────

 DietPi Team     : https://github.com/MichaIng/DietPi#the-dietpi-project-team
 Image by        : DietPi Core Team (pre-image: Meveric)
 Patreon Legends : Camry2731, Chris Gelatt
 Website         : https://dietpi.com/ | https://twitter.com/DietPi_
 Contribute      : https://dietpi.com/contribute.html
 Web Hosting by  : https://myvirtualserver.com

 dietpi-launcher : All the DietPi programs in one place
 dietpi-config   : Feature rich configuration tool for your device
 dietpi-software : Select optimised software for installation
 htop            : Resource monitor
 cpu             : Shows CPU information and stats

root@DietPiHC2:~# modinfo exfat
modinfo: ERROR: Module exfat not found.
root@DietPiHC2:~#

Hope this helps

Chers
andbroe

MichaIng, 2022-03-21:
Did you try to simply mount the drive with ` mount /dev7sda2 /mnt/M2SSD/` or via `dietpi-drive_manager`? And did you manually install `exfatprogs`?

General info:
- The `exfat` kernel module has been added with Linux v5.4, hence it not available on your Odroid HC2 with Linux v4.14. That `modprobe exfat` fails is hence expected.
- Since `modprobe exfat` does fail, `dietpi-drive_manager` should not install `exfatprogs` but instead `exfat-fuse` and `exfat-utils`, the FUSE based software driver for exFAT and related userland utilities. AFAIK , `exfatprogs` only works with the native Linux driver, not with the FUSE based one.
- Since with the FUSE based driver, the filesystem type for exFAT is shown as `fuse`, not as `exfat`, `mount -t exfat` may fail while it works just skipping the filesystem type and leaving it up to the system to auto-detect and use the correct mount method.

andbroe, 2022-03-22:
> With exfat-fuse and exfat-utils installed now, please retry:
>
> mount /dev/sda2 /mnt/M2SSD
>
> I just see you had a typo in there, 7 instead of slash.

Hi,

still the same

Regards
andbroe
=============================
***@***.***:~# apt update
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 https://deb.debian.org/debian bullseye-backports InRelease
Hit:5 https://dietpi.com/meveric all InRelease
Hit:6 https://dietpi.com/meveric bullseye InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
***@***.***:~# apt install exfat-fuse
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
exfat-fuse is already the newest version (1.3.0-2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
***@***.***:~# apt install exfat-utils
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
exfat-utils is already the newest version (1.3.0-2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
***@***.***:~# mount -t "exfat" /dev/sda2 /mnt/M2SSD/
mount: /mnt/M2SSD: unknown filesystem type 'exfat'.
***@***.***:~# mount -t exfat /dev/sda2 /mnt/M2SSD/
mount: /mnt/M2SSD: unknown filesystem type 'exfat'.
***@***.***:~#

MichaIng, 2022-03-22:
Please try it as I suggested and explained above without defining the filesystem type `exfat`:
```sh
mount /dev/sda2 /mnt/M2SSD
```

andbroe, 2022-03-23:
Sorry, same problem

***@***.***:~# mount /dev/sdb2 /mnt/M2SSD
mount: /mnt/M2SSD: unknown filesystem type 'exfat'.

andbroe, 2022-03-23:
> Is the fuse module loaded?
>
> modprobe fuse
> lsmod | grep fuse
> cat /proc/filesystems
>
> And please try to mount it explicitly as with as exfat-fuse
>
> mount.exfat-fuse /dev/sda2 /mnt/M2SSD

Yes, that worked

=======
***@***.***:~# modprobe fuse
***@***.***:~# lsmod | grep fuse
fuse                   86016  1
***@***.***:~# cat /proc/filesystems
nodev   sysfs
nodev   rootfs
nodev   ramfs
nodev   bdev
nodev   proc
nodev   cpuset
nodev   cgroup
nodev   cgroup2
nodev   tmpfs
nodev   devtmpfs
nodev   configfs
nodev   debugfs
nodev   tracefs
nodev   securityfs
nodev   sockfs
nodev   dax
nodev   bpf
nodev   pipefs
nodev   rpc_pipefs
nodev   devpts
        ext3
        ext4
        ext2
        cramfs
        squashfs
        vfat
        msdos
nodev   nfs
nodev   nfs4
        romfs
nodev   autofs
nodev   mqueue
nodev   gatorfs
        fuseblk
nodev   fuse
nodev   fusectl
***@***.***:~# mount.exfat-fuse /dev/sda2 /mnt/M2SSD
FUSE exfat 1.3.0

========

Now I can access the files.

Regards
andbroe

MichaIng, 2022-03-23:
I wonder why `mount.exfat` does not exist, and indeed also `exfat` is not shown in `/proc/filesystems`. Not sure whether the intermediate `exfatprogs` install messed something up, i.e. prevented the system to add the `exfat` filesystem driver via FUSE automatically. I'll try to replicate on another Bullseye system with older kernel (with no native exfat module).

andbroe, 2022-03-23:
Ok, thanks.

- Sent from my mobile device -

MichaIng, 2022-04-02:
Okay, so generally the exFAT module works fine when present, using `exfatprogs`.

Then I removed the module and rebooted. As expected, mounting the drive then failed. For the `modprobe -n exfat` dry-run to fail, I needed to run `depmod -a`, otherwise the symbol for the `exfat` module was still present. However, if the kernel really does not ship the module, this is not required.

Then I reformatted the drive, which installs `exfat-utils` and purges `exfatprogs`, as expected, but indeed mounting the drive still fails, the system still looks for the non-existing `exfat` filesystem type, not made available by `exfat-fuse` and FUSE not being invoked automatically.

Then I restored the VM and formatted the exFAT drive after removing the kernel module, so that `exfatprogs` isn't installed in the first place, and it fails the same way. The problem really is that on Bullseye, `exfat-fuse` does not ship the `mount.exfat` command, which it does no Buster. But also creating this command as symlink does not solve it. `blkid -s TYPE -o value -c /dev/null /dev/sdX1` still detects the drive as `exfat` while it seems to be required since Bullseye to detect it as `exfat-fuse` for the correct filesystem driver to be invoked. On Buster, `exfat` is correct.

Here is the related changelog of the Debian package:
```
fuse-exfat (1.3.0-2) unstable; urgency=medium

  * Drop "mount.exfat" symlink to the mount helper in the
    wake of the exfat support inside Linux. Adding a debian/NEWS
```
This makes of course sense when the kernel ships the module, which is true for the Debian kernel, but one cannot assume that all Debian systems use the Debian kernel, especially for ARM SBCs where manufacturers ship Debian images with own older kernel versions, like Hardkernel with Odroid XU4. For this to work properly, `exfat-fuse` either needs to keep providing the `exfat` filesystem (so that it is listed in `/proc/filesystems`, or `blkid`, `lsblk` and similar need to detect the filesystem type as `fuse-exfat` or so, so that `mount` invokes the correct filesystem drive.

I'll report this to Debian. Currently I have no good idea for a workaround in `dietpi-drive_manager`, especially since `mount` would still fail, unless this filesystem driver is explicitly forced:
```sh
mount -t exfat-fuse /dev/sdX1 /mnt/<mountpoint>
``` 
and the `/etc/fstab` entry in case manually adjusted.

MichaIng, 2022-04-02:
A workaround is, when the mount command fails, to add `-o exfat-fuse` explicitly as mount option, and changing the related `/etc/fstab` entry to contain `exfat-fuse` instead of `exfat`.

I just reported the issue to Debian, let's see whether there is an aim and idea to restore full compatibility with the FUSE exFAT driver: https://bugs.debian.org/1008842

MichaIng, 2022-04-02:
Okay, it won't get fixed by Debian, but I found a simple long-term workaround:
```sh
ln -s mount.exfat-fuse /sbin/mount.exfat
```
This doesn't even break anything when the kernel module and `exfatprogs` get installed later and/or `exfat-fuse` purged. That file is not shipped by anything else, and as fast as the symlink becomes invalid, it is gracefully ignored.

MichaIng, 2022-04-02:
Workaround applied with: https://github.com/MichaIng/DietPi/commit/77b1d57

closed 2022-04-02 by MichaIng