MichaIng/DietPi #5169: Downloading docker fails

klmkyo, 2022-01-14:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `bullseye 0`
- Kernel version | `Linux DietPi 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux`
- SBC model | RPI 3b+
- Power supply used | 5v 2a OnePlus charging brick, good enough to not cause undervoltage
- SDcard used | SanDisk ultra

#### Additional Information (if applicable)
- Software title | Docker
- Was the software title installed freshly or updated/migrated? | Freshly
- Can this issue be replicated on a fresh installation of DietPi? | Yes

- Bug report ID | 00d1b2e4-94f9-4e3f-925c-48f81b7ecd84

#### Steps to reproduce
Attempt to install docker

#### Expected behaviour
Docker is downloaded and installed

#### Actual behaviour
It is not.

#### Details:
- Date           | Fri Jan 14 16:57:58 CET 2022
- Bug report     | 00d1b2e4-94f9-4e3f-925c-48f81b7ecd84
- DietPi version | v8.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 3 Model B+ (aarch64) (ID=3)
- Kernel version | `Linux DietPi 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -q update`
- Exit code      | 100
- Software title | DietPi-Software
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Ign:3 https://download.docker.com/linux/debian 11 InRelease
Hit:4 https://archive.raspberrypi.org/debian bullseye InRelease
Hit:5 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:6 https://deb.debian.org/debian bullseye-backports InRelease
Err:7 https://download.docker.com/linux/debian 11 Release
  404  Not Found [IP: 54.192.230.81 443]
Hit:8 https://repo.jellyfin.org/debian bullseye InRelease
Reading package lists...
E: The repository 'https://download.docker.com/linux/debian 11 Release' does not have a Release file.
```

MichaIng, 2022-01-14:
Many thanks for your report.

When and how did you try to install the Docker repository? It is simply an invalid entry, which AFAIK was falsely installed that way by the Docker installer a while back. Try to use  `dietpi-software`:
```sh
rm /etc/apt/sources.list.d/*docker*.list
dietpi-software install 162
```

klmkyo, 2022-01-14:
Thanks for the quick response, your solution did work :) Thanks for the help!

closed 2022-01-14 by klmkyo