MichaIng/DietPi #5171: Pydio - inputfilter.php - magic_quotes_gpc() is deprecated error.

satnair, 2022-01-14:
### Creating a bug report/issue

DietPi 8 latest version
Pydio running on raspberry 400
inputfilter.php - function get_magic_quotes_gpc() is deprecated. 

Doesn't load any workspace , user access etc. Please help.

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2022-01-14:
Many thanks for your report.

Can you please full the bug report template, most importantly Debian version, hardware and OS architecture, so that we are able to try replicating it? Also which webserver do you use?

Usually a deprecation warning shouldn't break anything, is this the only error you see in Pydio, webserver error logs and browser console?

satnair, 2022-01-15:
[image]

I use dietpi v 8.2.3
raspberry pi 400 version
64 bit arch

I am seeing this as error in the logs output. Because of this when I try to access user details (for example) nothing shows up.

satnair, 2022-01-15:
Server logs here 
[image]

Joulinar, 2022-01-15:
Just as a side note, there is not DietPi version 8.2.3 existing. Lates version is 8.0.2

MichaIng, 2022-01-16:
> I am seeing this as error in the logs output. Because of this when I try to access user details (for example) nothing shows up.

You mean the Pydio logs, right? Why do you think that the deprecation warning is the reason for the other pages not showing up? As said, it is a warning/notice for developers that they should switch to a modern method/function by times, hence something to report to Pydio devs, but shouldn't cause any actual failure. Can you please show the browser console logs when trying to access a failing Pydio page?

I'll also try to replicate now.

MichaIng, 2022-01-16:
Okay before investing too much time into Pydio Core, we should instead migrate to Pydio Cells, the standalone (written in Go) replacement: #3469

I can see the warning, however it does not effect functionality. Can you be a bit more precise which page/feature does not work?

What I did find is that the internal updater does not work as of some false file owner and modes. To fix this:
```sh
chown -R www-data:www-data /var/www/pydio
chmod -R u+w /var/www/pydio
```
Then you can update from v8.2.3 to v8.2.5.

satnair, 2022-01-17:
Great thanks. Also, looks like a PHP compatibility issue. I will check this. Thanks.

closed 2022-01-17 by satnair

MichaIng, 2022-01-17:
I fixed the permissions:
- https://github.com/MichaIng/DietPi/commit/355a5c3
- https://github.com/MichaIng/DietPi/commit/a98adda

But aim to replace with Pydio Cells for next release as linked above.