MichaIng/DietPi #5172: DietPi-PREP | no network connection causing script to fail

thinkbig1979, 2022-01-14:
I am running the latest PREP_SYSTEM_FOR_DIETPI.sh from the master branch on an Allwinner H3 TV box (Beelink X2), with Armbian Bullseye as a base installation.
The script runs fine until at some point it tries to execute a 'apt dist-upgrade' but fails, giving an error "Temporarily unable to resolve deb.debian.org". The failure turned out to be that there was no network connection at all. 

Thanks to the script menu, I was able to open a subterminal and troubleshoot. I was able to get the network services up and running again and get an IP address, after which the script successfully resumed.

No idea what caused the lack of networking, but thought it may catch out more users, so good to report it.

Not sure if it's possible in a bash script to check that network services are running correctly before executing internet reliant tasks, but it could make things smoother if it were.

MichaIng, 2022-01-14:
Many thanks for your report.

Generally this is a known issue in case netman or connman are managing the network connection. When those are uninstalled, the network connection is dropped completely: #2054
This is quite difficult to workaround in an automated way, especially without a reboot. But not sure whether this is really the issue in your case. To check:
```sh
dpkg -l network-manager connman ifupdown
ip a
cat /etc/network/interfaces
```

The Armbian Beelink X2 images are hopelessly outdated, the box not supported anymore for years: https://armbian.hosthatch.com/archive/beelinkx2/archive/
Also I do not see any U-Boot or kernel packages provided anymore, at least non matching the kernel version this image ships with, and using a newer Linux 5.x will likely not work with the pre-installed U-Boot 🤔. However, generally the preinstalled ones should work. Since the PREP script requires Debian Buster as minimum now, did you upgrade the Armbian Stretch image manually?

thinkbig1979, 2022-01-14:
Thankf for your quick reply. I actually used the Armbian Bullseye image for the OrangePiPC from January 2021, as that is also an Allwinner H3 device with mostly the same basic specs as my little TV box. That ran just fine and has a relatively new kernel etc.

MichaIng, 2022-01-14:
Good idea, indeed as long as the SoC is the same, it mostly works fine. Okay there is a server image only, where when I tested it last time it did actually work well without a dropping network connection, even that netman is at least installed. But to be true recently I run it in containers where network is provided by the host 🤔.

Actually we could try to check for connectivity (a default route check to keep it simple) and in case it is down apply a minimal `ifupdown` DHCP configuration to reestablish it:
```sh
if ! G_GET_NET -q gateway
then
iface=$(G_GET_NET iface)
echo -e "auto $iface\niface $iface inet dhcp" > /etc/network/interfaces
ifup --force "$iface"
done
```
This requires `isc-dhcp-client` and `ifupdown` to be installed, but both are on Armbian images.

thinkbig1979, 2022-01-15:
Makes sense. 
It may also make sense to strip out all known network utils from the existing base image as the first step in the process and install and activate those components that you know with reasonable certainty will stay alive during the execution of the script. Combine that with testing for connectivity at crucial times, and the whole things becomes a lot more robust.

MichaIng, 2022-01-16:
> It may also make sense to strip out all known network utils from the existing base image

That is exactly what we are doing, and this exactly causes the issue, as purging netman/connman breaks the network connection (when it is really managed by those high level networking tools), even just stopping their services does. So far we didn't find a way to "take over" an established network connection from netman/connman with `ifupdown` without loosing it at least temporarily, which breaks the whole PREP script when called from SSH. Basically a misbehaviour of netman/connman I'd say since they could simply leave the interface/connection states as they are when services are stopped. Now it makes me thing whether hard killing them `kill -9` could prevent them from touching interface/connection states 🤔. I'll play with this when I find time.

MichaIng, 2022-04-03:
I'll mark this as closed in favour of: #2054

closed 2022-04-03 by MichaIng