MichaIng/DietPi #5173: DietPi-Software | OctoPrint and Docker Compose fail on ARMv6

MichaIng, 2022-01-14:
Verified for Docker Compose. Running a `strace` on it:
```
# strace docker-compose
...
...
...
stat64("/usr/local/lib/python3.9/dist-packages/cryptography/hazmat/bindings/_rust/__init__.cpython-39-arm-linux-gnueabihf.so", 0xbef5d980) = -1 ENOENT (No such file or directory)
stat64("/usr/local/lib/python3.9/dist-packages/cryptography/hazmat/bindings/_rust/__init__.abi3.so", 0xbef5d980) = -1 ENOENT (No such file or directory)
stat64("/usr/local/lib/python3.9/dist-packages/cryptography/hazmat/bindings/_rust/__init__.so", 0xbef5d980) = -1 ENOENT (No such file or directory)
stat64("/usr/local/lib/python3.9/dist-packages/cryptography/hazmat/bindings/_rust/__init__.py", 0xbef5d980) = -1 ENOENT (No such file or directory)
stat64("/usr/local/lib/python3.9/dist-packages/cryptography/hazmat/bindings/_rust/__init__.pyc", 0xbef5d980) = -1 ENOENT (No such file or directory)
stat64("/usr/local/lib/python3.9/dist-packages/cryptography/hazmat/bindings/_rust", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
stat64("/usr/local/lib/python3.9/dist-packages/cryptography/hazmat/bindings/_rust.abi3.so", {st_mode=S_IFREG|0755, st_size=6710144, ...}) = 0
openat(AT_FDCWD, "/usr/local/lib/python3.9/dist-packages/cryptography/hazmat/bindings/_rust.abi3.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\fu\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0755, st_size=6710144, ...}) = 0
mmap2(NULL, 1939444, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb5ad4000
mprotect(0xb5c95000, 65536, PROT_NONE)  = 0
mmap2(0xb5ca5000, 36864, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1c1000) = 0xb5ca5000
close(3)                                = 0
mprotect(0xb5ca5000, 32768, PROT_READ)  = 0
--- SIGILL {si_signo=SIGILL, si_code=ILL_ILLOPC, si_addr=0xb5adb56c} ---
+++ killed by SIGILL +++
Illegal instruction
```
Looks like `cryptography` is the culprit here, but also some bindings are not present, not sure whether this is expected.

Checking OctoPrint now.

MichaIng, 2022-01-14:
Aside of the `serve` and `systeminfo`, all works, so it seems to be related to some system info command. Ran those with `python3 -vvv`, and here it is:
```
import 'psutil._psposix' # <_frozen_importlib_external.SourceFileLoader object at 0xb39eb610>
# trying /mnt/dietpi_userdata/octoprint/.local/lib/python3.9/site-packages/psutil/_psutil_linux.cpython-39-arm-linux-gnueabihf.so
Illegal instruction
```
Downgrading `psutil` to v5.8.0 solves the issue:
```sh
sudo -u octoprint pip3 install --user psutil==5.8.0
```
It is used to obtain free disk space to assure that updates run through, as part of the OctoPrint update check. Reporting that one, then checking further on Docker Compose/cryptography.

MichaIng, 2022-01-14:
Issue with `cryptography` was reported already: https://github.com/piwheels/packages/issues/273
Looks like related to the manual build since CI/CD failed, obviously.

MichaIng, 2022-01-14:
The other one reported already as well, looks like more recently compiled wheels are affected: https://github.com/piwheels/packages/issues/269

Joulinar, 2022-01-14:
thx for taking the time to investigate the case 👍

MichaIng, 2022-01-16:
Basically here the central comment of Ben (piwheels creator): https://github.com/piwheels/packages/issues/269#issuecomment-1007273390

I think it's not worth it to start implementing workarounds or so, since an unknown (probably still growing) number of various wheels is affected, all requiring individual version pinning, including dependencies, or individual development headers when not using piwheels in those cases.

Joulinar, 2022-01-16:
hmm to bad. Maybe we need to remove `piwheels` from being used on ARMv6 🤔

MichaIng, 2022-01-16:
As said, this means a bunch of wheels need to be compiled, requiring build tools, development headers etc, unreasonable to implement all this for ARMv6 exclusively. We need to wait until the issue has been identified and affected wheels recompiled.

MichaIng, 2022-01-17:
New issue to track status: https://github.com/piwheels/packages/issues/276

bennuttall, 2022-01-17:
A fix has been deployed, should be resolved shortly.

MichaIng, 2022-01-17:
Many thanks for letting us know. Hopefully the build servers do not run too hot 😄. I'll test with OctoPrint and Docker Compose (psutils and cryptography) tomorrow.

Joulinar, 2022-01-17:
Just give me a ping once ready. Will bring up my RPi1

MichaIng, 2022-01-19:
Verified it works all fine now, many thanks @bennuttall 👍.

closed 2022-01-19 by MichaIng

MichaIng, 2022-01-22:
Hmm, OctoPrint installed fine on my test (after failed build date) on RPi Zero W 🤔. Probably it was compiled then successfully?

Joulinar, 2022-01-22:
probably you had GCC installed?

MichaIng, 2022-01-22:
Probably, will test again. Did report anyway: https://github.com/piwheels/packages/issues/279

MichaIng, 2022-01-22:
Issue solved, latest `regex` wheel is now available on piwheels 👍.