MichaIng/DietPi #5174: DietPi-PREP | Script failing unexpectedly on running dietpi-set_software

thinkbig1979, 2022-01-15:
Here's a screenshot of the error. Sorry to have to do it this way.
[image]

Basically the script exits with this error.

Upon investigation it turns out that the /boot directory is completely empty, so logically the command cannot be run from that location, but I'm stumped as to why there's nothing in there.

Joulinar, 2022-01-15:
I guess it is related to this post? https://github.com/MichaIng/DietPi/issues/5172

thinkbig1979, 2022-01-15:
> I guess it is related to this post? #5172

It is the same device, yes. and it is the same error I referred to later in that thread, but I don't think the two issues are related. 
This time I did a dist upgrade and I changed the network settings before starting the prep script. It did not drop the network connection this time, but failed at this point with a perfectly working network connection.

MichaIng, 2022-01-16:
Hmm, this is an image with a single `ext4` partition, right? So not change that the `/boot` partition could have been unmounted between the previous (obviously successful) `/boot/dietpi/func/dietpi-set_software apt-mirror default` call and this one.

When you run into this error, could you open a subshell and past the output of these commands:
```sh
ls -l /boot
df -T /boot
```

thinkbig1979, 2022-01-16:
```bash
root@orangepipcplus:~# ls -l /boot
total 0
root@orangepipcplus:~# df -T /boot
Filesystem     Type 1K-blocks  Used Available Use% Mounted on
/dev/mmcblk2p1 ext4    245671  2051    226417   1% /boot
root@orangepipcplus:~#
```

thinkbig1979, 2022-01-16:
```bash
root@orangepipcplus:~# fdisk -l /dev/mmcblk2
Disk /dev/mmcblk2: 7.22 GiB, 7752122368 bytes, 15140864 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xc2979200

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk2p1        8192   532479   524288  256M 83 Linux
/dev/mmcblk2p2      532480 14989439 14456960  6.9G 83 Linux
root@orangepipcplus:~#
```

MichaIng, 2022-01-16:
> Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk2p1        8192   532479   524288  256M 83 Linux
/dev/mmcblk2p2      532480 14989439 14456960  6.9G 83 Linux

Where is this partitioning coming from? I just checked the Armbian Orange Pi PC+ Bullseye image and it has a single partition only. More strangely, the mounts are touched some hundred code lines _after_ your failure, so if this partition was not mounted before (hence the scripts installed to the root filesystem below), it shouldn't have been mounted at the point of failure (which would indeed "overmount" the DietPi scripts then). Can you check:
```sh
journalctl -u boot.mount
```

thinkbig1979, 2022-01-16:
```bash
root@orangepipcplus:~# journalctl -u boot.mount
-- Journal begins at Fri 2022-01-14 13:55:42 CET, ends at Sun 2022-01-16 20:45:37 CET. --
-- No entries --
root@orangepipcplus:~#
```

MichaIng, 2022-01-16:
Mysterious. You flashed the Bullseye image from [here](https://www.armbian.com/orange-pi-pc-plus/) from Aug 26 2021, right? It contains a single `ext4` partition only, so your partitioning shouldn't be possible like that. Are there other drives attached to the TV box, or does it have an internal eMMC or such which you may have accidentally booted into?

thinkbig1979, 2022-01-17:
Here's my workflow: 
1. flash the Armbian image to SD card
2. boot device from SD card containing Armbian image
3. use parted to make sure emmc is free of partitions
3. run 'nand-sata-install' utillity
4. select btrfs file system
5. let the utility install the OS to emmc
6. remove sd card and reboot from emmc
7. adjust network settings to prevent errors during prep script
8. start prep script 

Thats it. 
It could be that step 4 is causing trouble... perhaps btrfs requires a second partition to be created or something along those lines?

thinkbig1979, 2022-01-17:
I'm going to try to flash the Armbian image directly to the EMMC and see if it will boot. If that is the case then I'll check and see if the EMMC contains only 1 partition and if that is the case I'll try the prep script from there and report back. :-)

MichaIng, 2022-01-17:
Ah okay, I guess `nand-sata-install` creates a dedicated boot partition automatically then as the bootloader likely does not support Btrfs directly. The problem most likely then is that this dedicated boot partition is not mounted directly. Can you check `/etc/fstab` and `df` whether `/boot` is mounted and even has an `/etc/fstab` entry, before running PREP?

thinkbig1979, 2022-01-18:
Will do.

MichaIng, 2022-02-20:
Sorry for the late reply. This is a very confusing setup: The boot partition is mounted to `/media/mmcboot`, while `/media` is intended for removable temporary mounts only, usually and the boot partition is strictly required to be mounted whenever kernel or bootloader are upgraded, initramfs rebuilt, a boot config change done etc. Instead `/boot` is a bind mount only, not mounted by default.

I've no idea why `nand-sata-install` does it like this, but I'd instead do:
```
tmpfs /tmp tmpfs nosuid
UUID=c7887069-e8f3-4220-815a-dbf8c57a9f39 /boot ext4 noatime,commit=600,errors=remount-ro,x-gvfs-hide 0 1
UUID=4c7f6096-28af-49a5-bc7b-b845395c0757 / btrfs noatime,commit=600,compress=lzo,x-gvfs-hide 0 2
```
And of course `mount /boot` so that the boot partition is mounted directly where its supposed to be and in fact mounted for current session.

I'll add a `mount -a` to DietPi-PREP to cover such cases where essential/defined mounts are not mounted.

MichaIng, 2022-02-20:
This should solve it: https://github.com/MichaIng/DietPi/commit/1810907

thinkbig1979, 2022-02-21:
Thanks for all your efforts!

MichaIng, 2022-02-21:
If you find time, please test whether this solves the issue with PREP after `nand-sata-install`.

closed 2022-02-25 by MichaIng