MichaIng/DietPi #5175: Drive manager stuck

Giulianini, 2022-01-15:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | bullseye 0
- Kernel version | Linux DietPi 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux
- SBC model | RPi4

#### Steps to reproduce

1. sudo dietpi-drive_manager 

#### Expected behaviour

- starting

#### Actual behaviour

- stuck

#### Extra details

[image]

Joulinar, 2022-01-15:
Can you share following

```
cat cat /etc/fstab
lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid
dmesg -l err,crit,alert,emerg

```

Giulianini, 2022-01-15:
@Joulinar Sure, thx in advance

```
dietpi@DietPi:~$ cat /etc/fstab 
# You can use "dietpi-drive_manager" to setup mounts.
# NB: It overwrites and re-creates physical drive mount entries on use.
#----------------------------------------------------------------
# NETWORK
#----------------------------------------------------------------

#----------------------------------------------------------------
# TMPFS
#----------------------------------------------------------------
tmpfs /tmp tmpfs size=1024M,noatime,lazytime,nodev,nosuid,mode=1777
tmpfs /var/log tmpfs size=50M,noatime,lazytime,nodev,nosuid,mode=1777

#----------------------------------------------------------------
# MISC: ecryptfs, vboxsf (VirtualBox shared folder), gluster, bind mounts
#----------------------------------------------------------------

#----------------------------------------------------------------
# SWAP SPACE
#----------------------------------------------------------------
/var/swap none swap sw

#----------------------------------------------------------------
# PHYSICAL DRIVES
#----------------------------------------------------------------
PARTUUID=8b0ac6f5-02 / ext4 noatime,lazytime,rw 0 1
PARTUUID=8b0ac6f5-01 /boot vfat noatime,lazytime,rw 0 2
UUID=070552bb-51f1-4b19-b12f-e6da89d9896c /mnt/backup ext4 noatime,lazytime,rw,nofail,noauto,x-systemd.automount
dietpi@DietPi:~$ 
```

```
dietpi@DietPi:~$ lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid
NAME FSTYPE LABEL  SIZE RO TYPE MOUNTPOINT PARTUUID                             UUID
sda                7.3G  0 disk                                                 
└─sda1
     ext4          7.3G  0 part            38339152-edeb-43a7-9999-0adde5fba990 103b19f2-b7d9-499c-8b0a-5b4462699c82
mmcblk0
│                   29G  0 disk                                                 
├─mmcblk0p1
│    vfat          128M  0 part /boot      8b0ac6f5-01                          22B4-00D8
└─mmcblk0p2
     ext4         28.8G  0 part /          8b0ac6f5-02                          f406011c-615d-4dee-abe8-501b09f46b7b
dietpi@DietPi:~$ 
```

```
dietpi@DietPi:~$ dmesg -l err,crit,alert,emerg
[    4.316426] sd 0:0:0:0: [sda] No Caching mode page found
[    4.316433] sd 0:0:0:0: [sda] Assuming drive cache: write through
[ 1135.236585] sd 0:0:0:0: [sda] No Caching mode page found
[ 1135.236601] sd 0:0:0:0: [sda] Assuming drive cache: write through
dietpi@DietPi:~$ 
```

Joulinar, 2022-01-15:
ok looks like you switched USB disks. I see one that should be mounted to `/mnt/backup`

```
UUID=070552bb-51f1-4b19-b12f-e6da89d9896c /mnt/backup
```

But this is not the one connected to your device

```
UUID
103b19f2-b7d9-499c-8b0a-5b4462699c82
```

Could that be the case? Might be best to remove the entry manually from `/etc/fstab` if the original device linked to `/mnt/backup` is not available anymore.

Giulianini, 2022-01-15:
Thanks, i had also connected a zigbee device. Removing /etc/fstab line and all devices the command started working again. Thanks. Hope others could find this question useful

closed 2022-01-15 by Giulianini