MichaIng/DietPi #5177: DietPi-PREP script fails with public key errors

MarkMD0OMY, 2022-01-16:
#### Details:
- Date           | Sun Jan 16 01:44:37 GMT 2022
- DietPi version | v8.0.2 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       |  (ID=0)
- Kernel version | `Linux DietPirate2 5.10.0-10-amd64 #1 SMP Debian 5.10.84-1 (2021-12-08) x86_64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -q update`
- Exit code      | 100
- Software title | DietPi-PREP
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 https://deb.debian.org/debian bullseye-backports InRelease
Get:5 https://archive.raspberrypi.org/debian bullseye InRelease [23.5 kB]
Err:5 https://archive.raspberrypi.org/debian bullseye InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 82B129927FA3303E
Reading package lists...
W: GPG error: https://archive.raspberrypi.org/debian bullseye InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 82B129927FA3303E
E: The repository 'https://archive.raspberrypi.org/debian bullseye InRelease' is not signed.
```

Joulinar, 2022-01-16:
It would be helpful if you could share a little bit more on what you are doing. What device you are running on? What base image you used? It looks like you selected a Raspberry Pi device during the PREP install. Which seems to be incorrect as your kernel is `amd64`, indicating some x86 device or VM.

MarkMD0OMY, 2022-01-16:
Hi Joulinar,
Thanks for the response.
It is indeed an ESXi VM.
It is a fresh build using "debian-11.2.0-amd64-netinst.iso" and I tried choosing the Pi device and the VM and I get the same error. Another paste of the blue section is below after my latest attempt.
I'm choosing Bullseye (stable) for the base image also.

Thanks

#### Details:
- Date           | Sun Jan 16 12:30:22 GMT 2022
- DietPi version | v8.0.2 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       |  (ID=20)
- Kernel version | `Linux DietPirate2 5.10.0-10-amd64 #1 SMP Debian 5.10.84-1 (2021-12-08) x86_64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -q update`
- Exit code      | 100
- Software title | DietPi-PREP
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Get:4 https://archive.raspberrypi.org/debian bullseye InRelease [23.5 kB]
Hit:5 https://deb.debian.org/debian bullseye-backports InRelease
Err:4 https://archive.raspberrypi.org/debian bullseye InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 82B129927FA3303E
Reading package lists...
W: GPG error: https://archive.raspberrypi.org/debian bullseye InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 82B129927FA3303E
E: The repository 'https://archive.raspberrypi.org/debian bullseye InRelease' is not signed.
```
---------------------------------------------------------------------
[FAILED] DietPi-PREP | Unable to continue, DietPi-PREP will now terminate.

Joulinar, 2022-01-16:
selecting RPi device added the respective source list to your configuration now. You would need to remove entries for `archive.raspberrypi.org` from source files.

MarkMD0OMY, 2022-01-16:
Thanks so much Joulinar,
It appears to be working now.
I commented out the entry for archive.raspberrypi.org in my /etc/apt/sources.list.d/raspi.list file and the script continued to do its thing.

Thanks for the help, this was my fault, please consider this to be closed now.

Joulinar, 2022-01-16:
It should be save to remove entire list 

`rm /etc/apt/sources.list.d/raspi.list`

closed 2022-01-16 by Joulinar

MichaIng, 2022-01-16:
Makes sense that the RPi repository cannot be installed on a plain Debian image, since its key is not installed as trusted key. Selecting `VM` as target systems is closer to correct for an ESXi system, but AFAIK currently PREP assumes VM == `x86_64` at some steps, leading to other (but different) errors. It's basically this request: #3831

So far people converted the ARMv8 RPi image to use it as an ESXi VM, but if `rm /etc/apt/sources.list.d/raspi.list` alone works, this looks cleaner. I'm just wondering whether it then can boot as no VM-compatible bootloader (GRUB) is installed 🤔.