MichaIng/DietPi #5178: EXT4 USB HDD Not Automounting

capriciousduck, 2022-01-16:
My external ext4 formatted USB HDD is not automounting. I've used dietpi's drive manager. The drive is visible after reboot but it's not mounting unless and until I cd into the mount point (in this case /mnt/240GB/). And I'm using the latest Dietpi on Raspberry Pi 3B. Here is the content of my fstab file.

`# You can use "dietpi-drive_manager" to setup mounts.
# NB: It overwrites and re-creates physical drive mount entries on use.
#----------------------------------------------------------------
# NETWORK
#----------------------------------------------------------------

#----------------------------------------------------------------
# TMPFS
#----------------------------------------------------------------
tmpfs /tmp tmpfs size=1024M,noatime,lazytime,nodev,nosuid,mode=1777
tmpfs /var/log tmpfs size=50M,noatime,lazytime,nodev,nosuid,mode=1777

#----------------------------------------------------------------
# MISC: ecryptfs, vboxsf (VirtualBox shared folder), gluster, bind mounts
#----------------------------------------------------------------

#----------------------------------------------------------------
# SWAP SPACE
#----------------------------------------------------------------
/var/swap none swap sw

#----------------------------------------------------------------
# PHYSICAL DRIVES
#----------------------------------------------------------------
PARTUUID=9bcc366d-02 / ext4 noatime,lazytime,rw 0 1
PARTUUID=9bcc366d-01 /boot vfat noatime,lazytime,rw 0 2
UUID=b5ddbf9f-c2b1-4a95-8a3c-f7943e1ffacb /mnt/240GB ext4 noatime,lazytime,rw,nofail,noauto,x-systemd.automount`

Joulinar, 2022-01-16:
This is fully correct and expected as the drive has the `systemd.automount` flag

capriciousduck, 2022-01-16:
Oh, but I want the drive to be mounted across reboots and the drive manager added this. So what do I do? Just remove that flag?

Joulinar, 2022-01-16:
There is a misunderstanding I guess. The drive is automatically mounted if needed. Means, as soon as some process/user is going to access the drive, it is mounted. This is what you demonstrate yourself by using `cd` to access the drive 😉

capriciousduck, 2022-01-16:
But I run some docker containers (they start on boot) and after reboot they don't start as the drive is mounted "when needed".

Joulinar, 2022-01-16:
How is the HDD powered? Does it have an own PSU?

Can you reboot the system again and try to export whole `journalctl` into a file and attach it to the issue pls.

capriciousduck, 2022-01-16:
For now I got around the issue by creating a corn job to create a temporary file at /mnt/240GB for the HDD to mount. Now it's working as I want it to.

MichaIng, 2022-01-16:
Also the Docker daemon/container should trigger the immediate mount as fast as it accesses the mount point in any way, as any other read of write access does. So if this does not happen, then we'd need to investigate indeed.

Which container do you use? Are containers/images themselves stored in `/mnt/240GB`, e.g. as you moved `dietpi_userdata` to it, or is a shared directory stored there which is mounted into the container?

Joulinar, 2022-01-16:
maybe docker is going to start to early, while the HDD takes time to spin up?

MichaIng, 2022-01-16:
Docker daemon starts (when done by `dietpi-postboot` > `dietpi-services`) very late, and why does adding a cron job (the cron daemon starts even after Docker) fix it 🤔? I mean even if the container fails at a later time, after `systemctl start docker` unblocks and the cron daemon is started, it can only fix the issue when it does the mount point access before the Docker container would do. So basically a too early Docker startup cannot be the reason, as long as @capriciousduck did not do other changes to the startup sequence aside of an additional `@reboot` cron job?

MichaIng, 2022-02-20:
Marking this as closed. Feel free to reopen when required.

closed 2022-02-20 by MichaIng