MichaIng/DietPi #5179: SC16IS752: driver failure?

danergo, 2022-01-17:
Folks, 

I have created a DTS for SC16IS752 which let me add external UARTs to PI.

It worked absolutely perfect on RPi1 (CM), but during migration to RPi3 (CM), it causes me a lot of problems, and I just have found this on dmesg:

`[ 1055.331262] sc16is7xx spi0.0: ttySC3: Possible RX FIFO overrun: 64`

Are you aware of any bugs in sc16is7xx driver recently?

I'm currently on v7.8.2, with RPi CM 3+ (aarch64).

Any help would be a great help, this is blocking me now from my further developments.

Thank you!
Happy new year!

Joulinar, 2022-01-17:
Just as a side note. DietPi is not, and will not do any kernel or driver development. Issues related to them would need to be reported to RPi devs directly.

MichaIng, 2022-01-17:
There is a very old report which may be related: https://github.com/raspberrypi/linux/issues/4711
Ah here another newer one where your error message appears: https://github.com/raspberrypi/linux/issues/4351

What are the problems you face with it, aside of these logs which may not be related?

danergo, 2022-01-18:
I'm trying to flash a firmware onto an ESP32 board over an UART port provided by this SC16IS7xx chip, and it always fails: sometimes with "unknown command", sometimes "wrong crc". Mileage varies, but never succeeds.

On the RPi1 (CM), it always worked.

danergo, 2022-01-18:
Note is is using kernel of "5.10.63+".

danergo, 2022-01-20:
Thanks, Michael, I'm on a short leave now (few days), but I'll try it asap afterwards.

Also, I'll first try it with the dedicated UART of RPi to see if that's also affected or not.

danergo, 2022-02-02:
Hi, just a quick update (status shall still remain on "Waiting for user reply"): my (modified) testcase is finally running on `ttyAMA0`.
I'll report here on how many hours/days can it survive without any glitch, then I'll move back onto SC16IS752 and retest.

MichaIng, 2022-02-07:
Did you try to use a different bus? https://github.com/raspberrypi/firmware/blob/1e138de/boot/overlays/README#L2925-L2951

danergo, 2022-02-07:
No, not yet!

However there are two things that came into my mind.
I've created the DTS for my own for RPi1.

And it uses 14745600 as `clock-frequency`. Is it possible that on RPi3+ clock frequency differs from this and this causes all my headaches?

I'm going to ask this on RPi forum, thanks for the idea MichaIng! :)

MichaIng, 2022-02-07:
It is `14745600` by default for all models, when using the official overlays. But of course it makes sense to try the official ones then.

danergo, 2022-02-07:
Thank you!

I'll give it a try, however I have 2 of these ICs on a single SPI, which is not available with official DTS. SPI can handle more than device with using CE (or CS): Chip-Enable (Chip-Select).

I might have to give up with using one SPI and move onto two completely separated SPIs. Strange however that on RPi1 this was not an issue.

MichaIng, 2022-02-07:
Certainly something which can be cleared via RPi forum. I don't know much about the internals and which bus is internally connected how on the different models 😅.

MichaIng, 2022-02-07:
FYI: https://github.com/MichaIng/DietPi/commit/f4fdc44
So boot/kernel outputs and a login console can now be toggled for this type of serial devices as well 🙂.

danergo, 2022-02-16:
Hi! I just got some good news on this topic: https://github.com/raspberrypi/linux/issues/4885.
It is being said: https://github.com/raspberrypi/linux/issues/4885#issuecomment-1040347460

@MichaIng: Im not at the office for a week (about), but after I got back, I really wish to try this out. What do I need to do this?
Is it something easy with DietPi? What would you recommend?

This is not urgent, thank you!

MichaIng, 2022-02-16:
Nice, looks like Dom just fixed it in master branch. You could test it via:
```sh
apt install rpi-update
rpi-update
```
Note that this will upgrade your kernel to the potentially unstable development branch. It will be reverted automatically once a new stable release via `raspberrypi-kernel` APT package is applied (`apt upgrade`), which hopefully will contain this fix and Linux 5.15 already 🙂.

danergo, 2022-02-16:
Who is Dom? :) Phil?

danergo, 2022-02-16:
I have some little trouble with my ˋmmcblk0p1ˋ:

[image]

Is there an easy way to resize it with DietPi? (If not I can find a way)

Thank you!

MichaIng, 2022-02-16:
No this cannot be resized as the ext4 partition is starting directly behind it, 128 MiB is however more than sufficient, safe to proceed. Nasty that this warning is showing up now. Probably we should raise the boot partition to 256 MiB as well then for our images. But ~200 MiB are really just wasted then, especially on RPi 4 with 64-bit kernel installed only.

danergo, 2022-02-16:
aha, cool thank you! 

wow, for popcornmix I thought is a robot :) (sorry popcornmix - if you happen to read this)

MichaIng, 2022-02-16:
This would be a masterpiece of robotics, regarding communication on GitHub and RPi forum 😄.

danergo, 2022-02-22:
I just had time to test this. Although not extensively, but very rude: last time (before the fix) @115.2kbps a simple data transmission could not reach 30% of its progress before failing. After trying at least 100 times, it could not make any better than 30%. At above 400kbps, progress was not able to go above 10%.

With the fix, from 3 testing, all test were done at above 400k, and all test trials were completelely succeeded at 100%.

Very great fix! :)

closed 2022-02-22 by danergo

MichaIng, 2022-02-22:
Great, it will take probably another month until the new kernel lands in stable package, since it is the first 5.15.y version, but if someone else lands here, it can be applied via `rpi-update`: https://github.com/MichaIng/DietPi/issues/5179#issuecomment-1041594090