MichaIng/DietPi #5180: DietPi-Software | Cuberite - failing on aarch64

Joulinar, 2022-01-17:
@MichaIng 

looks like `Cuberite` is not working well on aarch64. It's failing with a strange error message `No such file or directory`

```
root@DietPi4:~# journalctl -u cuberite.service
-- Journal begins at Mon 2022-01-17 02:13:58 CET, ends at Mon 2022-01-17 15:21:06 CET. --
Jan 17 14:58:43 DietPi4 systemd[1]: Starting Cuberite Server (DietPi)...
Jan 17 14:58:43 DietPi4 systemd[2047]: cuberite.service: Failed to execute /mnt/dietpi_userdata/cuberite/Cuberite: No such file or directory
Jan 17 14:58:43 DietPi4 systemd[2047]: cuberite.service: Failed at step EXEC spawning /mnt/dietpi_userdata/cuberite/Cuberite: No such file or directory
Jan 17 14:58:43 DietPi4 systemd[1]: cuberite.service: Control process exited, code=exited, status=203/EXEC
Jan 17 14:58:43 DietPi4 systemd[1]: cuberite.service: Failed with result 'exit-code'.
Jan 17 14:58:43 DietPi4 systemd[1]: Failed to start Cuberite Server (DietPi).
```

Doesn't matter if files existing

```
root@DietPi4:~# cd /mnt/dietpi_userdata/cuberite/
root@DietPi4:/mnt/dietpi_userdata/cuberite# ls -la
total 6816
drwxr-xr-x  7 cuberite cuberite    4096 Jan 17 15:01 .
drwxrwxr-x  7 dietpi   dietpi      4096 Jan 17 14:58 ..
-rw-r--r--  1 cuberite cuberite     389 Jan 16 18:48 BACKERS
-rw-r--r--  1 cuberite cuberite    1586 Jan 16 18:48 CONTRIBUTORS
-rwxr-xr-x  1 cuberite cuberite 6796912 Jan 16 18:53 Cuberite
-rw-r--r--  1 cuberite cuberite   11641 Jan 16 18:48 LICENSE
drwxr-xr-x 11 cuberite cuberite    4096 Jan 16 18:48 Plugins
drwxr-xr-x  5 cuberite cuberite    4096 Jan 16 18:48 Prefabs
drwxr-xr-x  5 cuberite cuberite    4096 Jan 16 18:48 Protocol
-rw-r--r--  1 cuberite cuberite     692 Jan 16 18:48 README.txt
drwxr-xr-x  2 cuberite cuberite    4096 Jan 16 18:48 ThirdPartyLicenses
-rw-r--r--  1 cuberite cuberite    2748 Jan 16 18:48 brewing.txt
-rw-r--r--  1 cuberite cuberite     128 Jan 16 18:53 buildinfo.txt
-rw-r--r--  1 cuberite cuberite   61733 Jan 16 18:48 crafting.txt
-rw-r--r--  1 cuberite cuberite    2233 Jan 16 18:48 favicon.png
-rw-r--r--  1 cuberite cuberite    8484 Jan 16 18:48 furnace.txt
-rw-r--r--  1 cuberite cuberite   26108 Jan 16 18:48 items.ini
-rw-r--r--  1 cuberite cuberite    3009 Jan 16 18:48 monsters.ini
drwxr-xr-x  3 cuberite cuberite    4096 Jan 16 18:48 webadmin
-rw-------  1 cuberite cuberite      64 Jan 17 14:58 webadmin.ini
root@DietPi4:/mnt/dietpi_userdata/cuberite# sudo -u cuberite /mnt/dietpi_userdata/cuberite/Cuberite -d
sudo: unable to execute /mnt/dietpi_userdata/cuberite/Cuberite: No such file or directory
root@DietPi4:/mnt/dietpi_userdata/cuberite# sudo -u cuberite /mnt/dietpi_userdata/cuberite/Cuberite
sudo: unable to execute /mnt/dietpi_userdata/cuberite/Cuberite: No such file or directory
root@DietPi4:/mnt/dietpi_userdata/cuberite#
```

same result while running `Cuberite` own installer

```
root@DietPi4:/tmp# mkdir demo
root@DietPi4:/tmp# cd demo/
root@DietPi4:/tmp/demo# curl -sSfL https://download.cuberite.org | sh
Identifying kernel: Linux
Identifying platform: aarch64
Downloading precompiled binaries.
Done.
Cuberite is now installed, run using 'cd Server; ./Cuberite'.
root@DietPi4:/tmp/demo# ls -la
total 6784
drwxr-xr-x  7 root root     380 Jan 17 15:35 .
drwxrwxrwt 15 root root     540 Jan 17 15:35 ..
-rw-r--r--  1 root root     389 Jan 16 18:48 BACKERS
-rw-r--r--  1 root root    1586 Jan 16 18:48 CONTRIBUTORS
-rwxr-xr-x  1 root root 6796912 Jan 16 18:53 Cuberite
-rw-r--r--  1 root root   11641 Jan 16 18:48 LICENSE
drwxr-xr-x 11 root root     280 Jan 16 18:48 Plugins
drwxr-xr-x  5 root root     100 Jan 16 18:48 Prefabs
drwxr-xr-x  5 root root     120 Jan 16 18:48 Protocol
-rw-r--r--  1 root root     692 Jan 16 18:48 README.txt
drwxr-xr-x  2 root root     180 Jan 16 18:48 ThirdPartyLicenses
-rw-r--r--  1 root root    2748 Jan 16 18:48 brewing.txt
-rw-r--r--  1 root root     128 Jan 16 18:53 buildinfo.txt
-rw-r--r--  1 root root   61733 Jan 16 18:48 crafting.txt
-rw-r--r--  1 root root    2233 Jan 16 18:48 favicon.png
-rw-r--r--  1 root root    8484 Jan 16 18:48 furnace.txt
-rw-r--r--  1 root root   26108 Jan 16 18:48 items.ini
-rw-r--r--  1 root root    3009 Jan 16 18:48 monsters.ini
drwxr-xr-x  3 root root     140 Jan 16 18:48 webadmin
root@DietPi4:/tmp/demo# ./Cuberite
-bash: ./Cuberite: No such file or directory
root@DietPi4:/tmp/demo#
```

What is working is compiling it from source. But this of course is taking time.

BTW: no issue on ARMv7. This seems to be an external issue as we use pre-compiled binaries directly from `Cuberite`. Can you have a look and probably confirm?

MichaIng, 2022-01-17:
The new build requires an `armhf` linker/`libc6` package explicitly now 😢:
```
arm-binfmt-P: Could not open '/lib/ld-linux-armhf.so.3': No such file or directory
```
Since even the official installer uses that build for `aarch64` it was at least no intentional change. Let's report and hope this can be liberalised again.

MichaIng, 2022-01-17:
Hmm, it is a statically linked binary. While the error message is misleading, I know this from some cases where the binary is incompatible with the architecture. Generally ARMv8 is backwards-compatible, so not sure what the issue is.

Since the official installer explicitly pulls the `armhf` binary for `aarch64`, this was at least no intended change. Let's report upstream.

MichaIng, 2022-01-17:
Reported: https://github.com/cuberite/cuberite/issues/5386

MichaIng, 2022-01-22:
Solved with: https://github.com/MichaIng/DietPi/commit/ae056a0

closed 2022-01-29 by MichaIng