MichaIng/DietPi #5181: DietPi-Software | MineOS - install is failing on Bullseye

Joulinar, 2022-01-18:
@MichaIng 
there might be an issue with MineOS. Install is failing as it seems to require Python2 🙄 

```
[ INFO ] DietPi-Software | npm i --unsafe-perm --no-audit, please wait...

> posix@4.2.0 install /mnt/dietpi_userdata/mineos/minecraft/node_modules/posix
> node-gyp rebuild

gyp ERR! configure error
gyp ERR! stack Error: Can't find Python executable "python", you can set the PYTHON env variable.
gyp ERR! stack     at PythonFinder.failNoPython (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:484:19)
gyp ERR! stack     at PythonFinder.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:406:16)
gyp ERR! stack     at F (/usr/local/lib/node_modules/npm/node_modules/which/which.js:68:16)
gyp ERR! stack     at E (/usr/local/lib/node_modules/npm/node_modules/which/which.js:80:29)
gyp ERR! stack     at /usr/local/lib/node_modules/npm/node_modules/which/which.js:89:16
gyp ERR! stack     at /usr/local/lib/node_modules/npm/node_modules/isexe/index.js:42:5
gyp ERR! stack     at /usr/local/lib/node_modules/npm/node_modules/isexe/mode.js:8:5
gyp ERR! stack     at FSReqCallback.oncomplete (fs.js:158:21)
gyp ERR! System Linux 5.10.63-v8+
```

Installations is explicitly looking for an executable `python`. Faking it using `update-alternatives` did not work. It just create other error massages. Only way to work around was to install apt package `python-is-python2`

MichaIng, 2022-01-18:
> Can't find Python executable "python", you can set the PYTHON env variable.

Probably `export PYTHON=$(which python3)` works. Strange is that I tested it just during last development cycle where I removed the `python` package from dependencies to fix install on Bullseye. Probably architecture dependant or a recent change 🤔. Will test on `x86_64` tomorrow.

MichaIng, 2022-01-21:
Okay, installing the `posix` modules with Node.js v11 requires Python 2 indeed, doing the same with Node.js v17 does not. A pain that the MineOS Node.js installer depends on ancient Node versions 😞: https://github.com/hexparrot/mineos-node/issues/374

Joulinar, 2022-01-21:
Means we need to remove support on Bullseye?

MichaIng, 2022-01-21:
Still investigating. I tried with latest Node, but the install fails again at the `posix` module, while the same can be installed without issues globally or to another directory:
<details>

```
npm ERR! code 1
npm ERR! path /mnt/dietpi_userdata/mineos/minecraft/node_modules/posix
npm ERR! command failed
npm ERR! command sh -c node-gyp rebuild
npm ERR! make: Entering directory '/mnt/dietpi_userdata/mineos/minecraft/node_modules/posix/build'
npm ERR!   CXX(target) Release/obj.target/posix/src/posix.o
npm ERR! make: Leaving directory '/mnt/dietpi_userdata/mineos/minecraft/node_modules/posix/build'
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@8.4.1
npm ERR! gyp info using node@17.4.0 | linux | x64
npm ERR! gyp info find Python using Python version 3.9.2 found at "/usr/bin/python3"
npm ERR! gyp info spawn /usr/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'make',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/mnt/dietpi_userdata/mineos/minecraft/node_modules/posix/build/config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/root/.cache/node-gyp/17.4.0/include/node/common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=/root/.cache/node-gyp/17.4.0',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=/usr/local/lib/node_modules/npm/node_modules/node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=/root/.cache/node-gyp/17.4.0/<(target_arch)/node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=/mnt/dietpi_userdata/mineos/minecraft/node_modules/posix',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! In file included from ../src/posix.cc:1:
npm ERR! ../../nan/nan.h: In function ‘void Nan::AsyncQueueWorker(Nan::AsyncWorker*)’:
npm ERR! ../../nan/nan.h:2298:7: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
npm ERR!  2298 |     , reinterpret_cast<uv_after_work_cb>(AsyncExecuteComplete)
npm ERR!       |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
npm ERR! In file included from ../../nan/nan.h:2884,
npm ERR!                  from ../src/posix.cc:1:
npm ERR! ../../nan/nan_typedarray_contents.h: In constructor ‘Nan::TypedArrayContents<T>::TypedArrayContents(v8::Local<v8::Value>)’:
npm ERR! ../../nan/nan_typedarray_contents.h:34:43: error: ‘class v8::ArrayBuffer’ has no member named ‘GetContents’
npm ERR!    34 |       data   = static_cast<char*>(buffer->GetContents().Data()) + byte_offset;
npm ERR!       |                                           ^~~~~~~~~~~
npm ERR! In file included from ../../nan/nan.h:54,
npm ERR!                  from ../src/posix.cc:1:
npm ERR! ../src/posix.cc: At global scope:
npm ERR! /root/.cache/node-gyp/17.4.0/include/node/node.h:843:7: warning: cast between incompatible function types from ‘void (*)(v8::Local<v8::Object>)’ to ‘node::addon_register_func’ {aka ‘void (*)(v8::Local<v8::Object>, v8::Local<v8::Value>, void*)’} [-Wcast-function-type]
npm ERR!   843 |       (node::addon_register_func) (regfunc),                          \
npm ERR!       |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
npm ERR! /root/.cache/node-gyp/17.4.0/include/node/node.h:877:3: note: in expansion of macro ‘NODE_MODULE_X’
npm ERR!   877 |   NODE_MODULE_X(modname, regfunc, NULL, 0)  // NOLINT (readability/null_usage)
npm ERR!       |   ^~~~~~~~~~~~~
npm ERR! ../src/posix.cc:709:1: note: in expansion of macro ‘NODE_MODULE’
npm ERR!   709 | NODE_MODULE(posix, init);
npm ERR!       | ^~~~~~~~~~~
npm ERR! make: *** [posix.target.mk:113: Release/obj.target/posix/src/posix.o] Error 1
npm ERR! gyp ERR! build error
npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
npm ERR! gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:520:28)
npm ERR! gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:291:12)
npm ERR! gyp ERR! System Linux 5.10.0-10-amd64
npm ERR! gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd /mnt/dietpi_userdata/mineos/minecraft/node_modules/posix
npm ERR! gyp ERR! node -v v17.4.0
npm ERR! gyp ERR! node-gyp -v v8.4.1
npm ERR! gyp ERR! not ok
```
</details>

Installing `posix` alone, without using the whole `package-lock.json`, however works, and afterwards also the whole project with `package-lock.json` can be installed 😕.

Now trying to install again everything with Node 17.

... and it works. Web UI opens fine, now trying to startup a Minecraft server. Not sure about the issue, why `posix` compiling fails when installing everything in one step while it installs fine alone. Also not sure why we found Node v11 to be the latest supported version, at least now it works, earlier it probably was as well somehow related to module compile failures or so.

MichaIng, 2022-01-21:
Works all fine now, but there is another issue on Bullseye and up which basically means that we do need to disable MineOS on Bullseye and up, or add a note to our docs with a fat warning on the workaround: #441

Until at least #4551, I however will not advise to our users to weaken the root user's password hash only to run a Minecraft server. When we have a dedicated `mineos` user for which we can store the its personal password (not any other) with SHA512, then okay, but much better would be to have the new default hash supported by MineOS itself.

Joulinar, 2022-01-21:
yep password was/is a know issue https://dietpi.com/phpbb/viewtopic.php?p=38223#p38223

Following fix it usually 

```sh
chpasswd -c SHA512 <<< 'root:dietpi'
```

closed 2022-01-29 by MichaIng