MichaIng/DietPi #5182: Dietpi 7.9.3 > 8.0.2 Upgrade Question

N7-BADA, 2022-01-18:
So since DietPi 8.x was released i wanted to wait till 2022 before upgrading as i saw the post about manually upgrading and didn't have the time for it.Today i logged into my DietPi and to my surprise i saw i could run the "dietpi-update" command. now my question is, is this possible and the manual upgrading isn't needed anynore or is this a bug that it allows me to do this as i didnt see it before?

thanks in advance.
[image]

Joulinar, 2022-01-18:
What manual update you are referring to? Probably a misunderstanding?

`dietpi-update` will update DietPi scripts.

A manual update would be needed in case you like to perform a destro upgrade from Debian Stretch to Buster. Or from Buster to Bullseye. Such a destro upgrade is **not** triggered by `dietpi-update`. What is the Debian version you are running on?

`cat /etc/os-release`

N7-BADA, 2022-01-18:
Oh probably a misunderstanding then. i thought that dietpi 7.x has reached EOL. and from there on 8.x would be Bullseye (hence me thinking 7.x > 8.x wouls be Buster > Bullseye).

cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

So it is fine if i do a 'dietpi-update' now?

Joulinar, 2022-01-18:
Yes `dietpi-update` is supported on Buster as well as Bullseye. We just dropped support for Stretch and migrate these Stretch systems to a separate branch as they will not get V8. We will continue supporting Buster the next 1.5 year. Nevertheless updating to Bullseye is a good idea as well

N7-BADA, 2022-01-18:
> Yes `dietpi-update` is supported on Buster as well as Bullseye. We just dropped support for Stretch and migrate these Stretch systems to a separate branch as they will not get V8. We will continue supporting Buster the next 1.5 year. Nevertheless updating to Bullseye is a good idea as well

Thanks a lot for the fast answers!. yes i will attempt an upgrade to Bullseye soon. just need to get some extra drives to make an remote backup.

closed 2022-01-18 by N7-BADA

MichaIng, 2022-01-18:
If you say so, all older DietPi versions are EOL as fast as a new release is there, meaning we do not provide patches or old versions and in case of related issues, we will refer to `dietpi-update` 😉. But this not related to Debian versions, where we usually support three versions, currently Debian 10 (Buster), Debian 11 (Bullseye) and Debian 12 (Bookworm), while the last is Debian testing and hence not 100% supportable by `dietpi-software` due to constantly appearing APT package upgrades with breaking changes included. We started to do regular testing of our software implementations for Bookworm: https://github.com/MichaIng/DietPi/wiki/Debian-Bookworm-testing

With DietPi v8.0 we dropped support for the older Debian 9 (Stretch), but still offered a DietPi patch version 7.9.4 for it, probably will ship further patches, when urgent security issues or bugs arise. Debian 10 (Buster) is however still fully supported. When you find time for it, an upgrade to Debian 11 (Bullseye) is however recommended, to benefit from newest APT packages.

Hehe, double post 😄.