MichaIng/DietPi #5183: dietpie-services drop-in paths are not set correctly

JJ-Author, 2022-01-18:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bullseye 1
- Kernel version | `uname -a`
Linux DietPi 5.10.63+
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
 rpi 1b

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce
1. run dietpie-services
2. add service (e.g. bluetooth)
3. select the service and hit "edit" 
4. confirm creation of drop-in with OK

problem b.
similar but changing the affinity and scheduling priority of the service

#### Expected behaviour
the edited file or affinitiy and priority config should be stored in the drop-in folder of the service
and the changes should take effect after restarting the service

#### Actual behaviour
the dropin folder is created correctly but the files are stored in the root path
root@DietPi:~# ll / | grep diet
-rw-r--r--   1 root root   216 Jan 17 22:50 dietpi-process_tool.conf
-rw-r--r--   1 root root   646 Jan 17 22:43 dietpi-services_edit.conf

of course these changes there dont have any effect

#### Extra details
tried to change service from systemd managed to dietpie managed but the same bug is happening
note:
the dropin-folder is created at a location I assume to be correct. so I guess just a variable was forgotten to be set correctly in dietpie-services tool

MichaIng, 2022-01-18:
Many thanks for reporting. This has been solved for v8.1 already 🙂: https://github.com/MichaIng/DietPi/commit/36d35e9

JJ-Author, 2022-01-18:
oh wow cool. thanks for the quick reply.

JJ-Author, 2022-01-18:
should have had a look on the dev branch commits not just on the master  🙈

MichaIng, 2022-01-18:
No problem, not bad to have an issue about this here on GitHub as well.

closed 2022-01-29 by MichaIng