MichaIng/DietPi #5184: ODroid C2 -  unmet dependencies installing minidsp-rs

younger38, 2022-01-19:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bullseye

- Kernel version | `uname -a`
Linux DietPi 5.10.81-meson64 #21.08.6 SMP PREEMPT Mon Nov 22 11:21:51 UTC 2021 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

Odroid C2 (aarch64)

- Power supply used | (EG: 5V 1A RAVpower)
USB port on Power Bar 
- SDcard used | (EG: SanDisk ultra)
- San Disk Extreme 32GB

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
minidsp-rs (https://github.com/mrene/minidsp-rs/releases)
- Was the software title installed freshly or updated/migrated?
Fresh install
- Can this issue be replicated on a fresh installation of DietPi?
Yes

- Bug report ID | `echo $G_HW_UUID`
d6baff9e-2bfb-49d5-8eaf-7da5f7e3eff5

#### Steps to reproduce

1. ... Follow directions on https://ezbeq.readthedocs.io/en/latest/rpi/ on installing minidsp-rs
2. ... wget https://github.com/mrene/minidsp-rs/releases/download/<version>/<filename> (tried latest, v0.1.3 and v.0.1.0 with same result)
3.  sudo apt install ./<filename>

#### Expected behaviour

- ... Should install minidsp-rs

#### Actual behaviour

Note, selecting 'minidsp:armhf' instead of './minidsp_0.1.3_armhf.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
minidsp:armhf : Depends: libusb-1.0-0:armhf (>= 2:1.0.22) but it is not installable
Depends: libc6:armhf (>= 2.17) but it is not installable
Depends: libudev1:armhf (>= 183) but it is not installable
E: Unable to correct problems, you have held broken packages.

I have confirmed all the noted packages seem to be installed (to the best of my limited Linux ability)

sudo apt-get install libusb-1.0-0
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libusb-1.0-0 is already the newest version (2:1.0.24-3).

sudo apt-get install libc6
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libc6 is already the newest version (2.31-13+deb11u2).

sudo apt-get install libudev1
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libudev1 is already the newest version (247.3-6).

#### Extra details

- ...
Hope this isnt wasting your time, I searched everywhere for something similar, this could be related to minidsp-rs but wanted to confirm here, thanks!

Joulinar, 2022-01-19:
Hi,

Looks like you tried to install a 32bit package on a 64bit system. This of course could not fullfil required dependencies. As far as I can see, MiniDSP did not offer a pre-compiled package for arm64. I guess you would need to compile it manually from source.

MichaIng, 2022-01-19:
You could try to revive this issue: https://github.com/mrene/minidsp-rs/issues/198
It was closed as OP managed to compile it while the maintainer already offered to look into providing `arm64` builds.

younger38, 2022-01-19:
Thanks @Joulinar and @MichaIng really appreciate your above and beyond type help for a newbie, I am trying to compile my own based on your feedback and if that doest work I will try reviving that issue! Thanks again, greatly appreciated!!

MichaIng, 2022-02-06:
Did you succeed, or can this issue be closed?

MichaIng, 2022-02-16:
Issue solved upstream: https://github.com/mrene/minidsp-rs/issues/198#issuecomment-1017054201
@mrene many thanks for magically reacting to this foreign issue (?) 👍🙂!

Mark this this as closed.

closed 2022-02-16 by MichaIng