MichaIng/DietPi #5185: Smokeping failing to curl v7.74 regression - needs package update to v7.80 or higher

matellis, 2022-01-19:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.0.2
- Distro version | bullseye
- Kernel version | Linux pihole-b 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux
- SBC model | RPi 4 Model B (aarch64)

#### Additional Information (if applicable)
- Software title | Smokeping
- Was the software title installed freshly or updated/migrated? Freshly installed
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce
1. Install smokeping
2. Setup a curl probe

#### Expected behaviour
It works.

#### Actual behaviour
The curl probes are disabled due to a lack of timeout support, and you see this in your logs:

```
Jan 18 20:08:25 pihole-b smokeping[346556]: Curl: Note: your curl doesn't support the timeout feature (option -m), disabling it
Jan 18 20:08:25 pihole-b smokeping[346556]: Curl: Note: your curl doesn't support the interface feature (option --interface), disabling it
```
#### Extra details
Curl 7.74 introduced a regression where it started reporting time values in microseconds instead of seconds. This is breaking Smokeping curl probes, rendering this version of Dietpi as unable to host any smokeping installation using curl probes.

The solution is to update the curl package to 7.80. See https://github.com/oetiker/SmokePing/issues/266 for more details.

Smokeping has decided not to fix this, citing an updated package will cure the problem. Not sure if this is something we wait on upstream for, or if it's already happened and we need to bring the change downstream. Mostly filing this ticket to help any lost poor souls who can't figure out why their smokeping installation is failing suddenly.

matellis, 2022-01-19:
https://tracker.debian.org/news/1294015/accepted-curl-7810-1bpo111-source-amd64-all-into-bullseye-backports-bullseye-backports/

matellis, 2022-01-19:
7.81 doesn't fix it either... grr.. off to smokeping

closed 2022-01-19 by matellis

Joulinar, 2022-01-19:
Just as side note. DietPi is not compiling any packages like `curl`. We use the packages available on global Debian and/or Raspbian repository.

MichaIng, 2022-01-19:
Just in case others land here, `curl` can be upgraded from backports via:
```sh
apt install curl/bullseye-backports
```
However, it shouldn't change anything on this matter since Debian added a patch for this to its v7.74 package already: https://metadata.ftp-master.debian.org/changelogs//main/c/curl/curl_7.74.0-1.3+deb11u1_changelog
```
  * Add upstream patch bc7ecc7 so curl -w times shown as seconds with
    fractions (Closes: #989064)
```
So you seem to face a different bug.

matellis, 2022-01-20:
It's a really silly bug: `curl --help` got summarized in v7.74 and later. So any script which uses the help option to see which features are available now breaks. Tobi at Smokeping accepted a PR to fix this, see https://github.com/oetiker/SmokePing/issues/266 for more details. Apologies for clogging up the issues backlog, it did seem that we needed a back port at the time. After applying the back port I figured out what was really going on.