MichaIng/DietPi #5186: [Question] How to make DietPi IP Address static when running on Hyper-V?

LazyGeniusMan, 2022-01-19:
Hi, I'm trying to get DietPi working on Hyper-V, mainly for pihole so I need static IP address for DietPi. How to do that ?

Right now everytimes I shutdown/restart my host OS, DietPi always have different IP address. I tried to make the guest OS MAC address static in Hyper-V setting but the IP still changed after host os restarted.

If I make network config in DietPi static, it won't connect to internet (propbably because the real IP is changed ?)

Joulinar, 2022-01-19:
I had as well challenges to create a new external vSwitch on HyperV. It always killed my WiFi. But is is needed to allow communication to extern. Especially if you like to use PiHole. For me it was needed to go down into Windows network settings and disable/enable WiFi adapter while HyperV was trying to create the new vSwitch. Something where it was needed to be quick to switch between HyperV and network settings 🤣 

Actually, does the HyperV VM is getting an IP address assigned from your local network? I mean is the same network segment?

LazyGeniusMan, 2022-01-19:
> I had as well challenges to create a new external vSwitch on HyperV. It always killed my WiFi

Yea, it's never succeed

> Actually, does the HyperV VM is getting an IP address assigned from your local network? I mean is the same network segment?

I'm not really understand networking, but probably yea ? it has LAN IP, but when I configure DNS to that IP it's not querying to pihole eventhough I can access the dashboard via browser in Host OS, probably because of what you said about allow communication to extern ?

MichaIng, 2022-01-19:
One thing with VMs connected via bridge mode through a WiFi adapter (on host) is that you cannot use IPv6. Also within the VM is keeps appearing as Ethernet adapter (emulated), but you seem to know that already. Indeed it may be required to disable and re-enable the WiFi adapter on the host when Hyper-V was freshly installed, since it adds a Hyper-V virtual switch driver. I just recognise that I cannot enable it on the WiFi adapter, at least not while it is enabled on the Ethernet adapter 🤔. But I remember I got this working once when testing. Probably you need to manually disable (uncheck) the Hyper-V virtual switch driver for the Ethernet adapter manually first.

With my routers I can assign a DHCP reservation for the VM (which appears as dedicated client) within the DHCP range, so that the static IP assigned within the VM does not conflict the one which the router aims to assign.

To assign a static IP, you can use `dietpi-config` > Network Options: Adapters > Ethernet > switch from DHCP to STATIC > Copy (to assure the previously via DHCP assigned settings are used) > Apply.

Joulinar, 2022-01-19:
The trick is to open HyperV Switch option and Windows Network option in parallel. Try to create a new external vSwitch and hit ok to create the switch. Now HyperV will take some time. Be quick and switch over to Windows network settings. Select the WiFi adapter and disable it. Once done, enable it again. If you are quick enough, HyperV should now create the new vSwitch

MichaIng, 2022-01-19:
From your Pi-hole dashboard vs DNS IP I see different address ranges. The Pi seems to believe it has the IP `172.24.64.1`, or is this the router IP and probably the address range of the clients/Windows host? When the clients are within a different subnet, Pi-hole denies to answer queries as long as you not not explicitly allow requests from all origins. Is that what you mean that the IP changes on every reboot? That the whole subnet changes/is a different one is indeed strange, in case I do not misinterpret the reverse proxy requests.

Joulinar, 2022-01-19:
I would recommend to setup HyperV to use external adapter if Pihole should be used as network AdBlocker. Otherwise other network device will most probably not able to reach Pihole from local network.

Joulinar, 2022-01-19:
`192.168.8.101` is the IP of your host system?

My system is in German but should be ok I guess. I have an adapter called `WLAN`. There I can right click and disable/enable

[image]

LazyGeniusMan, 2022-01-19:
> `192.168.8.101` is the IP of your host system?

No, it's guest system (dietpi), my host is 192.168.8.100

> My system is in German but should be ok I guess. I have an adapter called `WLAN`. There I can right click and disable/enable

The one on the left is bridged created by Hyper V external network?, btw could you show me your Hyper V Virtual Switch Manager setting?

> But I see you are using VirtualBox as well.

Yea, I was planning to switch to Hyper V since now I have Windows 10 Pro license so I dont need 3rd party software for my VM.

> Might be better to use this instead of HyperV.

You meant VBox?

Joulinar, 2022-01-19:
This is my vSwitch setting to create an external one.
[image]

> No, it's guest system (dietpi), my host is 192.168.8.100
Is VBox running? If yes, the IP 192.168.8.101 might be your VBox system

LazyGeniusMan, 2022-01-19:
Aight thanks, disabling and then enabling my wifi when adding Virtual Switch do the trick :)

LazyGeniusMan, 2022-01-20:
Bruh, now my device connected through my Laptop Mobile Hotspot from this menu (the one with `1/8 - Yansyah`) dont have internet connection.
[image]

I tried to set the Hotspot adapter (`Local Area Connection* 2`) DNS to my pihole but it's still dont have internet connection :/

MichaIng, 2022-01-20:
Try to disable/uncheck all virtualizers bridge drivers on the hotspot adapter. Doesn't it require a static IP when you want to use it as hotspot for other WiFi clients to connect? There is no DHCP server which provides IPs for the WiFi network I suppose. Also the DNS server for this adapter has no effect as it is used by the hotspot host only which does not connect to the Internet through the WiFi network.

MichaIng, 2022-01-20:
I guess it is bridged? At least the bridge driver is enabled ("Bridge Driver" is none of the virtualizers), but indeed the flag beside "enabled" is missing 🤔. But if I'm not mistaken, the bridging adapter still requires a static IP, or is it managed as well by the DHCP server of the router on `192.168.0.0/24` when successfully bridged? Long ago that I played with network bridges, aside of VMs, to be true 😅.

LazyGeniusMan, 2022-01-21:
> Try to disable/uncheck all virtualizers bridge drivers on the hotspot adapter.

@MichaIng like this?
[image]

> but indeed the flag beside "enabled" is missing 🤔

Even if I want to add the hotspot adapter to bridge I can't, there is no hotspot adapter in network bridge checklist
[image]

> but probably the hotspot IP range is a different one.

@Joulinar looks like it has different range
[image]

Joulinar, 2022-01-21:
can you try to set inside PiHole DNS settings your interface to `Permit all origins`

LazyGeniusMan, 2022-01-22:
No effect, looks like I have to use VirtualBox haha, it works flawlessly, but auto start VM headless kinda need manual conf :(, I thought Hyper V would be VirtualBox's flawless + easy autorun haha, but it's worse for me, only easy to autorun, maybe because I'm using Wireless Adapter :/

MichaIng, 2022-01-22:
> Also my PC is connected to vSwitch instead of directly to my WiFi Adapter, is this normal?

This actually shouldn't be required. Your PC should be connected directly via WiFi adapter, while the Hyper-V vSwitch is a driver which you an enable/disable within the WiFi adapter settings. But don't do it manually, it behaves very bad, messes with all other driver/option toggles etc, a total mess, instead it should be toggled correctly when managing via Hyper-V settings.

LazyGeniusMan, 2022-01-31:
Just a little bit update, now I can share my connection, apparently I just need to turn on sharing option in the virtual switch setting :)
[image]