MichaIng/DietPi #5187: After Kernel Update to 5.10.92+ (ARMv8(64Bit)) - Hold on RainbowScreen

Kebehsenuef, 2022-01-19:
#### Required Information
- DietPi version | 8.0.2
- Distro version | no Boot
- Kernel version | Linux Eurebia4 5.10.92-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux
- SBC model | Raspberry Pi4 4GB Ram
- Power supply used | Meanwell LRS-50-5
- SDcard used | USB WD Blue SATA SSD M.2 2280 1 TB

#### Steps to reproduce
apt update
apt upgrade
Upgrade raspberrypi-kernel and raspberrypi-bootloader from V5.10.63-V8+ to V5.10.92-V8+

After login I was offered the update. Updated with "apt update" + "apt upgrade". After a restart, the Pi4 was no longer accessible. Red LED was set up as a heartbeat. the heartbeat was not reached. "config.txt" has been adjusted.
After restarting, the Pi only showed the rainbow, USB disk was recognized.
clean Dietpi image copied to SD card.
Pi4 started without USB memory or anything else. monitor, network and keyboard only. The same update was carried out during the initial setup. after restart i have the same problem. Launches only to the rainbow. Sorry, everything is google translated. LG Steve

MichaIng, 2022-01-19:
I applied the same upgrade to my RPi's but didn't do a reboot yet. Will try that now.

> "config.txt" has been adjusted.

What do you mean by that? The upgrade shouldn't touch `config.txt` and it shouldn't be required to do that (manually).

Joulinar, 2022-01-19:
testing on my RPi3B+ 32bit and RPi4B. Both not booting after the upgrade as well.

MichaIng, 2022-01-19:
Same here, booting until a certain stage, then CPU usage LED (the way I configured it) stays lit. Dammit. I'll check the boot logs.

Joulinar, 2022-01-19:
Nothing reported yet on RPi side https://github.com/raspberrypi/firmware/issues

MichaIng, 2022-01-19:
I'm just writing a report.

MichaIng, 2022-01-19:
https://github.com/raspberrypi/firmware/issues/1686

Kebehsenuef, 2022-01-19:
I have modify the config.txt to the all hdmi outputs include see the rainbow

Joulinar, 2022-01-19:
kernel is the same as DietPi don't have own kernel.

Kebehsenuef, 2022-01-19:
> [raspberrypi/firmware#1686](https://github.com/raspberrypi/firmware/issues/1686)

Thx Micha

MichaIng, 2022-01-19:
System log was not even starting, so no chance to see what happened 😢. I'll try now to boot with the old bootloader, if still not working with the old kernel, to narrow it down.

vitostrano, 2022-01-19:
Same problem on rasp zero w. Maybe boot partition size? I remember a warning, using rpi-update kernel

MichaIng, 2022-01-19:
Nasty that it is so late at night in Europe, hopefully someone is still aware which can set this upgrade on hold repository wise...

No the boot partition size is not the problem, my systems have plenty of free space on it.

Joulinar, 2022-01-19:
@MichaIng could we supply a hotfix to disable `APT package update notification` on our banner?

Kebehsenuef, 2022-01-19:
> System log was not even starting, so no chance to see what happened 😢. I'll try now to boot with the old bootloader, if still not working with the old kernel, to narrow it down.

I have a backup. So i copied the old boot files to boot partition completely. After restart system go on. But after "apt install raspberrypi-kernel - - reinstall" sytem hangs @ rainbow

MichaIng, 2022-01-19:
So if really affects all systems and also all bootloader firmware files, e.g. my RPi 2 loads the cut-down firmware.

@Kebehsenuef 
Be sure to also copy the old /lib/modules/ content over, else the old kernel cannot load modules.

MichaIng, 2022-01-19:
The bootloader (alone) doesn't solve it, trying now to use old kernel with new bootloader.

vitostrano, 2022-01-19:
> So if really affects all systems and also all bootloader firmware files, e.g. my RPi 2 loads the cut-down firmware.
> 
> @Kebehsenuef Be sure to also copy the old /lib/modules/ content over, else the old kernel cannot load modules.

With Old /lib/modules and old boot partition can i repair my System?

MichaIng, 2022-01-19:
Yes

Joulinar, 2022-01-19:
but this require a backup done before right? To be able to copy back files.

MichaIng, 2022-01-19:
The new files are broken, so not sure what the backup shall be used for 😄. I'm just manually extracting the old package from https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/?C=M;O=D and copying them in place. Same can be done from a backup of course. All on external Linux system.

Joulinar, 2022-01-19:
we have first reports on forum as well 🤕

vitostrano, 2022-01-19:
Done. Fixed using backup

vitostrano, 2022-01-19:
Apt-mark hold Second step :/

MichaIng, 2022-01-19:
As the new package is installed already, no need for `apt-mark hold`. The next release will have it fixed 😉.

Joulinar, 2022-01-19:
crossing fingers for people who activate automatic apt package updates 🙄

ravenclaw900, 2022-01-19:
I haven't upgraded my RPi 4B yet (and I just backed it up). Is there anything I should test?

MichaIng, 2022-01-19:
Does someone have a UART console for RPi GPIO headers? I have only one for a dedicated debug UART plug 🤔.

Joulinar, 2022-01-19:
> Does someone have a UART console for RPi GPIO headers? I have only one for a dedicated debug UART plug 🤔.

nope sorry

MichaIng, 2022-01-19:
Definitely both needs to be restored, kernel and bootloader, but this is probably due to a strictly required version match, or a recent change with makes one depend on the other.

Joulinar, 2022-01-19:
Ok for time being, kernel package has been removed and is not available anymore. At least no failing systems over night 😄

vitostrano, 2022-01-19:
Before recovery boot and modules, i have tried to Restore all files from and old backup from Linux. I have Lost some file permission. There's an easy way to check and fix permiss?

Joulinar, 2022-01-19:
does your system is booting up? If yes, could you still execute a `dietpi-backup` restore?

```
dietpi-backup -1
```

macobra, 2022-01-19:
> testing on my RPi3B+ 32bit and RPi4B. Both not booting after the upgrade as well.

Same here. RPi4 2GB, RPi3B+ 1GB. No booting after update.

MichaIng, 2022-01-20:
The repo has been reverted already, so the upgrade is not offered anymore. @macobra do you manage to restore a backup, respectively the individual needed files from old kernel and bootloader or do you need help with that?

macobra, 2022-01-20:
> @macobra do you manage to restore a backup, respectively the individual needed files from old kernel and bootloader or do you need help with that?

Hi MichaIng, thanks for coming back so quickly. I could reinstall a backup but I would appreciate to learn what has happened so I might fix bugs like this myself in the future.

MichaIng, 2022-01-20:
The latest `raspberrypi-bootloader` and/or `raspberrypi-kernel` packages seem to be faulty, nothing you can really do something about to prevent as usually they are rock solid. When the system fails to boot at this early stage, then only bootloader, kernel or boot config can be the culprit. When there was a kernel/bootloader upgrade right before the reboot fails, then it is a clear thing 😉.

macobra, 2022-01-20:
Replacing kernel/bootloader/dtbs/dats/elfs and overlays from latest DietPi-image brought it back to live.

PieGuy314, 2022-01-20:
Why do I always seem to pick the wrong time to upgrade?

Used to think DietPi had some merits. Wasting time reinstalling and reconfiguring the system isn't high on my list of priorities. 

MIght I also suggest that encouraging people to do a system upgrade via a login banner is somewhat ill-advised given that said upgrades do not appear to have been peer reviewed/validated.

TL;DR I'm getting too old for this shit.

jnk19, 2022-01-20:
I tried to compile a custom kernel v7l and v8 on original Raspbian lite image the branches 5.10.y 5.15y and 5.16y which working flawless. No black screen no anything. But when tried to compile to DietPI image the black screen occurred both v7l and v8. The RPI's I have tested are: 2x RPI 4b 4gb ram 1x RPI 4b 8gb ram and one CM4 with emmc and 1gb ram.

MichaIng, 2022-01-20:
Found it. Please comment `dtoverlay=dietpi-disable_vcsm` in `/boot/config.txt`.

In case of less than 16 MiB GPU memory, the VCSM driver fails to load and throws kernel errors at boot, hence we deploy a device tree overlay to prevent it from loading in the first place. The path `/proc/device-tree/soc/vcsm` however does not exist anymore on newest kernel, and it seems that the bootloader denies to boot completely when a device tree overlay tries to alter an invalid path... This is a bummer as I explicitly asked about this and got an answer by RPi engineers that it shouldn't be able to break boot: https://forums.raspberrypi.com/viewtopic.php?t=290607

jnk19, 2022-01-20:
I confirm with commenting on `dtoverlay=dietpi-disable_vcsm` everything working normal, on DietPI image all the RPI that have take updates. Also I have compiled a new custom kernel and work flawless. 

Well done @MichaIng

Joulinar, 2022-01-20:
@PieGuy314 
At the moment it looks like am issue of the RPi kernel provided by RPi guys. On the kernel we don't have any influence. Therefore, at the moment it is not the right time to blame DietPi.

PieGuy314, 2022-01-20:
> @PieGuy314
> At the moment it looks like am issue of the RPi kernel provided by RPi guys. On the kernel we don't have any influence. Therefore, at the moment it is not the right time to blame DietPi.

Apologies for the tone of the email. I'm full of Covid and my wife needed access to one of the affected boxes. I'd rather had been resting than reconfiguring my network last night.

Joulinar, 2022-01-20:
if you are still offline, following should do a workaround as long as RPi guys looking into it.

 - Please comment `dtoverlay=dietpi-disable_vcsm` in `/boot/config.txt`.

MichaIng, 2022-01-20:
@PieGuy314 
> Apologies for the tone of the message.

No problem, I can imagine the anger when following a long prompt message which kills your system.

Usually all package upgrades shipped by the stable Debian/Raspbian and RPi repositories are well tested and reviewed and shouldn't cause any issues. Those from (stable) Debian/Raspbian do not even contain new features or major/minor software version increments but bug fixes and security patches, max important patch version raises, which is Debian stability strategy.

In this case it was a rare combination of this custom device tree overlay we apply to headless/non-GUI systems using a device tree path instead of abstracted symbol, and the unintended behaviour that the RPi adds it to the device tree even when the path is invalid (does not exist). Most RPi users (and RPi OS of course) do not use custom device tree overlays, and those which do mostly use symbols instead of paths, and of course that a specific path in the device tree changes is again a very rare case. All together loading to this issue, so it was hard to catch this, respectively the chance for this always existed was didn't happen or was not recognised as reason for a non-booting system until now.

Reminds me of the 100 characters issue with `config.txt`, which was as well extremely difficult to accidentally detect 😄.

What I take from this is that we should do `rpi-update` (in-development kernel/firmware) testing on a regular schedule to catch potential issues with common DietPi setups early, and non-DietPi specific issues to help with RPi kernel/firmware development anyway.

PieGuy314, 2022-01-20:
@MichaIng

Read a little about device tree overlays in an attempt to better educate myself about the specific issue. All I can say is thank you for identifying the problem and providing a work around in such a timely manner.

MichaIng, 2022-01-20:
Indeed, device trees and overlays are a great addition to the Linux kernel for SBCs and other embedded devices where hardware features are fixed/known, to toggle or configure these features outside of the kernel.

I can only forward the thanks to the RPi engineers who reverted the repository quickly, so that only a small number of our users where affected, and now providing a new set of kernel/bootloader packages which handles invalid device tree paths in overlays gracefully. On my end, critical enough to do a night session for narrowing the issue down to the invalid device tree overlay 😴.

I wonder whether we can use a dedicate a Raspberry Pi as GitHub actions runner to do `rpi-update` and basic functionality tests with DietPi on a Cron job like scheduled manner 🤔. Will investigate how this works, and the RPi could then also do compiling tasks for RPi ARMv6 builds.

Joulinar, 2022-01-20:
much appreciate the work you did guys. 👍

Joulinar, 2022-01-20:
new kernel + bootloader has been released by RPi devs

```
raspberrypi-bootloader/stable 1:1.20220120-1 armhf [upgradable from: 1:1.20211118-3]
raspberrypi-kernel/stable 1:1.20220120-1 armhf [upgradable from: 1:1.20211118-3]
```

MichaIng, 2022-01-20:
Marking this hence as closed.

closed 2022-01-20 by MichaIng