MichaIng/DietPi #5188: Reboot/Restart results in nothing

trueaspects, 2022-01-20:
### Creating a bug report/issue

#### Required Information
- G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Bullseye 0
- Linux DietPi 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux
- RPi 4 Model B (aarch64)
- Default RPI Power
- Samsung 32gb

I freshly installed at least 4 times and all resulted in same issue being unable to reboot/restart. 

#### Steps to reproduce

1. Download latest RPI ARMv8 x64 ISO (v7.9.3)
2. Install and run all updates (v8.0.2)
3. Reboot
4. Doesn't come online or connect to wifi

#### Expected behaviour

- Reboots normally

#### Actual behaviour

- Doesn't boot/start up

#### Extra details

- I am using the RPI headless and do not have the mini-hdmi to hdmi to connect it to a screen. I am not sure what is happening, but a reboot/restart does not work. 
- I have a feeling it has to do with debian itself rather than DietPi.

ravenclaw900, 2022-01-20:
Hi,
This is probably related to https://github.com/MichaIng/DietPi/issues/5187. The troublesome packages should be removed now, so installing should work.

trueaspects, 2022-01-20:
All 4 times with a fresh install was done today. When was it fixed? I am afraid to do a reboot/restart. Would apt update and full-upgrade command fix the issue?

Joulinar, 2022-01-20:
The issue was a kernel/bootloader from RPi side. An apt upgrade would not change anything. The kernel has been removed by RPi guys around 1am CET.

MichaIng, 2022-01-20:
I think the affected kernel packages were online for ~1.5h. However, to be sure, remove the `dtoverlay=dietpi-disable_vcsm` line in `/boot/config.txt`, which is not required anymore with newest kernel anyway.

TimBullas, 2022-01-21:
I'm having the same issue on RPi 3.
I'm using webmin interface and had the notification that updates were needed. Clicked update and it updated, then it prompted to install updates, so I did, but it failed when it tried to update webmin. Tried to navigate to a different page and it failed to load. Tried to SSH in and it also failed. 
Check the pi and it had solid red leds. Rebooted it, and it doesn't boot.

Files on sd are intact, but I don't have the: "dtoverlay=dietpi-disable_vcsm" line in /boot/config.txt to comment out, so I don't know what to do from here.

Joulinar, 2022-01-21:
Probably this is a different issue as the kernel package has been fixed and a new version was released already. Do you have a screen that could be attached temporary? Just to check if there is any output? Yes you have a screen, remove `quite` as well from `/boot/cmdline.txt`

MichaIng, 2022-01-21:
> but I don't have the: "dtoverlay=dietpi-disable_vcsm" line in /boot/config.txt to comment out, so I don't know what to do from here.

Please remove that line.

TimBullas, 2022-01-22:
Removed `quiet` from /boot/cmdline.txt and `dtoverlay=dietpi-disable_vcsm` from /boot/config.txt no HDMI output solid red leds on the pi.

MichaIng, 2022-01-22:
Sold red light is normal/power light by default, right? (I have it configured to indicate CPU activity)
Can you ping it?

TimBullas, 2022-01-22:
Tried to ping it "Destination host unreachable"

MichaIng, 2022-01-22:
Can you please paste the full `config.txt` and `cmdline.txt`? We are not aware of any other issue and the one with the overlay has been solved already. Probably the SD card is damaged, i.e. did you boot another OS successfully from it before?

I suggest it one more time flashing the image freshly, if it does not work, test another SD card or better a USB flash drive.

TimBullas, 2022-01-22:
cmdline.txt:

`root=PARTUUID=8b0ac6f5-02 rootfstype=ext4 rootwait fsck.repair=yes net.ifnames=0 logo.nologo console=tty1`

config.txt:

```# Docs: https://github.com/raspberrypi/documentation/blob/master/configuration/config-txt/README.md
# Overlays: https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README

#-------Display---------
# Max allocated framebuffers: Set to "0" in headless mode to reduce memory usage
# - Defaults to "2" on RPi4 and "1" on earlier RPi models
#max_framebuffers=0

# If you get no picture, set the following to "1" to apply most compatible HDMI settings.
hdmi_safe=1

# Uncomment to adjust the HDMI signal strength if you have interferences, blanking, or no display.
# - Values from "0" to "11" are allowed, use values above "7" only if required, e.g. with very long HDMI cable.
# - Default on first RPi1 A/B is "2", else "5", on RPi4 this setting is ignored.
#config_hdmi_boost=5

# Uncomment if HDMI display is not detected and composite is being outputted.
hdmi_force_hotplug=1

# Uncomment to disable HDMI even if plugged, e.g. to force composite output.
#hdmi_ignore_hotplug=1

# Uncomment to force a console size. By default it will be display's size minus overscan.
#framebuffer_width=1280
#framebuffer_height=720

# Uncomment to enable SDTV/composite output on RPi4. This has no effect on previous RPi models.
#enable_tvout=1
# SDTV mode
#sdtv_mode=0

# Uncomment to force a specific HDMI mode (this will force VGA).
#hdmi_group=1
#hdmi_mode=1

# Uncomment to force an HDMI mode rather than DVI. This can make audio work in DMT (computer monitor) modes.
#hdmi_drive=2

# Set "hdmi_blanking=1" to allow the display going into standby after 10 minutes without input.
# With default value "0", the display shows a blank screen instead, but will not go into standby.
# NB: With "1" some applications (e.g. Kodi, OMXPlayer) cannot prevent display standby due to missing DPMS signal.
#hdmi_blanking=1

# Set to "1" if your display has a black border of unused pixels visible.
disable_overscan=1

# Uncomment the following to adjust overscan. Use positive numbers if console goes off screen, and negative if there is too much border.
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# Rotation
#display_hdmi_rotate=0
#lcd_rotate=0

#-------RPi camera module-------
#start_x=1
disable_camera_led=1

#-------GPU memory splits-------
gpu_mem_256=192
gpu_mem_512=256
gpu_mem_1024=256

#-------Boot splash screen------
#disable_splash=1

#-------Onboard sound-----------
dtparam=audio=on

#-------I2C-------------
#dtparam=i2c_arm=off
#dtparam=i2c_arm_baudrate=100000

#-------SPI-------------
#dtparam=spi=off

#-------Serial/UART-----
# NB: Enabled for 1st run only. If you want to keep this setting, please set CONFIG_SERIAL_CONSOLE_ENABLE=1 in dietpi.txt before 1st boot.
# NB: "enable_uart=1" will forcefully set "core_freq=250" on WiFi/BT-capable RPi models, unless "force_turbo=1" is set as well.
enable_uart=0

#-------Overclock-------
temp_limit=75
initial_turbo=20

#over_voltage=0
#arm_freq=1400
#core_freq=400
#sdram_freq=500

#over_voltage_min=0
#arm_freq_min=300
#core_freq_min=250
#sdram_freq_min=400
arm_64bit=1
dtoverlay=disable-wifi
dtparam=sd_poll_once
```

The sd card was working fine, it might have been corrupted during the update.
I'm sure if I flashed it with a fresh image it would work again.
I haven't had a chance to do a backup, so I was hoping to try and recover the image on the card so I don't have to go through setting everything up again.

If there's nothing I can do to recover it, then I'll just flash it with a fresh image.

Joulinar, 2022-01-22:
what software you where running on that system? And do you have a change to place the SD card into another Linux box? or did you have a spare SD card? Just to be able to get access to the CD cad and to be able to perform some file system checks.

TimBullas, 2022-01-22:
I think I really only had Webmin, Samba & Radicale installed on it. Anything else was whatever is installed as part of DietPi.

I have another linux box I can use to access the sd card, but I don't know what I need to do once I have access.

MichaIng, 2022-01-22:
Attach the SD card to the other Linux box, check its device path via `lsblk -p`, something like `/dev/mmcblk0p2` is the ext4 partition. Run fsck on it, and on the FAT partition as well to be sure:
```sh
fsck /dev/mmcblk0p2
fsck /dev/mmcblk0p1
```

Joulinar, 2022-01-22:
If you use a SD card USB adaptor, the device should be `/dev/sdx`

MichaIng, 2022-01-22:
Right, only native SD card readers show `/dev/mmcblkN`/`/dev/mmcblkNpM` for attached SD cards/their partitions.

TimBullas, 2022-01-26:
Sorry I've taken so long to get back to you, has been a busy couple of days.

I ran `lsblk -p` :
```
/dev/sdb      8:16   1 14.8G  0 disk 
├─/dev/sdb1   8:17   1  128M  0 part 
└─/dev/sdb2   8:18   1 14.7G  0 part 
```

Here's the result of fsck on  /dev/sdb1
```
sudo fsck /dev/sdb1
fsck from util-linux 2.37.2
fsck.fat 4.2 (2021-01-31)
There are differences between boot sector and its backup.
This is mostly harmless. Differences: (offset:original/backup)
  65:01/00
1) Copy original to backup
2) Copy backup to original
3) No action
[123?q]? 3
Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
1) Remove dirty bit
2) No action
[12?q]? 2
/dev/sdb1: 66 files, 3582/258077 clusters
```
I've never used fsck so I just selected "No action" so I could post the results.

And here's the result of fsck on  /dev/sdb2:

```
sudo fsck /dev/sdb2                                                                                                                         130 ⨯
fsck from util-linux 2.37.2
e2fsck 1.46.4 (18-Aug-2021)
/dev/sdb2: clean, 100518/962880 files, 933650/3855744 blocks
```

MichaIng, 2022-01-26:
Since the RPi does not use a bootloader from the boot sector, you can safely solve the difference boot original and backup boot sectors in any way to mute fsck on `/dev/sdb1` FAT filesystem. Also remove the dirty bit, else a full fsck on that filesystem will run on every boot (though fast on FAT).

For the root partition `/dev/sdb2`, it has no dirty bit set, hence the fsck is basically skipped. You can force a full scan via:
```sh
fsck -f /dev/sdb2
```

TimBullas, 2022-02-01:
Results from: `fsck -f /dev/sdb2`
```
fsck from util-linux 2.37.2
e2fsck 1.46.4 (18-Aug-2021)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
/dev/sdb2: 100518/962880 files (0.3% non-contiguous), 933650/3855744 blocks
```

MichaIng, 2022-02-01:
Okay the filesystem is all clean. Two things to try:

GPU memory above the default is actually not required anymore, so you can safely revert this to:
```
gpu_mem_256=76
gpu_mem_512=76
gpu_mem_1024=76
```
At least I remember once for some reason a raised GPU memory broke boot, while it of course shouldn't cause it.

Another thing, try to commend the `dtoverlay=disable-wifi` setting:
```
#dtoverlay=disable-wifi
```
While debugging #5187 I thought that I fixed boot of my RPi Zero W by commenting not this but another custom overlay to disable onboard 3.5mm audio. Finally it was another one, but an idea I had in between was that something may have changed so that old device tree overlays may need to be recompiled. Worth to give it a try.

TimBullas, 2022-02-04:
Tried those suggestions, and still nothing.
Looks like I'm going to have to re-flash the sd card. 
Thanks so much for taking the time to help, much appreciated. :)

MichaIng, 2022-02-04:
🙁. As a list debug step, could you post the content of `/boot`, respectively the FAT partition of the SD card? Probably for some reason a boot relevant file is missing.

TimBullas, 2022-02-05:
Contents of `/boot`
```
└─$ ls -a -1
.
..
cmdline.txt
config.txt
dietpi
dietpi-CHANGELOG.txt
dietpi-LICENSE.txt
dietpi-README.md
dietpi.txt
dietpi-wifi.txt
overlays
ssh
'System Volume Information'

```

Joulinar, 2022-02-05:
This is all? The entire kernel is missing.

TimBullas, 2022-02-05:
I'm really not sure what version it was. I can hunt around and see if I still have the ISO I downloaded, would we be able to work it out from that?

Joulinar, 2022-02-05:
If you have time, we should wait on @MichaIng . Usually he has good ideas on how to find out and which files to put back in place.

TimBullas, 2022-02-05:
Yeah, that's cool. I'm happy to wait and learn some stuff too :)

MichaIng, 2022-02-05:
Okay, taking the kernel version you posted at first, download these two files:
- https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20211118-3_arm64.deb
- https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-bootloader_1.20211118-3_arm64.deb

Copy the content of the `boot` directory of each of those to packages (they can be extracted with e.g. [7-Zip]https://www.7-zip.org/) into the boot partition of your SD card. With this it should work. The first thing to do when it does it to upgrade/reinstall those two packages:
```sh
apt update
apt install --reinstall raspberrypi-kernel raspberrypi-bootloader
```

How ever this may have happened 😄.

Joulinar, 2022-02-13:
In this case I hope it will be not that difficult to start from scratch. I'm going to close this issue. Feel free to reopen if needed.

closed 2022-02-13 by Joulinar