MichaIng/DietPi #5189: Upgrade from 7.8.2 to 8.0.2

g7kse, 2022-01-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | 7.8.2
- Distro version | Bullseye
- Kernel version | Linux DietPi 5.10.63-v8+ 1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux
- SBC model | RPi 4
- Power supply used | Official RPi4 power supply
- SDcard used | SanDisk Ultra

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
 
- Bug report ID | d276ee80-6efc-42d4-9c37-e288c5a9291b

#### Steps to reproduce

1. ... run dietpi-upgrade or
2. ...apt-update and apt-upgrade

#### Expected behaviour
- Normal upgrade process

#### Actual behaviour
- exit with error apt-get: error while loading shared libraries: /lib/aarch64-linux-gnu/libapt-private.so.0.0: invalid ELF header

#### Extra details
- webserver looks like it is running but all apps unreachable (503 error). I might try it later but this looks unusual to me. Couldn't see a similar issue raised elsewhere 
[image]

MichaIng, 2022-01-20:
Please try:
```sh
apt install --reinstall apt
```

MichaIng, 2022-01-20:
If it doesn't work as the same underlying library is broken:
```sh
cd /tmp
curl -LO https://deb.debian.org/debian/pool/main/a/apt/apt_2.2.4_arm64.deb
dpkg -i apt_2.2.4_arm64.deb
rm apt_2.2.4_arm64.deb
apt-get update
```

g7kse, 2022-01-20:
Upgrade worked with the second suggestion. Software (Nextcloud and docker) not starting.Nextcloud doesn't show up as a service.Any other thoughts?

MichaIng, 2022-01-20:
Which webserver are you using?
```sh
journalctl -u php7.4-fpm -u lighttpd -u docker -u containerd
```

Joulinar, 2022-01-20:
can you show as well

`dietpi-services status`

g7kse, 2022-01-20:
First one is
`-- Journal begins at Thu 2022-01-20 21:18:55 GMT, ends at Thu 2022-01-20 22:20:01 GMT. --
Jan 20 21:19:24 DietPi systemd[1]: Starting Lighttpd Daemon...
Jan 20 21:19:24 DietPi systemd[1]: Started Lighttpd Daemon.
`

second is 
`[  OK  ] DietPi-Services | nmbd			 active (running) since Thu 2022-01-20 21:19:11 GMT; 1h 3min ago
[  OK  ] DietPi-Services | smbd			 active (running) since Thu 2022-01-20 21:19:11 GMT; 1h 3min ago
[  OK  ] DietPi-Services | nfs-kernel-server	 active (exited) since Thu 2022-01-20 21:19:14 GMT; 1h 3min ago
[  OK  ] DietPi-Services | mariadb		 active (running) since Thu 2022-01-20 21:19:24 GMT; 1h 3min ago
[  OK  ] DietPi-Services | lighttpd		 active (running) since Thu 2022-01-20 21:19:24 GMT; 1h 3min ago
[ INFO ] DietPi-Services | docker		 inactive (dead)
[  OK  ] DietPi-Services | cron			 active (running) since Thu 2022-01-20 21:19:24 GMT; 1h 3min ago
[  OK  ] DietPi-Services | fahclient		 active (running) since Thu 2022-01-20 21:19:25 GMT; 1h 3min ago
[  OK  ] DietPi-Services | dropbear		 active (running) since Thu 2022-01-20 21:18:59 GMT; 1h 3min ago
[ INFO ] DietPi-Services | dietpi-vpn		 inactive (dead)
[ INFO ] DietPi-Services | dietpi-cloudshell	 inactive (dead)
[  OK  ] DietPi-Services | dietpi-dashboard	 active (running) since Thu 2022-01-20 21:19:10 GMT; 1h 3min ago
[  OK  ] DietPi-Services | dietpi-ramlog	 active (exited) since Thu 2022-01-20 21:18:59 GMT; 1h 3min ago
[  OK  ] DietPi-Services | dietpi-preboot	 active (exited) since Thu 2022-01-20 21:18:59 GMT; 1h 3min ago
[  OK  ] DietPi-Services | dietpi-postboot	 active (exited) since Thu 2022-01-20 21:19:10 GMT; 1h 3min ago
[ INFO ] DietPi-Services | dietpi-wifi-monitor	 inactive (dead)`

Mot easy to read like that but Docker, VPN, Cloudshell and wifi monitor are showing as dead

Joulinar, 2022-01-20:
what about `dietpi-services start docker`?

g7kse, 2022-01-21:
Yeah, wasn't bothered about the VPN, Cloudshell and wifi monitor. Outputs are a bit confusing though

 ````
Failed to start docker.service: Unit containerd.service not found.

docker.service - Docker Application Container Engine
     Loaded: loaded (/lib/systemd/system/docker.service; disabled; vendor preset: enabled)
    Drop-In: /usr/lib/systemd/system/docker.service.d
             └─dietpi-simple.conf
     Active: inactive (dead)
TriggeredBy: ● docker.socket
       Docs: https://docs.docker.com
```
Biggest issue really is Nextcloud

Joulinar, 2022-01-21:
Did you installed NextCloud via DietPi software as native app or inside Docker? Could you share following

```
dietpi-software list | grep " =2 "
```

What about setting maintenance mode on NextCloud?

```
sudo -u www-data php /var/www/nextcloud/occ maintenance:mode --on
sudo -u www-data php /var/www/nextcloud/occ maintenance:mode --off
```

g7kse, 2022-01-21:
All installed via native dietpi software. I find this is the best way really. Only use docker occasionally for some unusual stuff. Output is
```
ID 1 | =2 | Samba Client: access SMB/CIFS/Samba network shares | |  dietpi-drive_manager > Add network drive
ID 2 | =2 | Folding@Home: distributed disease research project | | https://dietpi.com/docs/software/distributed_projects/#foldinghome
ID 55 | =2 | Wordpress: website blog and publishing platform | +MariaDB +PHP +webserver | https://dietpi.com/docs/software/social/#wordpress
ID 82 | =2 | LLMP: Lighttpd | MariaDB | PHP | +Lighttpd +MariaDB +PHP | https://dietpi.com/docs/software/webserver_stack/#llmp-web-stack
ID 84 | =2 | Lighttpd: Extremely lightweight webserver | +PHP | https://dietpi.com/docs/software/webserver_stack/#lighttpd
ID 88 | =2 | MariaDB: Persistent cached file-per-table database server | | https://dietpi.com/docs/software/databases/#mariadb
ID 89 | =2 | PHP: Hypertext Preprocessor for dynamic web content | | https://dietpi.com/docs/software/webserver_stack/#php
ID 91 | =2 | Redis: Volatile in-memory non-SQL database server | | https://dietpi.com/docs/software/databases/#redis
ID 96 | =2 | Samba Server: Feature-rich SMB/CIFS server | | https://dietpi.com/docs/software/file_servers/#samba
ID 103 | =2 | DietPi-RAMlog: Makes /var/log a RAM disk, preserves file structure on reboot | | https://dietpi.com/docs/software/log_system/#dietpi-ramlog
ID 104 | =2 | Dropbear: Lightweight SSH server | | https://dietpi.com/docs/software/ssh/#dropbear
ID 109 | =2 | NFS Server: Network File System server | | https://dietpi.com/docs/software/file_servers/#nfs
ID 110 | =2 | NFS Client: network file system client | |  dietpi-drive_manager > Add network drive
ID 114 | =2 | Nextcloud: File sync, sharing and collaboration platform | +MariaDB +PHP +Redis +webserver | https://dietpi.com/docs/software/cloud/#nextcloud
ID 130 | =2 | Python 3: Runtime system, pip package installer and development headers | | https://dietpi.com/docs/software/programming/#python-3
ID 134 | =2 | Docker Compose: Manage multi-container Docker applications | +Python 3 +Docker | https://dietpi.com/docs/software/programming/#docker-compose
ID 162 | =2 | Docker: Build, ship, and run distributed applications | | https://dietpi.com/docs/software/programming/#docker
ID 185 | =2 | Portainer: Simplifies container management in Docker (standalone host) | +Docker | https://dietpi.com/docs/software/programming/#portainer
ID 198 | =2 | File Browser: web based file manager | | https://dietpi.com/docs/software/cloud/#filebrowser
ID 200 | =2 | DietPi-Dashboard (beta): Official lightweight DietPi web interface (Rust) | | https://dietpi.com/docs/software/system_stats/#dietpi-dashboard
```
we could be getting somewhere here. When running `sudo -u www-data php /var/www/nextcloud/occ maintenance:mode --on` I get `sudo: php: command not found`. Does that mean that php is unhappy? something got broken somewhere perhaps

Joulinar, 2022-01-21:
something got corrupted on your system. You are missing a couple of software title (like PHP and Redis), even if they should be installed.

Can you perform a file system check

```
> /forcefsck
reboot
# then after reboot
journalctl -t systemd-fsck
```

g7kse, 2022-01-21:
Yep, sounds like it all had a bad day somewhere. Would be nice to recover it without having to re-install

```
Jul 13 18:29:24 DietPi systemd-fsck[134]: Please pass 'fsck.mode=force' on the kernel command line rather than creating /forcefsck on the root file system.
Jan 21 13:23:38 DietPi systemd-fsck[151]: e2fsck 1.46.2 (28-Feb-2021)
Jan 21 13:23:38 DietPi systemd-fsck[151]: Pass 1: Checking inodes, blocks, and sizes
Jan 21 13:23:38 DietPi systemd-fsck[151]: Pass 2: Checking directory structure
Jan 21 13:23:39 DietPi systemd-fsck[151]: Pass 3: Checking directory connectivity
Jan 21 13:23:39 DietPi systemd-fsck[151]: Pass 4: Checking reference counts
Jan 21 13:23:40 DietPi systemd-fsck[151]: Pass 5: Checking group summary information
Jan 21 13:23:40 DietPi systemd-fsck[151]: /dev/mmcblk0p2: 64211/1884960 files (0.1% non-contiguous), 674393/7560192 blocks
Jan 21 13:23:41 DietPi systemd-fsck[237]: Please pass 'fsck.mode=force' on the kernel command line rather than creating /forcefsck on the root file system.
Jan 21 13:23:41 DietPi systemd-fsck[270]: fsck.fat 4.2 (2021-01-31)
Jan 21 13:23:41 DietPi systemd-fsck[270]: /dev/mmcblk0p1: 346 files, 65326/258077 clusters
```

MichaIng, 2022-01-21:
Please try to reinstall Nextcloud, since PHP packages for some reason are not installed:
```sh
dietpi-software install 114
```

g7kse, 2022-01-21:
It gave me
```
[ INFO ] DietPi-Software | 114: Nextcloud is already installed
[ INFO ] DietPi-Software | Use "dietpi-software reinstall 114" to force rerun of installation and configuration steps for Nextcloud.
[  OK  ] DietPi-Software | No changes applied for: Nextcloud
```
when I `dietpi-software reinstall 114` we seem to get a bit closer. The installation fails with `Failed to stop redis-server.service: Unit redis-server.service not loaded.` 
I assume then that redis is at the bottom of this pile (so far :-) )

MichaIng, 2022-01-21:
Ah yes sorry, `reinstall` of course.

> Failed to stop redis-server.service: Unit redis-server.service not loaded.

It should have been installed just before with the `redis-server` package. What do the "APT install for: " lines above that one show?

Probably the service file itself got somehow corrupted or removed while the package is still installed please try to reinstall it manually:
```sh
apt install --reinstall redis-server
```

In case the same is true for PHP, can you show:
```sh
dpkg -l | grep php
```

MichaIng, 2022-01-21:
Ah, I should have had a look into your bug report. Okay it seems that various files from already installed packages somehow got removed, like the `php` command, the `php7.4-fpm.service` and the `redis-server.service`. So indeed those need to be reinstalled, hopefully we catch everything, no guarantee for that:
```sh
apt install --reinstall $(dpkg --get-selections 'php*' | mawk '{print $1}')
```

g7kse, 2022-01-21:
trying to reinstall redis is proving tricky
```
redis-server.service - Advanced key-value store
     Loaded: loaded (/lib/systemd/system/redis-server.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2022-01-21 13:49:59 GMT; 19s ago
       Docs: http://redis.io/documentation,
             man:redis-server(1)
    Process: 1507 ExecStart=/usr/bin/redis-server /etc/redis/redis.conf --supervised systemd --daemonize no (code=exited, status=203/EXEC)
   Main PID: 1507 (code=exited, status=203/EXEC)
        CPU: 73ms

Jan 21 13:49:59 DietPi systemd[1]: redis-server.service: Scheduled restart job, restart counter is at 5.
Jan 21 13:49:59 DietPi systemd[1]: Stopped Advanced key-value store.
Jan 21 13:49:59 DietPi systemd[1]: redis-server.service: Start request repeated too quickly.
Jan 21 13:49:59 DietPi systemd[1]: redis-server.service: Failed with result 'exit-code'.
Jan 21 13:49:59 DietPi systemd[1]: Failed to start Advanced key-value store.
```

ls -l /lost+found

```
-rw-r--r-- 1 root root  4633 Jan  2  2009 '#13107'
-rw-r--r-- 1 root root 21529 Apr 20  2009 '#13108'
-rw-r--r-- 1 root root  9113 Apr 20  2009 '#13109'
-rw-r--r-- 1 root root  6230 Jan 10  2021 '#13110'
-rw-r--r-- 1 root root  4086 Jan 10  2021 '#13111'
lrwxrwxrwx 1 root root    14 Jan 10  2021 '#13117' -> History.600.gz
drwxr-xr-x 2 root root  4096 Nov 13 18:19 '#16324'
drwxr-xr-x 2 root root  4096 Nov 13 18:19 '#16325'
```

would it be easier trying to recover from a backup or starting afresh and recovering Nextcloud? Don't want to use up loads of your time chasing a broken sys

g7kse, 2022-01-21:
recovery is no go. It's properly broken. To close this off I'll reinstall and attempt to recovery the user directory (not sure how to do that, if there is a guide then point me to that please). 

Poor old RPi. I should have treated it better

Joulinar, 2022-01-21:
Well it is most probably the SD card or some power shortage which caused it. 

Do you have files on Nextcloud only? Or as well something inside database like calendar?

Theoretically removing Nextcloud via `dietpi-software` should create a database backup that could be used to restore on a new system together with the data directory

g7kse, 2022-01-21:
I think the SD card has been (ab)used for sometime now and has just fallen over. Power shouldn't be a problem. Just a limitation of these things.

I can recover the files, there's just a lot of them. If there was an easy way I'd do that but no big deal.

I really appreciate your time.

closed 2022-01-21 by g7kse