MichaIng/DietPi #5190: Failed to fetch http://armbian.lv.auroradev.org

sunjam, 2022-01-20:
#### Details:
- Date           | Thu Jan 20 22:12:51 GMT 2022
- DietPi version | v8.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | Odroid C2 (aarch64) (ID=12)
- Kernel version | `Linux DietPi 5.10.81-meson64 #21.08.6 SMP PREEMPT Mon Nov 22 11:21:51 UTC 2021 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `apt-get -qq --allow-change-held-packages install gcc libc6-dev make libssl-dev zlib1g-dev libbz2-dev libreadline-dev libsqlite3-dev libffi-dev`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Failed to fetch http://armbian.lv.auroradev.org/apt/pool/main/l/linux-5.15.4-meson64/linux-libc-dev_21.08.6_arm64.deb  Could not connect to armbian.lv.auroradev.org:80 (209.141.46.199), connection timed out
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

Joulinar, 2022-01-20:
Looks like a temp issue with the package source server `armbian.lv.auroradev.org`. Personally I'm not able to reach the side as well. Could you try again? Maybe you will get another package server from pool.

sunjam, 2022-01-20:
Okay, restarted the device and apt-get update process worked.  Tried installing a piece of software, Home Assistant, and see:

E: Failed to fetch                                                                                                   │ 
                        │ http://armbian.lv.auroradev.org/apt/pool/main/l/linux-5.15.4-meson64/linux-libc-dev_21.08.6_arm64.deb  Could not     │ 
                        │ connect to armbian.lv.auroradev.org:80 (209.141.46.199), connection timed out                                        │ 
                        │                                                                                                                      │ 
                        │                         Retry          : Re-run the last command that failed                                         │ 
                        │                         DietPi-Config  : Edit network, APT/NTP mirror settings etc                                   │ 
                        │                         Open subshell  : Open a subshell to investigate or solve the issue                           │ 
                        │                         Send report    : Uploads bugreport containing system info to DietPi                          │ 
                        │                                        ●─ Devs only ──────────────────────────────────────●                          │ 
                        │                         Change command : Adjust and rerun the command

Joulinar, 2022-01-20:
on the error screen, you can select change command. Try to change URL into following

```
https://apt.armbian.com/pool/main/l/linux-5.15.4-meson64/linux-libc-dev_21.08.6_arm64.deb
```

sunjam, 2022-01-20:
[FAILED] DietPi-Config | Unable to continue, DietPi-Config will now terminate.
[ INFO ] DietPi-Software | Executing alternative command: apt-get -qq --allow-change-held-packages install gcc https://apt.armbian.com/pool/main/l/linux-5.15.4-meson64/linux-libc-dev_21.08.6_arm64.deb make libssl-dev zlib1g-dev libbz2-dev libreadline-dev libsqlite3-dev libffi-dev
E: Unable to locate package https://apt.armbian.com/pool/main/l/linux-5.15.4-meson64
E: Couldn't find any package by glob 'https://apt.armbian.com/pool/main/l/linux-5.15.4-meson64'
E: Couldn't find any package by regex 'https://apt.armbian.com/pool/main/l/linux-5.15.4-meson64

Joulinar, 2022-01-20:
ah sorry my fault. The command you changed did not contain an IP address. It was just the package name. Your issue is, that your system is still trying to connect to a faulty `apt package` mirror of Armbian. Not sure if we can do anything on it, as these package server are out of our control.

@MichaIng 
any idea?

MichaIng, 2022-01-21:
Did you manually (tried to) upgrade the kernel to a newer version? I just checked the package lists and `linux-image-current-meson64` is still pulling Linux 5.10 while in your case Linux 5.15 is tried to be installed. That one would be pulled by `linux-image-edge-meson64`, which is not pre-installed on our Odroid C2 image.

MichaIng, 2022-01-21:
It`s the `armbian.lv.auroradev.org` mirror which is currently down. I cannot connect to it from any system/location. Please simply retry from the error prompt until a different mirror is chosen (which is a bit random).

MichaIng, 2022-01-21:
Okay should be solved now on the mirror director 🙂. I'll mark this as closed.

closed 2022-01-21 by MichaIng