MichaIng/DietPi #5191: Install Failed for MINEOS using dietpi-software installer

tregraha, 2022-01-21:
#### Details:
- Date           | Fri Jan 21 07:46:07 GMT 2022
- DietPi version | v8.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 3 Model B (aarch64) (ID=3)
- Kernel version | `Linux DietPi 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `./update_webui.sh`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. Running insall of MINEOS using dietpi-software installer 
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Script execution started on: Fri Jan 21 07:45:09 GMT 2022
Updating mineos-node repository...OK
Pulling master branch...From https://github.com/hexparrot/mineos-node
 * branch            master     -> FETCH_HEAD
OK
Updating npm dependencies...npm WARN old lockfile
npm WARN old lockfile The package-lock.json file was created with an old version of npm,
npm WARN old lockfile so supplemental metadata must be fetched from the registry.
npm WARN old lockfile
npm WARN old lockfile This is a one-time fix-up, please be patient...
npm WARN old lockfile
npm ERR! code 1
npm ERR! path /mnt/dietpi_userdata/mineos/minecraft/node_modules/posix
npm ERR! command failed
npm ERR! command sh -c node-gyp rebuild
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@8.4.1
npm ERR! gyp info using node@17.4.0 | linux | arm64
npm ERR! gyp info find Python using Python version 3.9.2 found at "/usr/bin/python3"
npm ERR! gyp http GET https://nodejs.org/download/release/v17.4.0/node-v17.4.0-headers.tar.gz
npm ERR! gyp http 200 https://nodejs.org/download/release/v17.4.0/node-v17.4.0-headers.tar.gz
npm ERR! gyp WARN install got an error, rolling back install
npm ERR! gyp ERR! configure error
npm ERR! gyp ERR! stack TypeError: tar.extract is not a function
npm ERR! gyp ERR! stack     at go (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/install.js:164:15)
npm ERR! gyp ERR! stack     at processTicksAndRejections (node:internal/process/task_queues:96:5)
npm ERR! gyp ERR! stack     at async install (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/install.js:62:18)
npm ERR! gyp ERR! System Linux 5.10.92-v8+
npm ERR! gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd /mnt/dietpi_userdata/mineos/minecraft/node_modules/posix
npm ERR! gyp ERR! node -v v17.4.0
npm ERR! gyp ERR! node-gyp -v v8.4.1
npm ERR! gyp ERR! not ok

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2022-01-21T07_45_14_877Z-debug-0.log
FAILED
```

Joulinar, 2022-01-21:
Probably same as this https://github.com/MichaIng/DietPi/issues/5181

Joulinar, 2022-01-26:
closed as duplicate

closed 2022-01-26 by Joulinar