MichaIng/DietPi #5195: [NOOB] Make QF9700 Driver for RPi 0 fails because missing linux-headers

VirusBurst, 2022-01-22:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.0.2
- Distro version | Bullyseye 1
- Kernel version | Linux DietPi 5.10.92+ #1514 Mon Jan 17 17:35:21 GMT 2022 armv6l GNU/Linux
- SBC model | RPi Zero (armv6l)
- Power supply used | Anything that works lul
- SDcard used | Agfa 4GB (Life on Limit!)

#### Steps to reproduce
1. Download [https://www.etherbyte.net/drivers.zip](https://www.etherbyte.net/drivers.zip)
2. Extract the "Linux" Folder from \USB.LAN,RD9700Driver\Linux\
3. "apt-get install build-essential" (just in case)
4. "make clean" and "make" to build the driver 
5. MAGIC happens, driver has been build and just put it somewhere in and register and bla bla bla

#### Actual behaviour
TL;DR 
I always get THIS
```
root@DietPi:/tmp/driver# make
Building QF9700 USB2NET chip driver...
/bin/sh: 1: cd: can't cd to /lib/modules/5.10.92+/build
make: *** [Makefile:10: all] Error 2
```

#### Extra details

Yes it works without creating driver but only in 10Base-T and with the correct driver, it can handle 100Base-T FullDuplex

Joulinar, 2022-01-22:
Did you tried to install headers package?

`apt install raspberrypi-kernel-headers`

VirusBurst, 2022-01-22:
This brought me a step further!

Now I'm stuck here D:

```
root@DietPi:/tmp/qf9700/Linux# make
Building QF9700 USB2NET chip driver...
make[1]: Entering directory '/usr/src/linux-headers-5.10.92+'
make[2]: *** No rule to make target 'arch/arm/tools/syscall.tbl', needed by 'arch/arm/include/generated/uapi/asm/unistd-common.h'.  Stop.
make[1]: *** [arch/arm/Makefile:307: archheaders] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.92+'
make: *** [Makefile:10: all] Error 2
```

My Makefile;
```
KERNEL_DIR	:= /lib/modules/$(shell uname -r)/build
#KERNEL_DIR	:= /usr/src/linux-2.6.29/linux-2.6.29.i686/
KERNEL_VERSION	:= $(shell uname -r)
PWD		:= $(shell pwd)
MODULE_NAME := qf9700
obj-m		:= $(MODULE_NAME).o

all: 
	@echo "Building QF9700 USB2NET chip driver..."
	@(cd $(KERNEL_DIR) && make -C $(KERNEL_DIR) SUBDIRS=$(PWD) CROSS_COMPILE=$(CROSS_COMPILE) modules)

clean:
	-rm -f *.o *.ko .*.cmd .*.flags .mii.mod.o.d *.mod.c Module.symvers Module.markers modules.order version.h
	-rm -rf .tmp_versions
```

MichaIng, 2022-01-22:
Please try to prepare kernel sources for building modules first:
```
cd /lib/modules/$(uname -r)/build
make modules_prepare
```

VirusBurst, 2022-01-24:
```
root@DietPi:/lib/modules/5.10.92+/build# make modules_prepare
make[1]: *** No rule to make target 'arch/arm/tools/syscall.tbl', needed by 'arch/arm/include/generated/uapi/asm/unistd-common.h'.  Stop.
make: *** [arch/arm/Makefile:307: archheaders] Error 2

```

MichaIng, 2022-01-24:
Can you check:
```sh
find /lib/modules/$(uname -r)/build -name 'Makefile' -exec grep 'M=\$(PWD)' {} +
```

VirusBurst, 2022-01-25:
```
root@DietPi:~# find /lib/modules/$(uname -r)/build -name 'Makefile*' -exec grep 'M=\$(PWD)' {} +
root@DietPi:~#
```

I tried to modify my Makefile and swapped "shell pwd" to only "PWD" just in case but this didn't changed something...
I postet a "small" debug file while Make... maybe this help :)
[error.txt](https://github.com/MichaIng/DietPi/files/7931295/error.txt)

MichaIng, 2022-02-06:
Any news on this, did you finally succeed?

MichaIng, 2022-02-20:
Marking this as closed. Feel free to reopen when required.

closed 2022-02-20 by MichaIng