MichaIng/DietPi #5196: DietPi-Software | yt-dlp

rgabbo, 2022-01-22:
#### Formal software information
- Software title | yt-dlp
- Short description | download video from youtube and other sites
- GitHub URL (if available) | https://github.com/yt-dlp/yt-dlp

#### Are there similar/alternative software titles available with DietPi-Software?

- youtube-dl

#### What makes your requested software better than the above solutions, if available?

- youtube-dl is not actively mantained and the main maintainer left the project (see https://github.com/ytdl-org/youtube-dl/issues/29965). [Yt-dlp](https://github.com/yt-dlp/yt-dlp) is at the moment the most active fork. I suggest to consider switching to this software.

MichaIng, 2022-01-22:
Many thanks for reporting. Indeed it seems like a good direct replacement for youtube-dl.

MichaIng, 2022-05-17:
Another request from the forum: https://dietpi.com/phpbb/viewtopic.php?t=10373