MichaIng/DietPi #5197: Update to v8.0.2 from  v7.9.3 Failed.

philipeastbourne, 2022-01-22:
#### Details:
- Date           | Sat Jan 22 17:48:05 GMT 2022
- DietPi version | v7.9.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | `Linux RPi-IOT-173 5.10.63-v7+ #1496 SMP Wed Dec 1 15:58:11 GMT 2021 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `apt-get -q update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ran dietpi-update as root
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- Failed with error
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 https://deb.debian.org/debian buster-backports InRelease
Hit:3 https://archive.raspberrypi.org/debian buster InRelease
Ign:4 http://download.webmin.com/download/repository sarge InRelease
Hit:5 http://download.webmin.com/download/repository sarge Release
Error: No information about packages! (Maybe no deb entries?)
Reading package lists...
E: Problem executing scripts APT::Update::Post-Invoke-Success 'test -x /usr/bin/apt-show-versions || exit 0 ; apt-show-versions -i'
E: Sub-process returned an error code
```

MichaIng, 2022-01-22:
Many thanks for your report.

I guess you didn't install `apt-show-versions` manually but it was pulled by Webmin a longer time ago? It was a dependency of Webmin once, but not anymore, and is basically a pretty obsolete and broken package which can be replaced by native `dpkg` and `apt` calls. Hence I suggest to solve via:
```sh
apt purge --autoremove apt-show-versions
dietpi-update 1
```

philipeastbourne, 2022-01-23:
Thank you for your quick response. Now have another problem which is preventing the update in the pycompile is missing.
I tried to remove python2.7, and I think I have broken it.
Thanks again.

MichaIng, 2022-01-23:
For Python 3 the command is `py3compile`, so if anything tries to use `pycompile`, that fails without Python 2 indeed. Find out what is trying to use it, and you can restore Python 2 via:
```sh
apt install python
```
If the package is installed already, reinstall the essential files via:
```sh
apt install --reinstall python2-minimal python2.7-minimal libpython2.7-minimal
```

philipeastbourne, 2022-01-23:
Great, thank you. All sorted now, and working.
Thanks again.

closed 2022-01-23 by philipeastbourne