MichaIng/DietPi #5198: RockPi S - Can't start any docker container

4nner, 2022-01-23:
#### Required Information
- DietPi version | v8.0.2
- Distro version | bullseye
- Kernel version | Linux DietPi 4.4.143-65-rockchip-g58431d38f8f3 #1 SMP PREEMPT Sat Aug 14 09:31:07 UTC 2021 aarch64 GNU/Linux
- SBC model | ROCK Pi S (aarch64)
- Power supply used | 5V 1A
- SDcard used | Class 10 SD Card

#### Additional Information
- Software title | Docker
- Was the software title installed freshly or updated/migrated? Fresh install
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce
1. Install DietPi on a RockPi S
2. Firmware update through dietpi-config 
3. Install Docker & Docker-Compose through dietpi-software
4. Try to run any container (even the simple hello-world) will fail
5. This error will be displayed: `docker: Error response from daemon: OCI runtime create failed: cgroup namespaces aren't enabled in the kernel: unknown.`

#### Expected behaviour
Docker container should start successfully

#### Actual behaviour
This error is displayed, the docker fails to start.
`docker: Error response from daemon: OCI runtime create failed: cgroup namespaces aren't enabled in the kernel: unknown.`

#### Extra details
I have tried modifying the boot parameters as stated in #4980 as the issues appears to be similar, but without any success.

Output of `systemctl status docker`:
`root@DietPi:~# systemctl status docker
● docker.service - Docker Application Container Engine
     Loaded: loaded (/lib/systemd/system/docker.service; disabled; vendor preset: enabled)
    Drop-In: /usr/lib/systemd/system/docker.service.d
             └─dietpi-simple.conf
     Active: active (running) since Sun 2022-01-23 13:07:38 GMT; 1min 21s ago
TriggeredBy: ● docker.socket
       Docs: https://docs.docker.com
   Main PID: 1089 (dockerd)
      Tasks: 10
     Memory: 39.3M
     CGroup: /system.slice/docker.service
             └─1089 /usr/bin/dockerd -H fd:// --containerd=/run/containerd/containerd.sock

Jan 23 13:07:38 DietPi systemd[1]: Started Docker Application Container Engine.
Jan 23 13:07:39 DietPi dockerd[1089]: time="2022-01-23T13:07:39.861108922Z" level=warning msg="Unable to find cpu controller"
Jan 23 13:07:39 DietPi dockerd[1089]: time="2022-01-23T13:07:39.862767047Z" level=warning msg="Unable to find cpuset controller"
Jan 23 13:08:38 DietPi dockerd[1089]: time="2022-01-23T13:08:38.958923527Z" level=warning msg="Your kernel does not support cgroup namespaces.  Cgroup namespace setting discarded."
Jan 23 13:08:42 DietPi dockerd[1089]: time="2022-01-23T13:08:42.281869702Z" level=warning msg="seccomp is not enabled in your kernel, running container without default profile"
Jan 23 13:08:45 DietPi dockerd[1089]: time="2022-01-23T13:08:45.942365906Z" level=error msg="stream copy error: reading from a closed fifo"
Jan 23 13:08:45 DietPi dockerd[1089]: time="2022-01-23T13:08:45.947213973Z" level=error msg="stream copy error: reading from a closed fifo"
Jan 23 13:08:49 DietPi dockerd[1089]: time="2022-01-23T13:08:49.270579603Z" level=error msg="b022faacde13b101f84b1f9a4d54a492ba705666c6160f03eab94d8633f4f5a7 cleanup: failed to delete container from containerd: no such container"
Jan 23 13:08:49 DietPi dockerd[1089]: time="2022-01-23T13:08:49.270866602Z" level=error msg="Handler for POST /v1.41/containers/b022faacde13b101f84b1f9a4d54a492ba705666c6160f03eab94d8633f4f5a7/start returned error: OCI runtime create failed: cgroup namespaces aren't enabled in the kernel: unknown"`

MichaIng, 2022-01-23:
Many thanks for your report.

Can you show the output of:
```sh
grep 'unified_cgroup_hierarchy' /boot/boot.*
ls -l /sys/fs/cgroup
```

If I'm not mistaking this cgroup namespacing should be required only with cgroupv2 while our ROCK Pi S image should ship with cgroupv1 due to the limitations of the old v4.4 kernel.

4nner, 2022-01-23:
Here is the output you requested for:
```
root@DietPi:~# grep 'unified_cgroup_hierarchy' /boot/boot.*
root@DietPi:~# ls -l /sys/fs/cgroup
total 0
-r--r--r--  1 root root 0 Jan 23 15:31 cgroup.controllers
-rw-r--r--  1 root root 0 Jan 23 15:31 cgroup.procs
-rw-r--r--  1 root root 0 Jan 23 13:07 cgroup.subtree_control
drwxr-xr-x  2 root root 0 Jul 13  2021 dev-hugepages.mount
drwxr-xr-x  2 root root 0 Jul 13  2021 dev-mqueue.mount
drwxr-xr-x  2 root root 0 Jan 23 15:31 init.scope
drwxr-xr-x  2 root root 0 Jan 23 13:07 sys-fs-fuse-connections.mount
drwxr-xr-x  2 root root 0 Jan 23 13:07 sys-kernel-config.mount
drwxr-xr-x  2 root root 0 Jul 13  2021 sys-kernel-debug.mount
drwxr-xr-x 20 root root 0 Jan 23 13:22 system.slice
```

MichaIng, 2022-01-23:
Looks like you use an older image? The current image ships with force cgroupv1 while yours uses v2, which is not sufficiently supported by that kernel. Actually it should have been applied by the update to DietPi v7.7 🤔. Can you please show the output of this command:
```sh
(( $G_DISTRO > 5 && ( $(uname -r | mawk -F'.' '{print $1}') < 4 || ( $(uname -r | mawk -F'.' '{print $1}') == 4 && $(uname -r | mawk -F'.' '{print $2}') < 15 ) ) )) && echo 'Linux <4.15 detected'
```

You can apply the patch manually:
```sh
grep -q 'systemd.unified_cgroup_hierarchy=0' /boot/boot.cmd || G_EXEC sed -i '/^setenv bootargs "/s/"$/ systemd.unified_cgroup_hierarchy=0"/' /boot/boot.cmd
apt install u-boot-tools
mkimage -C none -A arm64 -T script -d /boot/boot.cmd /boot/boot.scr
```

4nner, 2022-01-23:
Here is the output requested:
```
root@DietPi:~# (( $G_DISTRO > 5 && ( $(uname -r | mawk -F'.' '{print $1}') < 4 || ( $(uname -r | mawk -F'.' '{print $1}') == 4 && $(uname -r | mawk -F'.' '{print $2}') < 15 ) ) )) && echo 'Linux <4.15 detected'
Linux <4.15 detected
```

The given patch works, and I am now able to successfully run docker containers.
However, I am unable to get a shell from a started container:
`docker: Error response from daemon: OCI runtime create failed: container_linux.go:380: starting container process caused: process_linux.go:545: container init caused: open /dev/ptmx: no such file or directory: unknown.`
`docker run -it alpine` or `docker exec -it bash` they both return the above error.

MichaIng, 2022-01-23:
Can you try:
```sh
docker exec -i bash
```
Looks like not all terminal devices are available within the container or it doesn't have permissions by default to access them. For reference: https://stackoverflow.com/questions/53478453

4nner, 2022-01-23:
The solution works.
Thank you for the support!

closed 2022-01-23 by 4nner