MichaIng/DietPi #5201: Missing boot files when generating image

nickshanks347, 2022-01-23:
This was tried on Virtualbox, using the latest Debian (bullseye) mini.iso.

Installed Debian based on the image generation instructions on the wiki, and then ran the DietPi script afterwards... seems like after the script is done, and you reboot, you get thrown into `grub>` because of missing files in /boot. I had to manually do `apt install linux-image-amd64`, that seemed to fix it.

Just thought I'd point it out, and, happy to demo if needed.

MichaIng, 2022-01-23:
`linux-image-amd64` is automatically installed on all systems detected as `x86_64`: https://github.com/MichaIng/DietPi/blob/7cedb8e/PREP_SYSTEM_FOR_DIETPI.sh#L746

Even when somehow the detection goes wrong, it should keep all `linux-image-*` packages installed by the Debian installer itself. So not sure how it can happen.

Do you have the whole DietPi-PREP log, so we can see what actually happened?

nickshanks347, 2022-01-23:
Sure, if this is what you mean:

```
root@DietPi:~# bash -c "$(curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/master/PREP_SYSTEM_FOR_DIETPI.sh)"
Get:1 http://ftp.uk.debian.org/debian bullseye InRelease [116 kB]
Get:2 http://ftp.uk.debian.org/debian bullseye-updates InRelease [39.4 kB]
Get:3 http://ftp.uk.debian.org/debian bullseye/main amd64 Packages [8183 kB]
Get:4 http://ftp.uk.debian.org/debian bullseye-updates/main amd64 Packages [2592 B]
Fetched 8341 kB in 2s (4430 kB/s)
Reading package lists... Done
[ INFO ] Selected Git branch: MichaIng/master
[ INFO ] DietPi-PREP | Detected distribution version: Bullseye (ID: 6)
[ INFO ] DietPi-PREP | Detected target CPU architecture: x86_64 (ID: 10)

 DietPi-PREP
─────────────────────────────────────────────────────
 Step: [0] Detecting existing DietPi system

[ INFO ] DietPi-PREP | No DietPi system found, skipping old instance uninstall...

 DietPi-PREP
─────────────────────────────────────────────────────
 Step: [1] Target system inputs

[ INFO ] DietPi-PREP | Entered image creator: Nick Shanks
[ INFO ] DietPi-PREP | Entered image creator: Nick Shanks
[ INFO ] DietPi-PREP | Entered pre-image info: Debian
[ INFO ] DietPi-PREP | Entered pre-image info: Debian
[ INFO ] DietPi-PREP | Selected hardware model ID: 20
[ INFO ] DietPi-PREP | Selected hardware model ID: 20
[ INFO ] DietPi-PREP | Marking WiFi as NOT required
[ INFO ] DietPi-PREP | Marking WiFi as NOT required
[ INFO ] DietPi-PREP | Selected Debian version: bullseye (ID: 6)

 DietPi-PREP
─────────────────────────────────────────────────────
 Step: [2] Downloading and installing DietPi source code

[ INFO ] DietPi-PREP | Selected Debian version: bullseye (ID: 6)

 DietPi-PREP
─────────────────────────────────────────────────────
 Step: [2] Downloading and installing DietPi source code

[  OK  ] DietPi-PREP | Checking URL: https://github.com/MichaIng/DietPi/archive/master.tar.gz
[  OK  ] DietPi-PREP | Downloading DietPi sourcecode
[  OK  ] DietPi-PREP | Extracting DietPi sourcecode
[ INFO ] DietPi-PREP | Moving kernel and boot configuration to /boot
[  OK  ] DietPi-PREP | mv DietPi-master/dietpi.txt /boot/
[  OK  ] DietPi-PREP | mv DietPi-master/README.md /boot/dietpi-README.md
[  OK  ] DietPi-PREP | mv DietPi-master/LICENSE /boot/dietpi-LICENSE.txt
[  OK  ] DietPi-PREP | Copy DietPi scripts to /boot/dietpi
[  OK  ] DietPi-PREP | Copy DietPi system files in place
[  OK  ] DietPi-PREP | Clean download location
[  OK  ] DietPi-PREP | Set execute permissions for DietPi scripts
[ INFO ] DietPi-PREP | Storing DietPi version info:
[  OK  ] DietPi-PREP | Desired setting in /boot/dietpi.txt was already set: DEV_GITBRANCH=master
[  OK  ] DietPi-PREP | Desired setting in /boot/dietpi.txt was already set: DEV_GITOWNER=MichaIng
[ INFO ] DietPi-PREP | Applying DietPi live patches to fix known bugs in this version
[  OK  ] DietPi-PREP | cp /boot/dietpi/.version /var/lib/dietpi/.dietpi_image_version
[  OK  ] DietPi-PREP | systemctl daemon-reload

 DietPi-PREP
─────────────────────────────────────────────────────
 Step: [3] APT configuration

[ INFO ] DietPi-PREP | Setting APT sources.list: bullseye 6
[  OK  ] DietPi-PREP | /boot/dietpi/func/dietpi-set_software apt-mirror default
[  OK  ] DietPi-PREP | mkdir -p /run/dietpi /var/tmp/dietpi/logs
[ INFO ] DietPi-PREP | APT update, please wait...
[... 720 lines cut ...]
removed '/etc/passwd-'
removed '/etc/shadow-'
removed '/etc/subgid-'
removed '/etc/subuid-'
removed '/var/cache/debconf/config.dat-old'
removed '/var/cache/debconf/templates.dat-old'
removed '/var/lib/dpkg/diversions-old'
removed '/var/lib/dpkg/status-old'
[ INFO ] DietPi-PREP | The used kernel version is:
        - Linux DietPi 5.10.0-10-amd64 #1 SMP Debian 5.10.84-1 (2021-12-08) x86_64 GNU/Linux
[ INFO ] DietPi-PREP | The following kernel images and modules have been found:
ls: cannot access '/lib/modules': No such file or directory
/boot:
total 60K
drwxr-xr-x 4 root root 4.0K Jan 23 20:07 dietpi
-rw-rw-r-- 1 root root  18K Jan  8 22:09 dietpi-LICENSE.txt
-rw-rw-r-- 1 root root  15K Jan  8 22:09 dietpi-README.md
-rw-rw-r-- 1 root root  15K Jan 23 20:07 dietpi.txt
drwxr-xr-x 5 root root 4.0K Jan 23 20:07 grub
[  OK  ] DietPi-PREP | Completed, disk can now be saved to .img for later use, or, reboot system to start first run of DietPi.
[  OK  ] DietPi-PREP | To create an .img file, you can "poweroff" and run the following command from the host/external DietPi system:
        - bash -c "$(curl -sSfL https://github.com/MichaIng/DietPi/blob/master/.meta/dietpi-imager)"
```

If there's an actual log file in a specific location, let me know.

MichaIng, 2022-01-23:
Hmm, when `initramfs-tools` is replaced by `tiny-initramfs`, it throws a dependency warning:
```
dpkg: initramfs-tools: dependency problems, but removing anyway as you requested:
 linux-image-5.10.0-10-amd64 depends on initramfs-tools (>= 0.120+deb8u2) | linux-initramfs-tool; however:
  Package initramfs-tools is to be removed.
  Package linux-initramfs-tool is not installed.
  Package initramfs-tools which provides linux-initramfs-tool is to be removed.
```
`tiny-initramfs` however provides `linux-initramfs-tool` and hence satisfies the dependency, which I just verified on my VirtualBox VM also for the very recent Linux 5.10.0-11 upgrade.

It sets up the new kernel package and initramfs as expected:
```
Preparing to unpack .../linux-image-5.10.0-11-amd64_5.10.92-1_amd64.deb ...
Unpacking linux-image-5.10.0-11-amd64 (5.10.92-1) ...
Preparing to unpack .../linux-image-amd64_5.10.92-1_amd64.deb ...
Unpacking linux-image-amd64 (5.10.92-1) over (5.10.84-1) ...
Setting up tiny-initramfs-core (0.1-5+b11) ...
Setting up tiny-initramfs (0.1-5) ...
```

But it seems somehow as of the dependency warning there is a removal mark, or no manual installed mark set for the kernel package, so that it gets again autoremoved as part of the dbus etc removal:
```
Removing linux-image-amd64 (5.10.92-1) ...
Removing linux-image-5.10.0-11-amd64 (5.10.92-1) ...
```

Not sure whether this is a temporary case as of the kernel upgrade which has gone live between you ran the Debian installer and DietPi-PREP. Strange are the dependency warnings anyway since one dependency option is replaced by another, hence the kernel is not left without satisfied dependencies at any time. Needs to be replicated once the kernel package has been synced to all Debian mirrors. If this is somehow a regression of a recent APT or kernel/initramfs dependency handling, it should be possible to solve it by another `apt-mark manual linux-image-amd64` or `G_AGI linux-image-amd64` to assure that the manual installed mark is set, respectively no uninstall mark left, when dbus etc is removed.
___________
Just trying to replicate on the VM, and installing `initramfs-tools` while `tiny-initramfs` is installed throws these dependency warnings as well. The kernel packages are however not loosing their installed marks and are not autoremoved, same when installing again `tiny-initramfs` 🤔.

I also downgraded the `linux-image-amd64` package and removed the new kernel, then installed/upgraded the new kernel together with `initramfs-tools` over `tiny-initramfs` then still, the new kernel is installed but not autoremoved afterwards, despite the warnings. Also removing `/etc/apt/apt.conf.d/01autoremove-kernels` doesn't make a difference, which makes sense as the kernel package is marked manual when being installed explicitly.

Probably this all was due to the pending repository state as of the Linux package upgrade being pushed to the Debian mirrors, but I'll try to replicate with the Debian installer when I find time.

nickshanks347, 2022-01-23:
If I'm very honest, I don't know the PREP script ins/outs so this doesn't make a lot of sense to me, but this error is easily reproduced if you follow the VM image prep scripts word for word. Let me know how you get on and happy to test anything if you need.

Joulinar, 2022-01-24:
I did a test using `debian-11.2.0-amd64-netinst.iso` on VirtualBox Version 6.1.32 r149290. Installation using PREP script was working without issue and system was booting fine afterwards.

below the PREP log

<details>
  <summary>Logfile attached. Click to expand!</summary>
```
root@debian:~# bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/master/PREP_SYSTEM_FOR_DIETPI.sh')"
Get:1 http://deb.debian.org/debian bullseye InRelease [116 kB]
Get:2 http://security.debian.org/debian-security bullseye-security InRelease [44.1 kB]
Get:3 http://deb.debian.org/debian bullseye-updates InRelease [39.4 kB]
Get:4 http://security.debian.org/debian-security bullseye-security/main Sources [93.7 kB]
Get:5 http://security.debian.org/debian-security bullseye-security/main amd64 Packages [111 kB]
Get:6 http://deb.debian.org/debian bullseye/main Sources [8626 kB]
Get:7 http://deb.debian.org/debian bullseye/main amd64 Packages [8183 kB]
Get:8 http://deb.debian.org/debian bullseye-updates/main Sources [1868 B]
Get:9 http://deb.debian.org/debian bullseye-updates/main amd64 Packages [2592 B]
Fetched 17.2 MB in 9s (1911 kB/s)
Reading package lists... Done
[ INFO ] Selected Git branch: MichaIng/master
[ INFO ] DietPi-PREP | Detected distribution version: Bullseye (ID: 6)
[ INFO ] DietPi-PREP | Detected target CPU architecture: x86_64 (ID: 10)

 DietPi-PREP
─────────────────────────────────────────────────────
 Step: [0] Detecting existing DietPi system

[ INFO ] DietPi-PREP | No DietPi system found, skipping old instance uninstall...

 DietPi-PREP
─────────────────────────────────────────────────────
 Step: [1] Target system inputs

[ INFO ] DietPi-PREP | Entered image creator: Joulinar
[ INFO ] DietPi-PREP | Entered pre-image info: Debian
[ INFO ] DietPi-PREP | Selected hardware model ID: 20
[ INFO ] DietPi-PREP | Marking WiFi as NOT required
[ INFO ] DietPi-PREP | Selected Debian version: bullseye (ID: 6)

 DietPi-PREP
─────────────────────────────────────────────────────
 Step: [2] Downloading and installing DietPi source code

[  OK  ] DietPi-PREP | Checking URL: https://github.com/MichaIng/DietPi/archive/master.tar.gz
[  OK  ] DietPi-PREP | Downloading DietPi sourcecode
[  OK  ] DietPi-PREP | Extracting DietPi sourcecode
[ INFO ] DietPi-PREP | Moving kernel and boot configuration to /boot
[  OK  ] DietPi-PREP | mv DietPi-master/dietpi.txt /boot/
[  OK  ] DietPi-PREP | mv DietPi-master/README.md /boot/dietpi-README.md
[  OK  ] DietPi-PREP | mv DietPi-master/LICENSE /boot/dietpi-LICENSE.txt
[  OK  ] DietPi-PREP | Copy DietPi scripts to /boot/dietpi
[  OK  ] DietPi-PREP | Copy DietPi system files in place
[  OK  ] DietPi-PREP | Clean download location
[  OK  ] DietPi-PREP | Set execute permissions for DietPi scripts
[ INFO ] DietPi-PREP | Storing DietPi version info:
[  OK  ] DietPi-PREP | Desired setting in /boot/dietpi.txt was already set: DEV_GITBRANCH=master
[  OK  ] DietPi-PREP | Desired setting in /boot/dietpi.txt was already set: DEV_GITOWNER=MichaIng
[ INFO ] DietPi-PREP | Applying DietPi live patches to fix known bugs in this version
[  OK  ] DietPi-PREP | cp /boot/dietpi/.version /var/lib/dietpi/.dietpi_image_version
[  OK  ] DietPi-PREP | systemctl daemon-reload

 DietPi-PREP
─────────────────────────────────────────────────────
 Step: [3] APT configuration

[ INFO ] DietPi-PREP | Setting APT sources.list: bullseye 6
[  OK  ] DietPi-PREP | /boot/dietpi/func/dietpi-set_software apt-mirror default
[  OK  ] DietPi-PREP | mkdir -p /run/dietpi /var/tmp/dietpi/logs
[ INFO ] DietPi-PREP | APT update, please wait...
Hit:1 https://deb.debian.org/debian bullseye InRelease
[... 812 lines cut ...]
        - Linux debian 5.10.0-11-amd64 #1 SMP Debian 5.10.92-1 (2022-01-18) x86_64 GNU/Linux
[ INFO ] DietPi-PREP | The following kernel DEB packages have been found:
ii  linux-image-5.10.0-11-amd64 5.10.92-1                      amd64        Linux 5.10 for 64-bit PCs (signed)
[ INFO ] DietPi-PREP | The following kernel images and modules have been found:
/boot:
total 7.7M
-rw-r--r-- 1 root root   83 Jan 18 15:54 System.map-5.10.0-11-amd64
-rw-r--r-- 1 root root 231K Jan 18 15:54 config-5.10.0-11-amd64
drwxr-xr-x 4 root root 4.0K Jan 24 00:43 dietpi
-rw-rw-r-- 1 root root  18K Jan  8 22:09 dietpi-LICENSE.txt
-rw-rw-r-- 1 root root  15K Jan  8 22:09 dietpi-README.md
-rw-rw-r-- 1 root root  15K Jan 24 00:43 dietpi.txt
drwxr-xr-x 5 root root 4.0K Jan 24 00:43 grub
-rw------- 1 root root 843K Jan 24 00:43 initrd.img-5.10.0-11-amd64
-rw-r--r-- 1 root root 6.6M Jan 18 15:54 vmlinuz-5.10.0-11-amd64

/lib/modules:
total 4.0K
drwxr-xr-x 3 root root 4.0K Jan 24 00:28 5.10.0-11-amd64
[  OK  ] DietPi-PREP | Completed, disk can now be saved to .img for later use, or, reboot system to start first run of DietPi.
[  OK  ] DietPi-PREP | To create an .img file, you can "poweroff" and run the following command from the host/external DietPi system:
        - bash -c "$(curl -sSfL https://github.com/MichaIng/DietPi/blob/master/.meta/dietpi-imager)"
root@debian:~#
```
<details>

nickshanks347, 2022-01-24:
> 

That’s interesting. I’m trying to think if anything is/would’ve been different between my VM and your VM? I can see you used netinst, I used mini. Would this make a difference?

MichaIng, 2022-01-24:
As said, you used the Debian installer while latest kernel was 5.10.0-10 and ran PREP while latest kernel was just raised to 5.10.0-11. So this may have caused a special situation, though so far I wasn't able to replicate the issue.

So a test would be to use mini.iso, downgrade `linux-image-amd64` to v5.10.0-10 (hopefully it is then still available) and then run PREP. At best for debugging add an interactive bash call after `linux-image-amd64` and `tiny-initramfs` is installed and before the `G_AGP dbus ...` call in PREP, to check `dpkg -l` for non-`ii` marks.

MichaIng, 2022-02-20:
I'll mark this as closed. Feel free if your face the issue again, but as said, I think this was a rare condition as of the small time frame you ran into when running Debian installer and PREP afterwards.

closed 2022-02-20 by MichaIng