MichaIng/DietPi #5206: Cannot find CONFIG_BOOT_WAIT_FOR_NETWORK in dietpi.txt

skleeschulte, 2022-01-24:
Hi, I'm new to DietPi and found the AUTO_SETUP_AUTOMATED option in dietpi.txt (DietPi_RPi-ARMv8-Bullseye.img, boot partition). The comment there says to also set CONFIG_BOOT_WAIT_FOR_NETWORK=2. However, CONFIG_BOOT_WAIT_FOR_NETWORK is not in that dietpi.txt file. Am I supposed to add it? Or has it been removed/replaced (e.g. by AUTO_SETUP_BOOT_WAIT_FOR_NETWORK)? Thank you!

MichaIng, 2022-01-25:
Many thanks for your report. Indeed this has changed but we forgot to update the comment. Yes it is now called `AUTO_SETUP_BOOT_WAIT_FOR_NETWORK`, but there is no `=2` anymore. `=1` is now sufficient and the default anyway.

MichaIng, 2022-01-25:
Solved with: https://github.com/MichaIng/DietPi/commit/a097895

skleeschulte, 2022-01-25:
Wow, that was quick! :) Just an additional hint: While trying to find an explanation myself I searched the repository for the string CONFIG_BOOT_WAIT_FOR_NETWORK and also found it in .update/patches. But that might still be correct for backwards compatibility?

MichaIng, 2022-01-25:
The change was made with DietPi v7.7, so the patch checks for this config and in case migrates it 😉.

I just recognised/remembered that `AUTO_SETUP_BOOT_WAIT_FOR_NETWORK` is completely irrelevant for AUTO_SETUP_AUTOMATED. I made some throughout testing recently: https://github.com/MichaIng/DietPi/issues/5143#issuecomment-1008080428

Hence removed that comment completely: https://github.com/MichaIng/DietPi/commit/747a37b

#### Background
Before DietPi v7.7, this waiting for network option indeed delayed the whole boot process until a default route (usually indicating proper network connection) has been found. Since DietPi v7.7 only `dietpi-postboot.service` is affected by this option, which checks for DietPi updates, assures that system time has been network synced and starts DietPi controlled services at the end of the boot sequence. This however is decoupled from the login prompts and hence the first run setup as well. Login prompts are implicitly deferred to appear after the network has been brought up on Debian with systemd, without any changes needed, at least since with another change to the `ifup@.service`'s we greatly strengthened the meaning of the systemd network targets, compared to plain Debian. So now the whole boot is not deferred at some point anymore, but the individual parts of the boot sequence, where network is required (or wanted with DietPi's waiting for network option), are ordered to start after network bring via native systemd targets, without affecting parallelism of others 🙂.

closed 2022-01-29 by MichaIng