MichaIng/DietPi #5207: Every two hours, like clockwork: "wpa_supplicant[313]: nl80211: kernel reports: key addition failed"

jwatt, 2022-01-25:
Exactly every two hours (to the second!) wpa_supplicant complains "kernel reports: key addition failed". Most things appear unaffected, but at the exact time this happens, shairport-sync gets into a bad state causing it to play pulsing, completely distorted audio. My SSH connection to the device stays up, and running:

```
systemctl stop shairport-sync && sleep 1 && systemctl start shairport-sync
```

gets things working again. shairport-sync doesn't report any errors at the time. What is written to the logs at the moment the issue occurs is:

```
Jan 25 12:05:10 MySpeaker wpa_supplicant[313]: nl80211: kernel reports: key addition failed  
Jan 25 12:05:10 MySpeaker wpa_supplicant[313]: wlan0: WPA: Failed to set GTK to the driver (alg=3 keylen=16 keyidx=2)  
Jan 25 12:05:10 MySpeaker wpa_supplicant[313]: wlan0: CTRL-EVENT-DISCONNECTED bssid=fc:ec:da:81:7a:b0 reason=1 locally_generated=1  
Jan 25 12:05:10 MySpeaker kernel: ieee80211 phy0: send_key_to_dongle: wsec_key error (-52)  
Jan 25 12:05:10 MySpeaker wpa_supplicant[313]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD  
Jan 25 12:05:10 MySpeaker wpa_supplicant[313]: wlan0: Trying to associate with SSID 'MySSID'  
Jan 25 12:05:11 MySpeaker wpa_supplicant[313]: wlan0: Associated with af:8c:a0:8e:73:e2  
Jan 25 12:05:11 MySpeaker wpa_supplicant[313]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0  
Jan 25 12:05:11 MySpeaker wpa_supplicant[313]: wlan0: WPA: Key negotiation completed with af:8c:a0:8e:73:e2 [PTK=CCMP GTK=CCMP]  
Jan 25 12:05:11 MySpeaker wpa_supplicant[313]: wlan0: CTRL-EVENT-CONNECTED - Connection to af:8c:a0:8e:73:e2 completed [id=0 id_str=]
```

### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`

bookworm 0

- Kernel version | `uname -a`

Linux MySpeaker 5.10.92-v7+ #1514 SMP Mon Jan 17 17:36:39 GMT 2022 armv7l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

RPi Zero 2 W (armv7l)

#### Additional Information (if applicable)
- Software title

shairport-sync

- Was the software title installed freshly or updated/migrated?

Fresh install.

MichaIng, 2022-01-25:
Many thanks for your report.

Can you try whether this solves the issue for the current boot session:
```sh
iw wlan0 set power_save off
```
We disable WiFi power management via `wireless-power off` in `/etc/network/interfaces`, but this triggers legacy `iwconfig` instead of `iw`, and I'm not 100% sure whether both have exactly the same intended effect in all cases.

Also it is not the static `wpa_supplicant` service running, but the one invoked by `ifupdown` right?
```sh
systemctl status wpa_supplicant
```

There is an old and long issue about WiFi issues on RPi 3 (a.o.), which could include RPi Zero 2 W with shares the same SoC with RPi 3 (non-plus): https://github.com/raspberrypi/linux/issues/1342

jwatt, 2022-01-25:
Thanks a lot for your reply. Much appreciated. :-)

Running `iw wlan0 set power_save off` in the current session does not appear to fix the issue, unfortunately.

I didn't change any network management configuration from what comes in the image other than setting an SSID/password, so the output from `systemctl status wpa_supplicant` is:

```
○ wpa_supplicant.service - WPA supplicant
     Loaded: loaded (/lib/systemd/system/wpa_supplicant.service; disabled; vendor preset: enabled)
     Active: inactive (dead)
```

Thanks for the link to the old issue about WiFi issues on RPi 3. Maybe I can dig something up from that when I can spend a bit more time on this.

MichaIng, 2022-01-25:
Okay, so power management and a conflicting `wpa_supplicant` are then not the issue. The linked issue or opening a new one at the RPi kernel repository then seems to be most beneficial.

MichaIng, 2022-02-20:
Any news on this from your end?

jwatt, 2022-02-20:
I went through the linked issue (it took a while) and none of the things that helped others seemed to help or apply. I've got it on my todo list to continue investigating but haven't got to that yet.

MichaIng, 2022-02-20:
Best would be to contribute to that issue, probably we get some attention from RPi developers or someone who knows more about the WiFi driver.

MichaIng, 2022-04-03:
As of general open issues cleanup, I'll mark this as closed. But feel free to reopen when you found time to further debug.

closed 2022-04-03 by MichaIng