MichaIng/DietPi #5208: Unable to install Radarr from DietPi-Software

NH1922, 2022-01-25:
#### Details:
- Date           | Tue Jan 25 17:52:14 GMT 2022
- DietPi version | v8.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | `Linux DietPi 5.10.92-v7l+ #1514 SMP Mon Jan 17 17:38:03 GMT 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `curl -ILfvm 20 https://github.com/Radarr/Radarr/releases/download/v3.2.2.5080/Radarr.master.3.2.2.5080.linux-core-arm.tar.gz`
- Exit code      | 28
- Software title | DietPi-Software
#### Steps to reproduce:

1. dietpi-software
2. Browse Software
3. Select radarr
4. Install
#### Expected behaviour:

- ...
#### Actual behaviour:

- Fails with 
``` Checking URL: https://github.com/Radarr/Radarr/releases/download/v3.2.2.5080/Radarr.master.3.2.2.5080.linux-core-arm.tar.gz
 - Command: curl -ILfvm 20 https://github.com/Radarr/Radarr/releases/download/v3.2.2.5080/Radarr.master.3.2.2.5080.linux-core-arm.tar.gz
```
#### Extra details:

- ...
#### Additional logs:
```
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:14 --:--:--     0*   Trying 13.234.210.38:443...
* Connected to github.com (13.234.210.38) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
} [5 bytes data]
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
} [512 bytes data]
* TLSv1.3 (IN), TLS handshake, Server hello (2):
{ [122 bytes data]
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
{ [19 bytes data]
* TLSv1.3 (IN), TLS handshake, Certificate (11):
{ [2363 bytes data]
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
{ [80 bytes data]
* TLSv1.3 (IN), TLS handshake, Finished (20):
{ [36 bytes data]
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
} [1 bytes data]
* TLSv1.3 (OUT), TLS handshake, Finished (20):
} [36 bytes data]
* SSL connection using TLSv1.3 / TLS_AES_128_GCM_SHA256
* ALPN, server accepted to use h2
* Server certificate:
*  subject: C=US; ST=California; L=San Francisco; O=GitHub, Inc.; CN=github.com
*  start date: Mar 25 00:00:00 2021 GMT
*  expire date: Mar 30 23:59:59 2022 GMT
*  subjectAltName: host "github.com" matched cert's "github.com"
*  issuer: C=US; O=DigiCert, Inc.; CN=DigiCert High Assurance TLS Hybrid ECC SHA256 2020 CA1
*  SSL certificate verify ok.
* Using HTTP2, server supports multi-use
* Connection state changed (HTTP/2 confirmed)
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
} [5 bytes data]
[... 85 lines cut ...]
* Connection #1 to host objects.githubusercontent.com left intact
curl: (28) Operation timed out after 15134 milliseconds with 0 bytes received
HTTP/2 302
server: GitHub.com
date: Tue, 25 Jan 2022 17:51:28 GMT
content-type: text/html; charset=utf-8
vary: X-PJAX, X-PJAX-Container, Accept-Encoding, Accept, X-Requested-With
permissions-policy: interest-cohort=()
location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/77323220/0f638c80-c442-11eb-9be6-46e61865c14b?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220125%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220125T175128Z&X-Amz-Expires=300&X-Amz-Signature=ff6056aa00d35d9992a7ae0893320a7b1891d5db2a0d42531bf84b6deeebe922&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=77323220&response-content-disposition=attachment%3B%20filename%3DRadarr.master.3.2.2.5080.linux-core-arm.tar.gz&response-content-type=application%2Foctet-stream
cache-control: no-cache
strict-transport-security: max-age=31536000; includeSubdomains; preload
x-frame-options: deny
x-content-type-options: nosniff
x-xss-protection: 0
referrer-policy: no-referrer-when-downgrade
expect-ct: max-age=2592000, report-uri="https://api.github.com/_private/browser/errors"
content-security-policy: default-src 'none'; base-uri 'self'; block-all-mixed-content; child-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/; connect-src 'self' uploads.github.com objects-origin.githubusercontent.com www.githubstatus.com collector.githubapp.com api.github.com github-cloud.s3.amazonaws.com github-production-repository-file-5c1aeb.s3.amazonaws.com github-production-upload-manifest-file-7fdce7.s3.amazonaws.com github-production-user-asset-6210df.s3.amazonaws.com cdn.optimizely.com logx.optimizely.com/v1/events translator.github.com wss://alive.github.com *.actions.githubusercontent.com wss://*.actions.githubusercontent.com online.visualstudio.com/api/v1/locations raw.githubusercontent.com github-production-repository-image-32fea6.s3.amazonaws.com github-production-release-asset-2e65be.s3.amazonaws.com insights.github.com; font-src github.githubassets.com; form-action 'self' github.com gist.github.com objects-origin.githubusercontent.com; frame-ancestors 'none'; frame-src render.githubusercontent.com viewscreen.githubusercontent.com notebooks.githubusercontent.com; img-src 'self' data: github.githubassets.com identicons.github.com collector.githubapp.com github-cloud.s3.amazonaws.com secured-user-images.githubusercontent.com/ *.githubusercontent.com; manifest-src 'self'; media-src github.com user-images.githubusercontent.com/; script-src github.githubassets.com; style-src 'unsafe-inline' github.githubassets.com; worker-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/
content-length: 683
x-github-request-id: 9494:7255:1257230:1533D30:61F038BF

```

Based on #5106, I've also run 
`G_CONFIG_INJECT 'CONFIG_G_CHECK_URL_TIMEOUT=' 'CONFIG_G_CHECK_URL_TIMEOUT=20' /boot/dietpi.txt
`

NH1922, 2022-01-25:
Increased the timeout to 60 and got it working

`G_CONFIG_INJECT 'CONFIG_G_CHECK_URL_TIMEOUT=' 'CONFIG_G_CHECK_URL_TIMEOUT=60' /boot/dietpi.txt`

closed 2022-01-25 by NH1922

Joulinar, 2022-01-25:
That's pretty slow connecting to GitHub. Do you experience any other network issues?

NH1922, 2022-01-25:
The network was doing fine on my personal windows laptop and my phone. However the processes were slow on the Raspberry Pi 4. The initial auto upgrade took extremely long and so did installation of these tools. I am using ethernet (200mbps).

Joulinar, 2022-01-26:
downloading `apt` updates could be slow in some cases due to `apt` package server performance. Keep an eye if you experience such slow download performance again in future.