MichaIng/DietPi #5209: DietPi-Backup to NFS share hangs after checking for rsync

whisdol, 2022-01-25:
#### Required Information
- DietPi version
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | bullseye 0
- Kernel version | Linux pi 5.10.92-v8+ # 1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | Offical Raspberry Pi 4 AC adapter, USB-C, 5.1V, 3A
- SDcard used | SanDisk Extreme Pro 64GB

#### Additional Information (if applicable)
- Software title | DietPi-Backup
- Was the software title installed freshly or updated/migrated? Updated (part of DietPi 8.0?)
- Can this issue be replicated on a fresh installation of DietPi? Did not attempt to replicate
- Bug report ID | efa1e045-5c07-40ae-ae0e-02945d637a59

#### Steps to reproduce
1. Configure NFS share from Synology Disk Station via DietPi-Drive_Manager (/mnt/titan in my case)
2. Configure DietPi-Backup to back up to mounted NFS share (/mnt/titan/dietpi-backup)
3. Run backup

#### Expected behaviour
Backup should start

#### Actual behaviour
Backup preparation hangs after checking that rsync is installed. Ctrl-C at that stage always prints the syntax error in line 188:
```
root@pi:/home/dietpi# dietpi-backup
[  OK  ] DietPi-Drive_Manager | Drive mount selected: /mnt/titan

 DietPi-Backup

 DietPi-Backup
─────────────────────────────────────────────────────
 Mode: Backup

[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.4-fpm
[ INFO ] DietPi-Backup | Checking for required APT packages: rsync
^Crsync error: received SIGINT, SIGTERM, or SIGHUP (code 20) at rsync.c(703) [sender=3.2.3]
rsync error: received SIGUSR1 (code 19) at main.c(1595) [generator=3.2.3]
/boot/dietpi/dietpi-backup: line 188: + (  +  ) * 131072 : syntax error: operand expected (error token is ") * 131072 ")
```

#### Extra details
- This worked when I initially set this up with an earlier DietPi version (apparently my last backup was on 2022-01-22)
- I tested
    1. mounting the NAS, Backup share as nested folder in `/mnt/Backup/volume1/Backup/` (original setup; `/mnt/Backup` and `/mnt/Backup/volume1` were read-only in this case)
    2. mounting my Backup share directly (current config in fstab: `192.168.0.2:/volume1/Backup /mnt/titan ...`)
- I can write to the Backup share (now `/mnt/titan`) and the dietpi-backup folder (`/mnt/titan/dietpi-backup`)
- Setting a new backup location and searching for a previous backup always just "hangs" as well (but maybe I'm not patient enough?)

MichaIng, 2022-01-25:
Many thanks for your report.

You interrupt `rsync` while it does a dry run to obtain required disk space. I'll add another notification, indicating that the dry run is running.

whisdol, 2022-01-25:
Ha, indeed! I was just too impatient, it seems.
A message that something is happening would be helpful. Thank you!

MichaIng, 2022-01-25:
Done: https://github.com/MichaIng/DietPi/commit/cd3bd5f

Does this work for you? The dry run does not actually transfer anything but collects info about all files which would be transferred and gives a summary about this. That way we know exactly the transferred size + the number of files and filters, which in combination with target drive block size allows to calculate the worst case disk space which is used by the target.

closed 2022-01-29 by MichaIng