MichaIng/DietPi #5210: Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service is masked.

Carl0152, 2022-01-26:
Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service is masked.### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

Carl0152, 2022-01-26:
#### Details:
- Date           | Wed Jan 26 02:00:01 CET 2022
- DietPi version | v8.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux NAS 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `systemctl start systemd-timesyncd`
- Exit code      | 1
- Software title | DietPi-TimeSync
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service is masked.
```

closed 2022-01-26 by Carl0152

MichaIng, 2022-01-26:
You solved it yourself?
```sh
systemctl unmask systemd-timesyncd
```
The question is why it got masked in the first place 🤔.

MichaIng, 2022-01-26:
OMV is very intrusive indeed, it installs and sets up a bunch of OS level things, basically it is an own OS, I mean there are images provided. When using OMV to configure/setup things, you cannot use DietPi tools to setup the same things, both simply do the same, but not in the same way. Likely OMV installs a different time sync daemon and masks `systemd-timesyncd` then. This mask wouldn't be required since the `systemd-timesyncd.service` detects other time sync daemons automatically and in case does not start.

However, if there is really a different time sync daemon now, you need to switch to "Custom" time sync mode with DietPi, so that it does not try to start `systemd-timesyncd` in the first place:
```sh
/boot/dietpi/func/dietpi-set_software ntpd-mode 0
```

Joulinar, 2022-01-26:
2 year's ago I did some testing around OMV, prerequisite and challenges https://dietpi.com/phpbb/viewtopic.php?p=24668#p24668

Carl0152, 2022-01-26:
das ging aber schnell - und war genau das richtige.
Ich hatte schon befürchtet, daß ich das OS wechseln muß.
Danke

> OMV is very intrusive indeed, it installs and sets up a bunch of OS level things, basically it is an own OS, I mean there are images provided. When using OMV to configure/setup things, you cannot [...]

Sehr geehrte Damen und Herren,

Mit freundlichen Grüßen

K.-H. Graf
-Versicherungsmakler -

Versicherungsbüro Karl-Heinz Graf
Schwaansche Str. 1
18055 Rostock
Tel. 0381 - 44 71 533 e-mail: ***@***.***
_______________________________________________________________________________________________________________________________________________ 
Umsatzsteuer-ID-Nummer: DE170499822 Registrierungsnr.: D-KNOT-YRUQU-28
Diese Nachricht kann vertrauliche Informationen enthalten. Wenn Sie nicht der in der Nachricht angegebene Adressat sind, darf diese Nachricht nicht kopiert oder an Dritte weitergeleitet werden. In einem solchen Fall vernichten Sie bitte die Nachricht und informieren Sie den Absender. In dieser Nachricht enthaltene Informationen, die nicht in Zusammenhang mit einer offiziellen Geschaeftstaetigkeit zwischen Absender und Empfaenger stehen, gelten als nicht erteilt. Rechte aus solchen Informationen koennen nicht geltend gemacht werden.

Joulinar, 2022-01-26:
Pls try to stick to English language if possible. Even if we understood German, it's quite a challenge for other user. Thx for understanding 😊

Carl0152, 2022-01-26:
I vow to do better

(my english knowledge is about 0)😊

> 
>  
> Pls try to stick to English language if possible. Even if we understood German, it's quite a challenge for other user. Thx for understanding 😊

> https://github.com/MichaIng/DietPi/issues/5210#issuecomment-1022341800

Sehr geehrte Damen und Herren,

Mit freundlichen Grüßen

K.-H. Graf
-Versicherungsmakler -

Versicherungsbüro Karl-Heinz Graf
Schwaansche Str. 1
18055 Rostock
Tel. 0381 - 44 71 533 e-mail: ***@***.***
_______________________________________________________________________________________________________________________________________________ 
Umsatzsteuer-ID-Nummer: DE170499822 Registrierungsnr.: D-KNOT-YRUQU-28
Diese Nachricht kann vertrauliche Informationen enthalten. Wenn Sie nicht der in der Nachricht angegebene Adressat sind, darf diese Nachricht nicht kopiert oder an Dritte weitergeleitet werden. In einem solchen Fall vernichten Sie bitte die Nachricht und informieren Sie den Absender. In dieser Nachricht enthaltene Informationen, die nicht in Zusammenhang mit einer offiziellen Geschaeftstaetigkeit zwischen Absender und Empfaenger stehen, gelten als nicht erteilt. Rechte aus solchen Informationen koennen nicht geltend gemacht werden.

Joulinar, 2022-01-26:
That's not a problem. Feel free to use think's like Google Translate. 😉

MichaIng, 2022-01-26:
Better: https://www.deepl.com/translator
🙂

Carl0152, 2022-01-26:
I join Joulinar:
deepl is in my bookmarks bar

> 
>  
> Better: https://www.deepl.com/translator

> 🙂

> https://github.com/MichaIng/DietPi/issues/5210#issuecomment-1022351541

Sehr geehrte Damen und Herren,

Mit freundlichen Grüßen

K.-H. Graf
-Versicherungsmakler -

Versicherungsbüro Karl-Heinz Graf
Schwaansche Str. 1
18055 Rostock
Tel. 0381 - 44 71 533 e-mail: ***@***.***
_______________________________________________________________________________________________________________________________________________ 
Umsatzsteuer-ID-Nummer: DE170499822 Registrierungsnr.: D-KNOT-YRUQU-28
Diese Nachricht kann vertrauliche Informationen enthalten. Wenn Sie nicht der in der Nachricht angegebene Adressat sind, darf diese Nachricht nicht kopiert oder an Dritte weitergeleitet werden. In einem solchen Fall vernichten Sie bitte die Nachricht und informieren Sie den Absender. In dieser Nachricht enthaltene Informationen, die nicht in Zusammenhang mit einer offiziellen Geschaeftstaetigkeit zwischen Absender und Empfaenger stehen, gelten als nicht erteilt. Rechte aus solchen Informationen koennen nicht geltend gemacht werden.