MichaIng/DietPi #5212: DietPi-Letsencrypt | certificate not create on Apache2

Joulinar, 2022-01-27:
At the moment SSL certificate creation is failing with following error message

```
    raise ValueError("Unable to insert label!")
ValueError: Unable to insert label!
```

Looks like `certbot` is expecting a config value inside Apache configuration file `/etc/apache2/sites-available/000-default.conf`

MichaIng, 2022-01-27:
Solved with: https://github.com/MichaIng/DietPi/commit/bca964e
Changelog: https://github.com/MichaIng/DietPi/commit/8f606c4

MichaIng, 2022-01-27:
Hmm, we should also remove the comment which states to add directives for port 80 only, copied from `000-default.conf`. Another idea is to add `ServerName` to `000-default.conf` in the first place. It doubles with the one in `dietpi.conf` on a fresh install, but finally it is probably fine to have a generic global server name and then override it per vhost 🤔.

closed 2022-01-29 by MichaIng