MichaIng/DietPi #5213: DietPi-Software | OpenBazaar: Project dead, switch to fork?

MichaIng, 2022-01-28:
https://github.com/OpenBazaar/openbazaar-go/issues/2109

The fork, as far as I can see, didn't get much attention yet. Not sure whether we want to implement/migrate without a notable amount of (non project team) users, reviews etc? The desktop client is provided with compiled binary releases as well. Both undergoes quite some rework here and there, so without having enough users which use, test, review, report bugs, I fear that it may not be able to get sufficiently stable soon.

Currently only 15 reported users, hence probably ~100 actual users, and it never was more since we track it, hence from that end we'd simply drop it. Only thing is that I like the project as it is quite unique and when I tested it a while ago the marketplace was quite large, with interesting gems to discover. I didn't have any crypto currency that time, and do not have now (aside of DietPi BAT donations, but they need to stay to hopefully undergo a raising exchange rate 😄), but generally nice to have an alternative to ebay etc with crypto payment.

cocoflan, 2022-01-29:
Yes i think this platform will stop soon.

MichaIng, 2022-01-29:
Link to the fork: https://github.com/mobazha/openbazaar-go
But I have no idea whether the community (a major part) knows about it an migrates or whether the marketplace itself will simply die. The website is down, and in case any forum running on that host.

MichaIng, 2022-01-30:
Okay the original server fails to start since it uses `openbazaar.org` URLs for several APIs. Testing the fork now.

MichaIng, 2022-01-30:
Done: #5222

MichaIng, 2022-02-01:
Done so far what we can do. For ARM support (compiling) there is still either a repo-wise change required or a different method than `go get`: https://github.com/mobazha/openbazaar-go/issues/1

closed 2022-02-01 by MichaIng