MichaIng/DietPi #5214: Cannot install Mate Desktop -> "you have held broken packages"

n0valis, 2022-01-28:
#### Details:
- Date           | Fri Jan 28 13:10:42 CET 2022
- DietPi version | v7.6.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | From scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux THORIN 5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -qq --allow-change-held-packages install mate-desktop-environment-core mate-media upower xterm`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. install mate from dietpi-softwate
I did apt update/upgrade before

E: Unable to correct problems, you have held broken packages.
```

Joulinar, 2022-01-28:
What is the output of

`apt update && apt upgrade`

n0valis, 2022-01-28:
All packages are up to date.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

MichaIng, 2022-01-28:
And:
```sh
apt install mate-desktop-environment-core mate-media upower xterm
```

n0valis, 2022-01-28:
Then, this time I get a little bit more info:
The following packages have unmet dependencies:
 libgail-3-0 : Depends: libgtk-3-0 (= 3.24.24-4+rpt3) but 3.24.24-4 is to be installed
E: Unable to correct problems, you have held broken packages.

n0valis, 2022-01-28:
confirmed. That was the problem. Thank you for pointing that out.

Maybe you can tell me how the config is changed to get Mate instead of LXDE via vnc.
I have configured Mate as Desktop in Dietpi-Software, but LXDE still shows up. As I (can) only access via vnc I guess there is another config to be made.

MichaIng, 2022-01-28:
`startx` on the local console then also starts LXDE, right? Hmm, I would have expected that MATE sets itself up as default desktop (X client) automatically on install, obviously it does not.

Can you check:
```sh
cat /etc/X11/default-display-manager
```
You should be able to select it via:
```sh
update-alternatives --config x-window-manager
```

MichaIng, 2022-01-28:
Original issue solved with: https://github.com/MichaIng/DietPi/commit/fa1c70c

n0valis, 2022-01-28:
> `startx` on the local console then also starts LXDE, right? Hmm, I would have expected that MATE sets itself up as default desktop (X client) automatically on install, obviously it does not.
> 
> Can you check:
> 
> ```shell
> cat /etc/X11/default-display-manager
> ```
/usr/sbin/lightdm

> You should be able to select it via:
> 
> ```shell
> update-alternatives --config x-window-manager
> ```
I am sorry but that does not say me anything:

~# update-alternatives --config x-window-manager
There are 2 choices for the alternative x-window-manager (providing /usr/bin/x-window-manager).

  Selection    Path              Priority   Status
------------------------------------------------------------
* 0            /usr/bin/openbox   90        auto mode
  1            /usr/bin/marco     30        manual mode
  2            /usr/bin/openbox   90        manual mode

Press <enter> to keep the current choice[*], or type selection number:

Joulinar, 2022-01-28:
@MichaIng 
should we in general allow to have 2 different desktop installed? Or should we handle it same way like we do for AdBlocker and WebServer?

n0valis, 2022-01-28:
I think for Mate it probably is marco. How can I change to
 /usr/bin/marco 30 auto mode
?

MichaIng, 2022-01-28:
Jep, MATE uses Marco display manager, run the command again, then hit `1`. I hope the display manager alone is sufficient 😅. Nasty that is has much lower priority than OpenBox, not sure whether this choice survives e.g. OpenBox package upgrades and such or whether it may be reverted automatically by some triggers.

@Joulinar 
Generally desktops do not conflict with each other and with mentioned commands it can be selected which desktop to start via `startx`, login masks and VNC servers, so I don't see a need to unallow multiple desktop installs. But it may be an idea to add an info to our docs about how to select a different default desktop, how to start one explicitly (`mate-session`, `startlxde` etc), and it may be an idea to select the newly installed desktop as new default desktop automatically in `dietpi-software`, but we need to verify that the selection stays active even when priority is lower.

n0valis, 2022-01-28:
> Jep, MATE uses Marco display manager, run the command again, then hit `1`. I hope the display manager alone is sufficient 

Ok, here is what I do:
check  option 1
reboot

goto PC's vnc client 
open the Dietpi
LXDE opens
:(

n0valis, 2022-01-28:
I found this one. That kind of did the trick
update-alternatives --config x-session-manager
There are 4 choices for the alternative x-session-manager (providing /usr/bin/x-session-manager).

  Selection    Path                      Priority   Status
------------------------------------------------------------
* 0            /usr/bin/startlxde         50        auto mode
  1            /usr/bin/lxsession         49        manual mode
  2            /usr/bin/mate-session      50        manual mode
  3            /usr/bin/openbox-session   40        manual mode
  4            /usr/bin/startlxde         50        manual mode

Press <enter> to keep the current choice[*], or type selection number:

n0valis, 2022-01-28:
It is Linux after all.
The 1.000.000 Dollar question.
You want to change your click environment. What do chose?
A window manager
B display manager 
C desktop manager 
D session manager
;)

MichaIng, 2022-01-28:
Ahh, that makes sense, I mixed it up as well 😄. I guess the default `x-window-manager` then is without effect, as long as a desktop environment does not support multiple ones.
- The display manager is usually a login mask, e.g. LightDM in your case (pulled by LXDE), started by `startx` when any is installed.
- The session manager is the actual desktop environment, started by `startx` when no display manager is installed, else started by the display manager after successful login.
- The window manager is responsible for drawing the windows used by the desktop environment. Usually a desktop depends on a specific window manager, but it is generally possible that it can use different ones. A window manager can also be started as standalone application, in theory, the i3 wm is quite usable without an actual desktop environment installed. Things like a panel, menu desktop icons etc are then missing, of course.

A "desktop manager" AFAIK does not exist as general term.

n0valis, 2022-01-28:
Yes,  that makes sense. 
Looking forward to 8.1

closed 2022-01-28 by n0valis