MichaIng/DietPi #5217: [Question] Is it possible to mount docker root directory to VM shared folder?

LazyGeniusMan, 2022-01-29:
I 'm not a fan on how WSL2 backend Docker use my resource, it's kinda not optimal, so I was thinking to create some kind of DietPi VM backend docker.  Is that possible ?

I have `docker-data` and `docker-compose` folder in my host OS `~/docker` directory. Using VirtualBox, I have set those folder into shared folder, I want to make `docker-compose` in my host OS connected to `~/docker-compose` in dietpi. I have succesfully do this.
```
root@DietPi:~# df -H
Filesystem             Size  Used Avail Use% Mounted on
udev                   509M     0  509M   0% /dev
/dev/sda1              8.5G  4.0G  4.1G  50% /
tmpfs                  512M     0  512M   0% /dev/shm
tmpfs                  205M  5.6M  200M   3% /run
tmpfs                  5.3M     0  5.3M   0% /run/lock
tmpfs                  1.1G  8.2k  1.1G   1% /tmp
tmpfs                   53M   25k   53M   1% /var/log
shared_docker-compose  374G  314G   61G  84% /root/docker-compose
overlay                8.5G  4.0G  4.1G  50% /mnt/dietpi_userdata/docker-data/overlay2/4e172227a64304f899cad8498053c5256762b57f8445bc84b21259e332137e23/merged
overlay                8.5G  4.0G  4.1G  50% /mnt/dietpi_userdata/docker-data/overlay2/def5f73392598d54ff28511fe6ece53909bac10859578a95cc33ef1b9e86219a/merged
```

But when I tried to connect docker root dir (which is `/mnt/dietpi_userdata/docker-data`, CMIIW) I can't get it work. peobably because of that overlay thingy ? is there  a workaround?

I want to connect docker root dir so I don't have to worry about disk space (don't have to resizing the virtual disk if it's full), since it will use host disk space instead (unless I missunderstood ?).

MichaIng, 2022-01-29:
`/mnt/dietpi_userdata/docker-data` is the place for the Docker images, mounted as overlayfs when the container starts. I indeed guess that it causes issues when this is tried to be done into onto a Windows NTFS filesystem. Also performance-wise I'm not sure whether for running applications it is a good idea to R/W to a shared VM host/guest directory 🤔. How exactly do you not "get it work"? Does Docker fail to start of the VM host/guest directory fail to mount?

LazyGeniusMan, 2022-01-29:
> Also performance-wise I'm not sure whether for running applications it is a good idea to R/W to a shared VM host/guest directory 🤔

So my option if I want flexible disk space is Hyper-V then ? since VirtualBox can't shink disk size AFAIK, only expand supported. But I will not have static IP since my Hyper V behave so weird when I use external vSwitch from my WiFi :( (ref: https://github.com/MichaIng/DietPi/issues/5186 )

> How exactly do you not "get it work"? Does Docker fail to start of the VM host/guest directory fail to mount?

The mount is kinda (?) successful but the file is gone and not synced
```
root@DietPi:/mnt/dietpi_userdata/docker-data# df -H
Filesystem             Size  Used Avail Use% Mounted on
udev                   509M     0  509M   0% /dev
/dev/sda1              8.5G  4.0G  4.1G  50% /
tmpfs                  512M     0  512M   0% /dev/shm
tmpfs                  205M  3.2M  202M   2% /run
tmpfs                  5.3M     0  5.3M   0% /run/lock
tmpfs                  1.1G     0  1.1G   0% /tmp
shared_docker-compose  374G  314G   61G  84% /root/docker-compose
tmpfs                   53M   25k   53M   1% /var/log
overlay                8.5G  4.0G  4.1G  50% /mnt/dietpi_userdata/docker-data/overlay2/def5f73392598d54ff28511fe6ece53909bac10859578a95cc33ef1b9e86219a/merged
overlay                8.5G  4.0G  4.1G  50% /mnt/dietpi_userdata/docker-data/overlay2/4e172227a64304f899cad8498053c5256762b57f8445bc84b21259e332137e23/merged
root@DietPi:/mnt/dietpi_userdata/docker-data# ls
buildkit  containers  image  network  overlay2  plugins  runtimes  swarm  tmp  trust  volumes
root@DietPi:/mnt/dietpi_userdata/docker-data# sudo mount -t vboxsf shared_docker-data /mnt/dietpi_userdata/docker-data
root@DietPi:/mnt/dietpi_userdata/docker-data# cd ..
root@DietPi:/mnt/dietpi_userdata# cd docker-data
root@DietPi:/mnt/dietpi_userdata/docker-data# ls
root@DietPi:/mnt/dietpi_userdata/docker-data# df -H
Filesystem             Size  Used Avail Use% Mounted on
udev                   509M     0  509M   0% /dev
/dev/sda1              8.5G  4.0G  4.1G  50% /
tmpfs                  512M     0  512M   0% /dev/shm
tmpfs                  205M  3.2M  202M   2% /run
tmpfs                  5.3M     0  5.3M   0% /run/lock
tmpfs                  1.1G     0  1.1G   0% /tmp
shared_docker-compose  374G  314G   61G  84% /root/docker-compose
tmpfs                   53M   25k   53M   1% /var/log
shared_docker-data     374G  314G   61G  84% /mnt/dietpi_userdata/docker-data
```

MichaIng, 2022-01-29:
Yes, shrinking virtual disks is not supported, I guess since so many users run into issues when not shrinking filesystem and partition sizes within the VM first. However, you can use a very large virtual disk size with dynamic allocation: Only the actually used size is allocated on host level. When much disk space is freed, you can even compact the VDI file, I'm using this batch script:
```bat
@echo off
for %%i in (%*) do (
	"C:\Program Files\Oracle\VirtualBox\VBoxManage" modifymedium disk %%i --compact
)
pause
```
Making a file of this, you can drag&drop the `*.vdi` files onto the script and it will be compacted to match again actually used size.

Of course when you mount something onto `/mnt/dietpi_userdata/docker-data` after Docker (resp. the containers) got installed, you overmount the data and hence the directory appears empty. If you really want to achieve this, mount the shared folder first, then install Docker/the containers so that they get installed into the shared folder instead of into the virtual disk below it.

LazyGeniusMan, 2022-01-30:
> Making a file of this, you can drag&drop one or more `*.vdi` files onto the script and they will be compacted to match again actually used size.

So I have to run this everytime my actual virtual disk in decreased ? (e.g. after deleting docker images/volume)
Does the VM has to be turned off ?

> If you really want to achieve this, mount the shared folder first, then install Docker/the containers so that they get installed into the shared folder instead of into the virtual disk below it.

It's failed to install
```
││  - Command: docker volume create portainer_data
││  - Exit code: 1
││  - DietPi version: v8.0.2 (MichaIng/master) | HW_MODEL: 20 | HW_ARCH: 10 | DISTRO: 6
││  - Image creator: DietPi Core Team
││  - Pre-image: from scratch
││  - Error log:
││ Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?
```

It's successfully synced tho
```
PS C:\Users\LazyGeniusMan\docker\docker-data> tree /F
Folder PATH listing
Volume serial number is B84C-EB95
C:.
├───check-overlayfs-support209265903
│   └───work
│       └───work
├───check-overlayfs-support622291920
│   └───work
│       └───work
├───containers
├───devicemapper
│   ├───devicemapper
│   │       data
│   │
│   └───metadata
├───plugins
│   ├───storage
│   │   └───ingest
│   └───tmp
├───runtimes
└───tmp
```

MichaIng, 2022-01-30:
Please check:
```sh
journalctl -u docker containerd
```
I however believe that it indeed requires an underlying filesystem with UNIX permissions and symlinks support, hence `vboxsf` cannot be used.

> So I have to run this everytime my actual virtual disk in decreased ? (e.g. after deleting docker images/volume)
Does the VM has to be turned off ?

Not sure how tight your host system's drive space is, that you could need to shrink the virtual drive whenever possible, and how often you aim to remove containers/images without adding new ones (so that freed space is used shortly after)? I do this in rare cases when I did very large test installs on my testing VMs while there is actually not really a need to do so due to sufficient host drive space. I haven't tested it for a running VM, but believe it indeed needs to be stopped, otherwise would advise to stop it.

LazyGeniusMan, 2022-01-30:
> I however believe that it indeed requires an underlying filesystem with UNIX permissions and symlinks support, hence `vboxsf` cannot be used.

Ah ok I'll just do what you said using big dynamic disk then.

> Not sure how tight your host system's drive space is, that you could need to shrink the virtual drive whenever possible, and how often you aim to remove containers/images without adding new ones (so that freed space is used shortly after)?

I just want to know if it's like one time only thing or not. I have .bat script in my startup folder to start my VM Headless on boot, so I guess I'll just add your script above it ?

Currently I have 2 DietPi in my VBox, 1 for pi-hole and 1 for docker, do you think it's better if they are separated or not?

Sorry this question is not directly related to DietPi, I'm dual boot Windows and Linux in my PC, is there any way so that they shared the same docker image so I don't have to pull the same image from internet twice?

MichaIng, 2022-01-31:
> I just want to know if it's like one time only thing or not.

It shrinks the disk as much as possible depending on its current content.

> I have .bat script in my startup folder to start my VM Headless on boot, so I guess I'll just add your script above it?

That's actually a nice idea. But it can take a while, ~30 seconds for a 16 GiB disk on a quick test I just did. Strangely the file size was not reduced at all this time, this worked much better last time I tried, not sure whether the location of raw data within the disk plays a role 🤔.

LazyGeniusMan, 2022-01-31:
> Strangely the file size was not reduced at all this time, this worked much better last time I tried

Maybe because of this[^1]. I read on the archlinux wiki, looks like your script is the last step?

[^1]: https://wiki.archlinux.org/title/VirtualBox#Compact_virtual_disks

MichaIng, 2022-01-31:
Ah, I'm an idiot, totally forgot to zero out free space. The host system tools of course cannot know which non-zero parts of the disk are non-deleted data and which parts are obsolete/deleted data, hence for best results, one needs to e.g. run the following within the VM:
```sh
mount -o remount,ro /
zerofree -v "$G_ROOTFS_DEV"
mount -o remount,rw /
```
Remounting the root filesystem as R/O requires all services to be stopped and swap files disabled:
```sh
dietpi-services stop
swapoff -a
```
Of course `zerofree` can also be done offline by mounting the virtual disk to another virtual machine.

LazyGeniusMan, 2022-02-02:
Aight thanks

closed 2022-02-02 by LazyGeniusMan