MichaIng/DietPi #5218: Open Beta v8.1 | Please help testing and hardening the upcoming release

MichaIng, 2022-01-29:
RC version | v8.1.0
---------- | -------
v8.1.0 PR | #5215
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2022-02-05**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av8.1
____
### Beta v8.1.0
_(2022-01-29)_

### SBC support
- Odroid N2 | Support for Odroid N2 with modern mainline Linux images has been added, including proper boot configuration, serial console selection, HDMI and 3.5mm audio selection and more.

### Changes
- DietPi-Benchmark | During RAM I/O benchmarks, the CPU governor is now changed to "performance" so that the result is not affected by possible CPU frequency scaling but the raw/max RAM I/O performance is measured. A significant effect of this has been recognised on Odroid N2 images with modern mainline kernel. Since the CPU benchmark is done first, this has no effect on its score and CPU temperature results.
- DietPi-Drive_Manager | Btrfs subvolume mounts are now preserved in /etc/fstab. Many thanks to @laddde for implementing this feature: https://github.com/MichaIng/DietPi/pull/5176
- DietPi-Drive_Manager | The Samba mount dialogues do now indicate that a hostname can be entered instead of a server IP and a shared folder path instead of a share name only.
- DietPi-Backup | A notification is not printed when rsync dry-run to obtain required disk space is starting, since this can take a while. Many thanks to @whisdol for reporting the the impression of a hanging script: https://github.com/MichaIng/DietPi/issues/5209

### Fixes
- DietPi-Services | Resolved an issue where service edits and process tool settings were not effective as they were stored to the wrong file path. Many thanks to @TopFord and @MicDG for reporting this issue: https://dietpi.com/phpbb/viewtopic.php?p=41509#p41509
- DietPi-LetsEncrypt | Resolved an issue where Certbot failed on fresh Apache installs since the 000-default.conf vhost by default does not contain any directive anymore. The ServerName directive with the entered domain name is now added to 000-default.conf instead of apache2.conf, with the benefit that it overrides the global ServerName directive in dietpi.conf, which uses the local IP address only.
- DietPi-Software/DietPi-AutoStart | Chromium: Resolved an issue where the Chromium kiosk autostart option failed on all non-RPi systems. 
- DietPi-Software | FuguHub: Resolved an issue where the uninstall failed as the service was not stopped and removed as expected, before attempting to remove the service user.
- DietPi-Software | PHP: Resolved an issue where the PATH variable was not available for PHP applications with Lighttpd and Nginx webservers. It is now set to the static value "/usr/local/bin:/usr/bin:/bin", the default PATH excluding super user executables "sbin". Many thanks to @shanew1694 for reporting a related issue with rTorrent: https://github.com/MichaIng/DietPi/issues/5161#issuecomment-1013381362
- DietPi-Software | MineOS: Resolved an issue where the install failed on Bullseye: https://github.com/MichaIng/DietPi/issues/5181
- DietPi-Software | Cuberite: Resolved an issue where the server failed to start on ARMv8 systems: https://github.com/MichaIng/DietPi/issues/5180
- DietPi-Software | MATE: Resolved an issue where the install failed when LXDE is installed already. Many thanks to @n0valis for reporting this issue: https://github.com/MichaIng/DietPi/issues/5214
- DietPi-Software | Ampache: Resolved an issue on Bullseye systems where Ampache v5 installs/reinstalls did not work because of a changed webroot directory: https://github.com/MichaIng/DietPi/pull/5205

Joulinar, 2022-01-31:
@MichaIng 
it is not directly related to the Beta but might be something we could/should include. I noticed that desktop systems are not able boot on HyperV VM systems. Because video package `xserver-xorg-video-fbdev` is missing. I know we only install `xserver-xorg-video-vmware` in general on all VM's (HW ID 20)

https://github.com/MichaIng/DietPi/blob/fa396c402ca1dfa902de2e074d8e50f97b60ef6d/dietpi/dietpi-software#L2661-L2662

MichaIng, 2022-01-31:
Hyper-V does not support KMS/DRM?
```sh
ls -l /dev/dri
```
We could check whether any `/dev/dri/card[0-9]` exists and based on this decide whether to install `xserver-xorg-video-fbdev` or `xserver-xorg-video-vmware`.

Joulinar, 2022-02-01:
there is no `/dev/dri` on the HyperV VM

```
root@DietPi:~# ls -l /dev/dri
ls: cannot access '/dev/dri': No such file or directory
root@DietPi:~#
```

MichaIng, 2022-02-01:
Solved with: https://github.com/MichaIng/DietPi/commit/a6340fa

MichaIng, 2022-02-02:
#### DietPi Beta v8.1.1 has been merged: #5226

Joulinar, 2022-02-02:
I tested desktop install on HyperV which seems to be working fine now. Correct package `server-xorg-video-fbdev` has been selected 👍 

```
 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing X.Org X Server: aka X11 - X Window System implementation

[  OK  ] DietPi-Software | cd /tmp/DietPi-Software
[  OK  ] DietPi-Software | Checking URL: https://raw.githubusercontent.com/MichaIng/DietPi/beta/.conf/dps_6/98-dietpi-disable_dpms.conf
[ INFO ] DietPi-Software | G_THREAD_START_0 | curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/beta/.conf/dps_6/98-dietpi-disable_dpms.conf -o 98-dietpi-disable_dpms.conf
[ INFO ] DietPi-Software | APT install for: xserver-xorg-core xserver-xorg-input-libinput xinit dbus-x11 xfonts-base x11-xserver-utils x11-utils libgl1-mesa-dri mesa-utils mesa-utils-extra xserver-xorg-video-fbdev, please wait...
```

Desktop is correctly started afterwards.

MichaIng, 2022-02-02:
Great, thanks for testing 👍🙂

Joulinar, 2022-02-05:
@MichaIng 
Just as information. There seems to be an issue to restart HA from GUI as the service is going to be killed by own application. Similar to what we had on Webmin. Unfortunately the same Workaround we applied there, is not working on HA. Maybe you can have a look if you find a better solution. https://dietpi.com/phpbb/viewtopic.php?p=42859#p42859

MichaIng, 2022-02-05:
A very recent change indeed. Here it is: https://github.com/home-assistant/core/issues/65450
Hence:
```sh
echo -e '[Service]\nRestartForceExitStatus=100' > /etc/systemd/system/home-assistant.service.d/dietpi-services_edit.conf
systemctl daemon-reload
systemctl restart home-assistent
```
I'll add it to `dietpi-software` as well.

MichaIng, 2022-02-05:
#### DietPi Beta v8.1.2 has been merged: #5230

After a last review, this will be merged to stable.

MichaIng, 2022-02-05:
#### DietPi v8.1 has been released: #5229

Many thanks to all testers ❤️