MichaIng/DietPi #5219: OwnCloud installation failed on dependecies

Breeze2201, 2022-01-30:
#### Details:
- Date           | Sun Jan 30 02:07:44 CET 2022
- DietPi version | v7.9.4 (MichaIng/stretch)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- Hardware       | Odroid N2 (aarch64) (ID=15)
- Kernel version | `Linux Server-2 4.9.241-arm64 #1 SMP PREEMPT Thu Feb 25 18:56:07 CET 2021 aarch64 GNU/Linux`
- Distro         | stretch (ID=4)
- Command        | `a2enconf php7.3-fpm`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. Simply a new installation of the OwnCloud System
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
ERROR: Conf php7.3-fpm does not exist!
```

MichaIng, 2022-01-30:
Many thanks for your report.

Can you show the output of:
```sh
dpkg -l | grep php
dpkg -L php7.3-fpm
```

Note that your Debian version is not supported anymore. You may upgrade using the instructions given here: https://dietpi.com/docs/usage/

Or you can flash a very new image I uploaded today for testing, which uses mainline Linux 5.10: https://github.com/MichaIng/DietPi/issues/5039

Breeze2201, 2022-01-30:
dpkg -l | grep php :

ii  php-common                         2:92+0~20220117.43+debian9~1.gbpe0d14e            all          Common files for PHP packages
ii  php7.3-apcu                        5.1.21+4.0.11-7+0~20220117.31+debian9~1.gbp9a887a arm64        APC User Cache for PHP
ii  php7.3-cli                         7.3.33-1+0~20211119.91+debian9~1.gbp618351        arm64        command-line interpreter for the PHP scripting language
ii  php7.3-common                      7.3.33-1+0~20211119.91+debian9~1.gbp618351        arm64        documentation, examples and common module for PHP
ii  php7.3-curl                        7.3.33-1+0~20211119.91+debian9~1.gbp618351        arm64        CURL module for PHP
ii  php7.3-fpm                         7.3.33-1+0~20211119.91+debian9~1.gbp618351        arm64        server-side, HTML-embedded scripting language (FPM-CGI binary)
ii  php7.3-gd                          7.3.33-1+0~20211119.91+debian9~1.gbp618351        arm64        GD module for PHP
ii  php7.3-igbinary                    3.2.6+2.0.8-2+0~20211115.31+debian9~1.gbpfaf284   arm64        igbinary PHP serializer
ii  php7.3-intl                        7.3.33-1+0~20211119.91+debian9~1.gbp618351        arm64        Internationalisation module for PHP
ii  php7.3-json                        7.3.33-1+0~20211119.91+debian9~1.gbp618351        arm64        JSON module for PHP
ii  php7.3-mbstring                    7.3.33-1+0~20211119.91+debian9~1.gbp618351        arm64        MBSTRING module for PHP
ii  php7.3-mysql                       7.3.33-1+0~20211119.91+debian9~1.gbp618351        arm64        MySQL module for PHP
ii  php7.3-opcache                     7.3.33-1+0~20211119.91+debian9~1.gbp618351        arm64        Zend OpCache module for PHP
ii  php7.3-readline                    7.3.33-1+0~20211119.91+debian9~1.gbp618351        arm64        readline module for PHP
ii  php7.3-redis                       5.3.5+4.3.0-2+0~20211231.39+debian9~1.gbp134a26   arm64        PHP extension for interfacing with Redis
ii  php7.3-sqlite3                     7.3.33-1+0~20211119.91+debian9~1.gbp618351        arm64        SQLite3 module for PHP
ii  php7.3-xml                         7.3.33-1+0~20211119.91+debian9~1.gbp618351        arm64        DOM, SimpleXML, WDDX, XML, and XSL module for PHP
ii  php7.3-zip                         7.3.33-1+0~20211119.91+debian9~1.gbp618351        arm64        Zip module for PHP

dpkg -L php7.3-fpm :

/.
/etc
/etc/apache2
/etc/apache2/conf-available
/etc/apache2/conf-available/php7.3-fpm.conf
/etc/init.d
/etc/init.d/php7.3-fpm
/etc/logrotate.d
/etc/logrotate.d/php7.3-fpm
/etc/php
/etc/php/7.3
/etc/php/7.3/fpm
/etc/php/7.3/fpm/conf.d
/etc/php/7.3/fpm/php-fpm.conf
/etc/php/7.3/fpm/pool.d
/etc/php/7.3/fpm/pool.d/www.conf
/lib
/lib/systemd
/lib/systemd/system
/lib/systemd/system/php7.3-fpm.service
/usr
/usr/lib
/usr/lib/php
/usr/lib/php/7.3
/usr/lib/php/7.3/sapi
/usr/lib/php/7.3/sapi/fpm
/usr/lib/php/php7.3-fpm-reopenlogs
/usr/lib/tmpfiles.d
/usr/lib/tmpfiles.d/php7.3-fpm.conf
/usr/sbin
/usr/sbin/php-fpm7.3
/usr/share
/usr/share/bug
/usr/share/bug/php7.3-fpm
/usr/share/bug/php7.3-fpm/control
/usr/share/bug/php7.3-fpm/script
/usr/share/doc
/usr/share/lintian
/usr/share/lintian/overrides
/usr/share/lintian/overrides/php7.3-fpm
/usr/share/man
/usr/share/man/man8
/usr/share/man/man8/php-fpm7.3.8.gz
/usr/share/php
/usr/share/php/7.3
/usr/share/php/7.3/fpm
/usr/share/php/7.3/fpm/status.html
/usr/share/doc/php7.3-fpm

MichaIng, 2022-01-30:
Strange, `php7.3-fpm` has been successfully installed and the Apache config `/etc/apache2/conf-available/php7.3-fpm.conf` is there, is it? Can you check and repeat:
```sh
ls -l /etc/apache2/conf-available/php7.3-fpm.conf
a2enconf php7.3-fpm
```

Breeze2201, 2022-01-30:
I Think its on its place

-rw-r--r-- 1 root root 804 Jul  2  2021 /etc/apache2/conf-available/php7.3-fpm.conf

root@Server-1:~# a2enconf php7.3-fpm
-bash: a2enconf: command not found

MichaIng, 2022-01-30:
Even stranger. Your original error message `ERROR: Conf php7.3-fpm does not exist!` is coming from `a2enconf`, so the command existed, which also must have been true as otherwise `dietpi-software` wouldn't have tried to execute it. Now bash says the command does not exist 🤔. Can you check:
```sh
command -v a2enconf
ls -l /usr/sbin/a2enconf
echo $PATH
```

Breeze2201, 2022-01-30:
I have found that the http server is an Lighttpd not the Apache, Maybe this is the fault.

 [ ] 75   LASP: Apache  | SQLite  | PHP
│    [ ] 76   LAMP: Apache  | MariaDB | PHP
│    [ ] 78   LESP: Nginx    | SQLite  | PHP
│    [ ] 79   LEMP: Nginx    | MariaDB | PHP
│    [*] 81   LLSP: Lighttpd | SQLite  | PHP
│    [*] 82   LLMP: Lighttpd | MariaDB | PHP

MichaIng, 2022-01-30:
Hmm, obviously the Apache package is or was installed. Can you please run the above commands and additionally:
```sh
dpkg -l apache2 lighttpd
```

Breeze2201, 2022-01-30:
dpkg-query: no packages found matching apache2
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version          Architecture Description
+++-==============-================-============-===============================
=============
ii  lighttpd       1.4.53-4+deb10u2 arm64        fast webserver with minimal mem
ory footprint

MichaIng, 2022-01-30:
Okay strange, to sure, run the following, to assure no Apache package is installed anymore and repeat the ownCloud installation:
```sh
G_AGP 'apache2-*' 'libapache2-*'
dietpi-software install 47
```

Btw, to show commands and console logs nicely on GitHub (Markdown syntax), wrap them into code tags. Either mark all copied lines and hit the "Insert code" button `<>`  or add them manually, like:
````
```
line 1
line 2
```
````

Breeze2201, 2022-01-30:
Ah OK Sorry next time :)

Breeze2201, 2022-01-30:
I Think its now running...... Accesible only from LAN but i can fix this later.

Thank you very much......

MichaIng, 2022-01-30:
Okay, that is great.

closed 2022-01-30 by Breeze2201