MichaIng/DietPi #5223: DietPi-Software | Turn preference choices into ondemand choices

MichaIng, 2022-01-30:
I was thinking about how to solve the issue that the preference menus keep confusing users and have the following idea:
- We have a good dependency system now, so what about asking users interactively which webserver, desktop etc they want to install, when one is required? This could be done already when selecting a dependant via software list/search menu, to cover CLI as well after dependencies and conflicts have been resolved.
- The `dietpi.txt` preference settings could be shipped commented by default so that no preference is applied during first run setup when not actively done by users.
- Also the related `.installed` settings could be skipped as long as no interactive or `dietpi.txt` choice has been done.
- The preference menus could then be turned into install selections, at least for the desktops this would still be nice, addressing the idea of @StephanStS to make desktop installs easier.

What do you think?

Joulinar, 2022-01-31:
actually a good idea. It would help to reduce confusion, especially for experienced users.

StephanStS, 2022-01-31:
Some more thoughts about `dietpi.txt`.

In my opinion it would also help to clearify the dietpi.txt operation:
- Which entry is valid in case of doubled entries (e.g. two lines, one with `AUTO_SETUP_KEYBOARD_LAYOUT=gb` and one with `AUTO_SETUP_KEYBOARD_LAYOUT=de`)?  
  To have a clear message (e.g. "the first entry wins") would give a more clear handling.
- Alternatively to have the option for a kind of `dietpi_override.txt`, where only the entries to be changed by the user are present, would simplify the automated setup: Just edit this file with the few lines you need and the rest are the default values from `dietpi.txt`.  With this, we e.g. could give a small file for a desktop environment, like:

```
AUTO_SETUP_LOCALE=de_DE.UTF-8
AUTO_SETUP_KEYBOARD_LAYOUT=de
AUTO_SETUP_TIMEZONE=Europe/Berlin
AUTO_SETUP_NET_HOSTNAME=myDietPiSystem
AUTO_SETUP_BROWSER_INDEX=-2
AUTO_SETUP_AUTOSTART_TARGET_INDEX=16
AUTO_SETUP_AUTOMATED=1
AUTO_SETUP_INSTALL_SOFTWARE_ID=25
CONFIG_CHECK_DIETPI_UPDATES=1
CONFIG_CHECK_APT_UPDATES=2
```

Maybe this should be a different issue...

MichaIng, 2022-01-31:
> Maybe this should be a different issue...

Exactly 😉. I'll add some info to the top of the file. For now, basically all settings must exactly appear one time, since the override behaviour is not consistent: Most are checked with `sed ...;exit` so that only the first of multiple entries is effective, but in some cases we check via `grep -q`, so that only the presence of the setting with a specific value is relevant, hence any doubled entry is ineffective.

Joulinar, 2022-02-04:
yes more argument to have it changed. https://dietpi.com/phpbb/viewtopic.php?t=10023

closed 2022-04-24 by MichaIng