MichaIng/DietPi #5225: add cgroups to command line when installing docker

isarrider, 2022-01-31:
IMHO the flags
cgroup_enable=memory
and
cgroup_memory=1
should be added to the cmdline.txt
when installing docker
(so the memory can be monitored with docker stats)

#### Is your feature request related to a problem? Please describe:

cant view mem usage in docker without ;)

#### Describe the solution you'd like:

add feature flags to the commandline

#### Describe alternatives you've considered:

do it by hand ;)

#### Additional context

- ...

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2022-02-01:
Can you verify that `cgroup_memory=1` is redundant and/or has no effect anymore? The common argument is `cgroup_enable=<group>` and AFAIK the other one got accidentally added to RPi firmware/kernel and is now left for backwards compatibility even the the proper one is supported.

MichaIng, 2022-02-01:
Jep, `cgroup_enable=memory` is the correct syntax: https://github.com/raspberrypi/linux/blob/f9bd396/mm/memcontrol.c#L7115-L7128
`cgroup_memory=` is actually used to pass options to the memory cgroup and `1` is not a valid option, only `nosocket` and `nokmem` are: https://github.com/raspberrypi/linux/blob/f9bd396/mm/memcontrol.c#L7115-L7128

Armbian also passes `swapaccount=1` in case of Docker optimisation enabled, but it is the default in Linux, probably overwritten by Armbian but not in RPi kernel.

MichaIng, 2022-02-01:
Done with: https://github.com/MichaIng/DietPi/commit/e983c80

closed 2022-02-01 by MichaIng

isarrider, 2022-02-02:
many thanks to you & your research...
highly appreceated!

isarrider, 2022-02-08:
works - awesome...
maybe one thing - if people already have that in their commandline, dont append it.
(I took the double cgroup_enable=memory on my Pis today after upgrading to 8.1)...

MichaIng, 2022-02-08:
Ayy, there is a syntax error which causes this. Fixed with: https://github.com/MichaIng/DietPi/commit/a5951ef
Duplicates are removed with next DietPi update when present. Luckily they do not cause any issue.

isarrider, 2022-02-08:
I can only say Muchas Gracias again :)