MichaIng/DietPi #5227: DietPi-Update | Bootloader updates aren't included for Raspberry Pi?

xRiot, 2022-02-01:
#### Required Information
- DietPi version | `8.0.2`
- Distro version | `bullseye`
- Kernel version | `5.10.92-v8+`
- SBC model | `RPi 4 Model B (aarch64)`
- Power supply used | `15.3W 5.1V 3A Raspberry Pi official power supply`
- SDcard used | `SanDisk Extreme - 64GB - 160MB/s`

#### Additional Information (if applicable)
- Software title | `dietpi-update`
- Was the software title installed freshly or updated/migrated? `Included`
- Can this issue be replicated on a fresh installation of DietPi? `Yes`

#### Steps to reproduce
1. `$ dietpi-update`
2. `$ rpi-eeprom-update`

#### Expected behaviour
Shouldn't the bootloader be updated as a part of `dietpi-update` since it's a part of the system?

#### Actual behaviour
```shell
:~# dietpi-update

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
[  OK  ] Network time sync | Completed
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
[  OK  ] DietPi-Update | Got valid latest version: 8.0.2
[  OK  ] DietPi-Update | No update required, your DietPi installation is already up to date:
[ INFO ] DietPi-Update | Current version : v8.0.2
[ INFO ] DietPi-Update | Latest version  : v8.0.2
[ INFO ] DietPi-Update | Checking for new available live patches
[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 http://deb.xanmod.org releases InRelease
Hit:5 https://deb.debian.org/debian bullseye-backports InRelease
Hit:6 https://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | No APT upgrades were found, not creating file: /run/dietpi/.apt_updates

~# rpi-eeprom-update
*** UPDATE AVAILABLE ***
BOOTLOADER: update available
   CURRENT: Thu Apr 29 16:11:25 UTC 2021 (1619712685)
    LATEST: Tue Jan 25 14:30:41 UTC 2022 (1643121041)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
```

Joulinar, 2022-02-01:
No they are not part of our update. Usually `rpi-eeprom` updates are done outamatically during boot.

MichaIng, 2022-02-01:
Jep, the package is installed on first boot on all RPi 4 variants and the update service enabled which should update the EEPROM on boot automatically:
```sh
systemctl status rpi-eeprom-update
```

It is also possible to do it via `dietpi-config` > Advanced Options > Update RPi4 EEPROM firmware.

xRiot, 2022-02-01:
#### SOLVED
I changed `FIRMWARE_RELEASE_STATUS` from `default` to `latest` in `/etc/default/rpi-eeprom-update`, then I re-executed `rpi-eeprom-update` and lastly `reboot`.

---

@Joulinar This is strange. I actually rebooted my system after running `dietpi-update` and `rpi-eeprom-update` but the bootloader didn't update itself (maybe because your supposed to use `rpi-eeprom-update -a` to initialize it or perhaps because rpi-eeprom-update service wasn't running in the first place?).

@MichaIng I didn't actually get the chance to try that before successfully upgrading the firmware (`dietpi-config` > Advanced Options > Update RPi4 EEPROM firmware) but from the console logs it looks like a probable solution since it runs the right commands, thanks for sharing 👍 

Anyhow, this ticket can probably be closed unless you guys think otherwise 😃

Joulinar, 2022-02-01:
The setting for `rpi-eeprom-update` is the default one, provided directly from RPi devs. The value `default` means, only critical updates will be installed on your system. They will be released rarely. The version you had installed was `Thu Apr 29 16:11:25 UTC 2021`. This is the last `default` 

https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/release-notes.md#2021-04-30---update-default-version-to-2021-04-29

Now you switched to `latest`, which is the one from `Tue Jan 25 14:30:41 UTC 2022`

https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/release-notes.md#2022-01-25---create-new-release-from-2022-01-20---lateststable

closed 2022-02-01 by Joulinar

MichaIng, 2022-02-01:
The command I posted was meant to check whether the EEPROM update service really did run on boot as intended. But now we know the difference between "default" and "latest" EEPROM update and that by default only the "default"/critical updates are applied.