MichaIng/DietPi #5228: unable to install software using dietpi-software (ping command fails)

leesouthern, 2022-02-01:
#### Details:
- Date           | Tue Feb  1 15:36:40 EST 2022
- Bug report     | 3073119d-8acf-4a51-8861-b181c46eda48
- DietPi version | v7.9.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi Zero 2 W (armv7l) (ID=3)
- Kernel version | `Linux DietPi 5.10.92-v7+ #1514 SMP Mon Jan 17 17:36:39 GMT 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
` Command        | `ping -nc 1 -W 10 9.9.9.9
- Exit code      | 2
- Software title | DietPi-Software
#### Steps to reproduce:

1. Try to install software via dietpi-software after latest dietpi update (doesn’t seem to matter which software is selected)
#### Expected behaviour:

- selected software should be installed
#### Actual behaviour:

- the ping command always fails (also fails when selecting ‘retry’), but manually typing in the same ping command always works
#### Extra details:

 the following is the output when running the same command manually:

root@DietPi:/tmp/DietPi-Software# ping -nc 1 -W 10 9.9.9.9
PING 9.9.9.9 (9.9.9.9) 56(84) bytes of data.
64 bytes from 9.9.9.9: icmp_seq=1 ttl=47 time=120 ms

--- 9.9.9.9 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 119.951/119.951/119.951/0.000 ms

#### Additional logs:
```
: Name or service not known

Joulinar, 2022-02-01:
Hi,

can you try following. Do `nano /boot/dietpi.txt` and search for the following 2 lines and replace them as follows.

```
# General connection and DNS testing
# - IP to ping when checking network connectivity. Default: 1.1.1.1 (Cloudflare DNS, should be very fast world-wide)
CONFIG_CHECK_CONNECTION_IP=1.1.1.1
# - Domain to ping when checking DNS resolver. Default: one.one.one.one (Cloudflare DNS domain, see above)
CONFIG_CHECK_DNS_DOMAIN=one.one.one.one
```

Once done, try again

leesouthern, 2022-02-01:
Same issue (and again, there is no issue with the ping command when entered manually).

[FAILED] DietPi-Software | Checking network connectivity
 - Command: ping -nc 1 -W 10 1.1.1.1
root@DietPi:/tmp/DietPi-Software# ping -nc 1 -W 10 1.1.1.1
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=1 ttl=49 time=81.9 ms

--- 1.1.1.1 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 81.868/81.868/81.868/0.000 ms
root@DietPi:/tmp/DietPi-Software#

Joulinar, 2022-02-01:
let's try to cheat and set `CONFIG_CHECK_CONNECTION_IP=127.0.0.1`

MichaIng, 2022-02-01:
Another test:
```sh
G_CHECK_CON
G_CHECK_CON 9.9.9.9
```
`: Name or service not known` actually means an invalid/non-existing IP/host. Probably there is an unintended space somewhere 🤔.

leesouthern, 2022-02-01:
Setting “ CONFIG_CHECK_CONNECTION_IP=127.0.0.1” didn’t fix it.

G_CHECK_CON on its own fails, but it works with an IP address.

dietpi@DietPi:~$ G_CHECK_CON
[FAILED] Checking network connectivity
 - Command: ping -nc 1 -W 10 127.0.0.1
[FAILED] Unable to continue, command will now terminate.

dietpi@DietPi:~$ G_CHECK_CON 9.9.9.9
[  OK  ] Checking network connectivity

MichaIng, 2022-02-01:
Can you run those:
```sh
echo "\"$(sed -n '/^[[:blank:]]*CONFIG_CHECK_CONNECTION_IP=/{s/^[^=]*=//p;q}' /boot/dietpi.txt)\""
[[ $(sed -n '/^[[:blank:]]*CONFIG_CHECK_CONNECTION_IP=/{s/^[^=]*=//p;q}' /boot/dietpi.txt) == '127.0.0.1' ]] && echo same
```

leesouthern, 2022-02-01:
Note sure if I copied and pasted the above correctly, this is what I got:

dietpi@DietPi:~$ echo "\"$(sed -n '/^[[:blank:]]*CONFIG_CHECK_CONNECTION_IP=/{s/^[^=]*=//p;q}' /boot/dietpi.txt)\"" > [[ $(sed -n '/^[[:blank:]]*CONFIG_CHECK_CONNECTION_IP=/{s/^[^=]*=//p;q}' /boot/dietpi.txt) == '127.0.0.1' ]] && echo same
same

MichaIng, 2022-02-01:
Hehe it was two dedicated commands. I guess you created now a bunch of strange files in your current directory by forwarding the output with the additional `>`. At best copy&paste from GitHub into the SSH client. In PuTTY you can simply right-click to paste it into the console.

leesouthern, 2022-02-01:
Lol.  Does this look better:
dietpi@DietPi:~$ echo "\"$(sed -n '/^[[:blank:]]*CONFIG_CHECK_CONNECTION_IP=/{s/^[^=]*=//p;q}' /boot/dietpi.txt)\""
"9.9.9.9
dietpi@DietPi:~$ [[ $(sed -n '/^[[:blank:]]*CONFIG_CHECK_CONNECTION_IP=/{s/^[^=]*=//p;q}' /boot/diet

leesouthern, 2022-02-01:
Not what’s up with the strike through text above…

leesouthern, 2022-02-01:
FYI, I had put CONFIG_CHECK_CONECTION_IP back to 9.9.9.9 before running the above commands.

MichaIng, 2022-02-01:
Hmm, on the first output I'm missing the trailing `"`. Probably there is a strange control character. Windows style line endings shouldn't be an issue, though.

The second seems to be not complete:
```sh
[[ $(sed -n '/^[[:blank:]]*CONFIG_CHECK_CONNECTION_IP=/{s/^[^=]*=//p;q}' /boot/dietpi.txt) == '9.9.9.9' ]] && echo same
```

That `sed ... dietpi.txt` is used by `G_CHECK_CON` to get the configured IP. What I want to find out is whether this really is `9.9.9.9` (now after you changed it back) or whether some control character or other bit after the IP is corrupting it. As of the `echo` command, I added the two literal double quotes to check whether there is something unintended. Both double quotes should appear. The second command compares it to the intended IP, to check for possible differences which we may not see in the `echo` output.

> Not what’s up with the strike through text above…

What do you mean?

Joulinar, 2022-02-01:
> What do you mean?

the output on GitHub before I adjusted and added the markdown tags 😉

MichaIng, 2022-02-01:
However, before digging too deep here, try to recreate the line by changing the setting like this:
```sh
G_CONFIG_INJECT 'CONFIG_CHECK_CONNECTION_IP=' 'CONFIG_CHECK_CONNECTION_IP=1.1.1.1' /boot/dietpi.txt
```
Whatever is there, this rewrites the whole line, hence the check should then succeed:
```sh
G_CHECK_CON
```

I'll check whether Windows line endings are the issue.

> the output on GitHub before I adjusted and added the markdown tags 😉

Ah yes, best to wrap any pasted console output and commands into code fences:
````
```
line 1
line 2
```
````
Can be also achieved by selecting the pasted lines and hitting "Insert code" button on the GitHub editor `<>`.

leesouthern, 2022-02-01:
I fixed the problem by running dos2unix on dietpi.txt, G_CHECK_CON is now working without adding the IP address after it.  Thanks for pointing me in the right direction, I guess there must have been a weird character there after editing the file in Windows.  And I’ll try to work on my copy/pasting skills 😄

closed 2022-02-01 by leesouthern

MichaIng, 2022-02-01:
Jep, just verified, Windows line endings really are the issue here. Can be also fixed via:
```sh
sed -i 's/\r$//' /boot/dietpi.txt
```

Hmm, most tools, like wordpad, notepad and notepad++ preserve the existing UNIX line endings, so nowadays I'm not sure how they find their way into the file.

On UNIX, lines by default end with the `\n` control character only, on Windows its `\r\n` by default. This `\n` (not visible in editor) is kept when parsing with `grep`/`sed`/`awk` and obviously the `ping` command doesn't ignore it. Same with `curl`. Also not sure whether this can break the first login password and such. Probably we should run the above at least once on first boot, before the DietPi-FirstBoot script parses it and applies settings.

@leesouthern 
Was this a freshly flashed image? And do you know which editor on which OS you used to edit `dietpi.txt`?

leesouthern, 2022-02-01:
It was an image I flashed a couple of weeks ago, I had no problem installing software at that time.  I updated to the latest dietpi yesterday (the pi had been powered off in the mean time) then I ran into this issue.  I’m fairly sure I edited this file in notepad (Windows 10).

MichaIng, 2022-02-01:
Hmm, I guess you didn't edit `dietpi.txt` with notepad after booting the first time, did you? Then even more a mystery how it appeared 😄.

notepad on Windows 10 detects and preserves line ending format, same as character encoding. The status bar at the bottom right shows both, which should be `UNIX (LF)` and `UTF-8` when editing `dietpi.txt`, `config.txt` or any other text file.

So it remains a bid of a mystery how it happened. Good to keep it in mind however when a command which depends on a `dietpi.txt` settings fails due to invalid input. And in case this appears more often, we need to add `sed -i 's/\r$//' /boot/dietpi.txt` at some point.

Joulinar, 2022-02-01:
Hmm strange, your DietPi version above is shown as `v7.9.3` while we already on v8 as latest on. Could you try running `dietpi-update` to check.

leesouthern, 2022-02-02:
It’s now showing that v8.0.2 is available, I’m just about to do the upgrade.