MichaIng/DietPi #5233: OpenSSH instead of Dropbear

adamgreenberg07, 2022-02-03:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
brand new software installations on a raspberry pi 2 and odroid hc2. same issue experienced on both
- trying to enable openssh replaceing dropbear
- fresh install on both machines.
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`
- 8685ca75-39a1-4220-96d0-b98c56e38d82
- 50a6b127-d985-49e3-8570-468daa2e349f

#### Steps to reproduce

1. Installed a fresh version of dietpi. Installed Pihole, unbound wireguard
2. installed gravity sync
3. installed openssh 

#### Expected behaviour

-Open SSH should replace dropbear

#### Actual behaviour

- The change in program is failing

#### Extra details

- ...

Joulinar, 2022-02-03:
Did you used the preference system to switch SSH server? Or did you go via `search` software and just selected OpenSSH?

adamgreenberg07, 2022-02-03:
I used the preference system to switch between the two programs.

adamgreenberg07, 2022-02-03:
I decided to try a fresh install on one of my two devices and before I installed any other software, I installed OpenSSH. It worked.

closed 2022-02-03 by adamgreenberg07

MichaIng, 2022-02-03:
> I used the preference system to switch between the two programs.

Aside, it is not a "preference" menu but a "selection" in this case. Another reason to do #5223 🙂.

It shouldn't play a role whether the SSH server menu is used or OpenSSH marked via software list or search. When OpenSSH is installed, and any Dropbear package is installed, it is uninstalled automtically.

Interesting would be at which point with which error message the OpenSSH install failed. Actually it should always work as the package install does not depend on a successful service start (which expectedly fails while the port is still used by Dropbear) and when Dropbear is uninstalled, the possibly currently used SSH session stays active. Other installed software shouldn't play a role either.