MichaIng/DietPi #5234: Odroid C2 X Server  - does not start - failed to load module "fbturbo"

berni2a, 2022-02-03:
### Creating a bug report/issue

#### Required Information
- DietPi version | CORE 8 - 0 - 2
- Distro version | 6
- Kernel version | 5.10.81-meson64
- SBC model | 12
- Power supply used | (EG: 5V 2A
- SDcard used | (EG: SanDisk ultra 16 GB

#### Additional Information (if applicable)
- Software title | X Server
- Was the software title installed freshly 
- Can this issue be replicated on a fresh installation of DietPi? --> yes

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...I wanted to start MATE - Desktop - ended with error message
2.  /var/log/Xorg.0.log shows --> (EE) Failed to module "fbturbo"
2. ...

#### Expected behaviour

- ...Should start X - Server

#### Actual behaviour

- ...Server terminated 

#### Extra details

- ...

MichaIng, 2022-02-03:
Ah, I see, we still install the old xorg.conf, which has been fixed for DietPi v8.1 already. Please try:
```sh
rm /etc/X11/xorg.conf
```
Then the X server should start.

berni2a, 2022-02-03:
Hello Micha,

Sorry, still not working...
(EE) Cannot run in framebuffer mode. Please specify bus IDs for all framebuffer devices

BR Bernd

berni2a, 2022-02-03:
ls -l /dev/dri
insgesamt 0
drwxr-xr-x 2 root root        100  3. Feb 15:51 by-path
crw-rw---- 1 root video  226,   0  3. Feb 15:51 card0
crw-rw---- 1 root video  226,   1  3. Feb 15:51 card1
crw-rw---- 1 root render 226, 128  3. Feb 15:51 renderD128
----------------------------------------------------------------------

[    18.512]    Entry deleted from font path.
[    18.512] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    18.512]    Entry deleted from font path.
[    18.512] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
        built-ins
[    18.512] (==) ModulePath set to "/usr/lib/xorg/modules"
[    18.513] (**) Extension "DPMS" is disabled
[    18.513] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[    18.513] (II) Loader magic: 0xaaaab6741e28
[    18.513] (II) Module ABI versions:
[    18.513]    X.Org ANSI C Emulation: 0.4
[    18.513]    X.Org Video Driver: 24.1
[    18.513]    X.Org XInput driver : 24.1
[    18.513]    X.Org Server Extension : 10.0
[    18.515] (++) using VT number 1

[    18.523] (II) systemd-logind: took control of session /org/freedesktop/login1/session/_31
[    18.526] (II) xfree86: Adding drm device (/dev/dri/card1)
[    18.531] (II) systemd-logind: got fd for /dev/dri/card1 226:1 fd 11 paused 0
[    18.532] (II) xfree86: Adding drm device (/dev/dri/card0)
[    18.534] (II) systemd-logind: got fd for /dev/dri/card0 226:0 fd 12 paused 0
[    18.535] (II) no primary bus or device found
[    18.535]    falling back to /sys/devices/platform/soc/d0000000.apb/d00c0000.gpu/drm/card1
[    18.535] (II) LoadModule: "glx"
[    18.539] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    18.607] (II) Module glx: vendor="X.Org Foundation"
[    18.607]    compiled for 1.20.11, module version = 1.0.0
[    18.607]    ABI class: X.Org Server Extension, version 10.0
[    18.607] (==) Matched modesetting as autoconfigured driver 0
[    18.607] (==) Matched fbdev as autoconfigured driver 1
[    18.607] (==) Assigned the driver to the xf86ConfigLayout
[    18.607] (II) LoadModule: "modesetting"
[    18.607] (II) Loading /usr/lib/xorg/modules/drivers/modesetting_drv.so
[    18.613] (II) Module modesetting: vendor="X.Org Foundation"
[    18.613]    compiled for 1.20.11, module version = 1.20.11
[    18.613]    Module class: X.Org Video Driver
[    18.613]    ABI class: X.Org Video Driver, version 24.1
[    18.613] (II) LoadModule: "fbdev"
[    18.615] (WW) Warning, couldn't open module fbdev
[    18.615] (EE) Failed to load module "fbdev" (module does not exist, 0)
[    18.615] (II) modesetting: Driver for Modesetting Kernel Drivers: kms
[    18.615] (WW) Falling back to old probe method for modesetting
[    18.615] (II) modeset(0): using default device
[    18.615] (II) modeset(G0): using drv /dev/dri/card0
[    18.615] (EE)
Fatal server error:
[    18.615] (EE) Cannot run in framebuffer mode. Please specify busIDs        for all framebuffer devices
[    18.615] (EE)
[    18.615] (EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
[    18.615] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[    18.616] (EE)
[    18.643] (EE) Server terminated with error (1). Closing log file.

MichaIng, 2022-02-03:
Hmm, so KMS/DRM is supported, it successfully loads the modesetting module, the fbdev module fails to load expectedly. Not sure why it crashes even trying to run in framebuffer mode.

GLES is installed, right?
```sh
G_AGI libegl1 libgles2'
```
Are there any other X11 configs?
```sh
for i in /etc/X11/xorg.conf{,.d/*.conf}
do
echo "${i}:"
cat "$i"
done
```

berni2a, 2022-02-03:
G_AGI libegl1 libgles2' -- it installed something
-------------------------
second question is to hard for me - sorry

MichaIng, 2022-02-03:
Ah, the trailing single quote was not intended. I hope both packages did now install.

> second question is to hard for me - sorry

If you are on an SSH console, simply copy and paste the above code block into the SSH console and hit return. It loops through all X11 configurations and prints their content.

berni2a, 2022-02-03:
i am using PUTTY on my WINDOWS Computer and have NO idea where i shoud paste it? Sooory

Joulinar, 2022-02-03:
Directly into Putty

MichaIng, 2022-02-03:
Hit the copy button, which appears when you hover this code block:
```sh
for i in /etc/X11/xorg.conf{,.d/*.conf}
do
[[ -f $i ]] || continue
echo "${i}:"
cat "$i"
done
```
Then go to your PuTTY window and click the right mouse button, which pastes the copied code into the PuTTY terminal. Then hit return/enter to execute the command loop.

berni2a, 2022-02-03:
Oh - thank you so much - this makes LINUX  live much easier... :-) --> thats the outcome:

/etc/X11/xorg.conf.d/98-dietpi-disable_dpms.conf:
Section "Extensions"
    Option "DPMS" "Disable"
EndSection

Section "ServerFlags"
    Option "BlankTime" "0"
EndSection
root@DietPi:~#

berni2a, 2022-02-03:
i have installed it  but still not working :-(
see log-file
----------------------------------
[    45.361] (--) using VT number 2

[    45.361] (II) systemd-logind: logind integration requires -keeptty and -keeptty was no>
[    45.363] (II) xfree86: Adding drm device (/dev/dri/card1)
[    45.364] (II) xfree86: Adding drm device (/dev/dri/card0)
[    45.376] (II) no primary bus or device found
[    45.376]    falling back to /sys/devices/platform/soc/d0000000.apb/d00c0000.gpu/drm/ca>
[    45.376] (II) LoadModule: "glx"
[    45.381] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    45.437] (II) Module glx: vendor="X.Org Foundation"
[    45.437]    compiled for 1.20.11, module version = 1.0.0
[    45.437]    ABI class: X.Org Server Extension, version 10.0
[    45.437] (==) Matched modesetting as autoconfigured driver 0
[    45.437] (==) Matched fbdev as autoconfigured driver 1
[    45.437] (==) Assigned the driver to the xf86ConfigLayout
[    45.437] (II) LoadModule: "modesetting"
[    45.437] (II) Loading /usr/lib/xorg/modules/drivers/modesetting_drv.so
[    45.443] (II) Module modesetting: vendor="X.Org Foundation"
[    45.443]    compiled for 1.20.11, module version = 1.20.11
[    45.443]    Module class: X.Org Video Driver
[    45.443]    ABI class: X.Org Video Driver, version 24.1
[    45.443] (II) LoadModule: "fbdev"
[    45.443] (II) Loading /usr/lib/xorg/modules/drivers/fbdev_drv.so
[    45.445] (II) Module fbdev: vendor="X.Org Foundation"
[    45.446]    compiled for 1.20.0, module version = 0.5.0
[    45.446]    Module class: X.Org Video Driver
[    45.446]    ABI class: X.Org Video Driver, version 24.0
[    45.446] (II) modesetting: Driver for Modesetting Kernel Drivers: kms
[    45.446] (II) FBDEV: driver for framebuffer: fbdev
[    45.500] (WW) Falling back to old probe method for modesetting
[    45.509] (II) modeset(0): using default device
[    45.509] (WW) Falling back to old probe method for fbdev
[    45.509] (II) Loading sub module "fbdevhw"
[    45.509] (II) LoadModule: "fbdevhw"
[    45.511] (II) Loading /usr/lib/xorg/modules/libfbdevhw.so
[    45.514] (II) Module fbdevhw: vendor="X.Org Foundation"
[    45.514]    compiled for 1.20.11, module version = 0.0.2
[    45.514]    ABI class: X.Org Video Driver, version 24.1
[    45.515] (II) FBDEV(1): using default device
[    45.526] (II) modeset(G0): using drv /dev/dri/card0
[    45.526] (EE)
Fatal server error:
[    45.526] (EE) Cannot run in framebuffer mode. Please specify busIDs        for all fra>
[    45.526] (EE)
[    45.526] (EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
[    45.526] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional i>
[    45.527] (EE)
[    45.685] (EE) Server terminated with error (1). Closing log file.

MichaIng, 2022-02-04:
Okay, that is very strange. You do try to start it via `startx` as root user, right?

Another attempt would be to install the whole palette of possible DDX drivers, including the unaccelerated but very generic VESA driver:
```sh
apt install xserver-xorg-video-all
```

Also, while it should be all default, X11 could be explicitly configured to use modesetting:
```sh
cat << '_EOF_' > /etc/X11/xorg.conf/99-modesetting.conf
Section "Device"
    Identifier "Mali Graphics"
    Driver     "modesetting"
    Option     "AccelMethod"   "glamor"
EndSection
_EOF_
```

Also, since it is there, you can try to use the second DRI device by adding another line:
```
Option "kmsdev" "/dev/dri/card1"
```
to the section block of that file.

If it still does not work, would be interesting whether it does with the original Armbian image: https://dl.armbian.com/odroidc2/Bullseye_current
If this doesn't work either, it can be reported to the Armbian bug tracker: https://forum.armbian.com/forum/16-amlogic-s905x-s922x/

berni2a, 2022-02-10:
Today I updated  to version 8.1 - XSERVER still not starting. 

Thank you for your support so far.

MichaIng, 2022-02-10:
Have you tried the steps in my last post?

PD-Kerman, 2022-02-24:
Has the problem solved? I have the same problem and i wanted to start LXDE desktop on k2, and after installation i've got this on my terminal:
root@DietPi:~# startx

X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
Build Operating System: linux Debian
Current Operating System: Linux DietPi 5.10.81-meson64 #21.08.6 SMP PREEMPT Mon                                                                                                     Nov 22 11:21:51 UTC 2021 aarch64
Kernel command line: root=UUID=17e9cd02-f1e8-4d56-b60e-bc7cc23604c9 rootwait roo                                                                                                    tfstype=ext4 console=ttyAML0,115200 console=tty1 consoleblank=0 coherent_pool=2M                                                                                                     loglevel=4 ubootpart=2826a2a0-01 libata.force=noncq usb-storage.quirks=0x2537:0                                                                                                    x1066:u,0x2537:0x1068:u "net.ifnames=0"
Build Date: 16 December 2021  05:08:23PM
xorg-server 2:1.20.11-1+deb11u1 (https://www.debian.org/support)
Current version of pixman: 0.40.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Thu Feb 24 04:41:31 2022
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(EE)
Fatal server error:
(EE) Cannot run in framebuffer mode. Please specify busIDs        for all frameb                                                                                                    uffer devices
(EE)
(EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
(EE) Please also check the log file at "/var/log/Xorg.0.log" for additional info                                                                                                    rmation.
(EE)
(EE) Server terminated with error (1). Closing log file.
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error
root@DietPi:~# (EE) Cannot run in framebuffer mode. Please specify busIDs        for all framebuffer devices(EE) Cannot run in framebuffer mode. Please specify busIDs        for all framebuffer devicesdevices(EE) Cannot run in framebuffer mode. Please specify                                                                                                     xinit: giving upr all framebuffer devices
xinit: unable to connect to X server: Connection refused
xinit: server errornnect to X server: Connection refused
xinit: server error
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server errornnect to X server: Connection refused
xinit: server error
root@DietPi:~# (EE) Cannot run in framebuffer mode. Please specify busIDs                                                                                                            for all framebuffer devices(EE) Cannot run in framebuffer mode. Please specify                                                                                                     busIDs        for all framebuffer devices
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error

xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error

PD-Kerman, 2022-02-24:
and ive tried running lshw -c display and i cant see any thing about my display infomations , and then ive tried installing xserver-xorg-video-all on it but it didnt help

PD-Kerman, 2022-02-24:
it looks like we're having a graphic card issue,(im using NanoPi K2, OS:  DietPi v8.1.2, )(my lxde desktop was installed from dietpi-software), and ive been having this issue for a long time since i tried to install xfce4 on dietpi lastweek (ive tried both installation from dietpi-software and from apt, and ive got the same error)

MichaIng, 2022-02-24:
Please test this:
```sh
cat << '_EOF_' > /etc/X11/xorg.conf.d/02-dietpi-aml-s905.conf
Section "OutputClass"
    Identifier "Amlogic"
    MatchDriver "meson"
    Driver "modesetting"
    Option "PrimaryGPU" "true"
EndSection
Section "Screen"
    Identifier "Default Screen"
    Device "Meson"
    Monitor "foo"
    DefaultDepth 24
    SubSection "Display"
       Depth 24
       Modes "1920x1080" "1440x900" "1280x720" "1280x1024" "1280x960" "1024x768" "800x600" "640x480" "720x400"
    EndSubSection
EndSection
_EOF_
```

MichaIng, 2022-02-25:
Solution implemented with: https://github.com/MichaIng/DietPi/pull/5291/commits/572d1b3

PD-Kerman, 2022-02-26:
OK i'll test that as soon as i can, and thank you for your reply btw

PD-Kerman, 2022-02-26:
And One more issue by the way

PD-Kerman, 2022-02-26:
we can not reach raw.githubusercontent.com in some area especially in china and for that reason, we can neither update nor initialize the dietpi system and use dietpi-software some times, so i hope we can have a solution on this issue

PD-Kerman, 2022-02-26:
And hmmmm another question ,is it possible to provide a new image with installed desktop? it will be much more stable for us to use dietpi with desktop. Have a nice day!!!  :D

PD-Kerman, 2022-02-26:
And hmmmm another question ,is it possible to provide a new image with installed desktop? it will be much more stable for us to use dietpi images with installed desktop. Have a nice day!!! :D

Joulinar, 2022-02-26:
We are thinking to provide an image that contains a desktop ootb. But nothing we have a ETA for as it is just an idea. That you are not able to reach GitHub is a pity, as all our scripts are located there.

PD-Kerman, 2022-02-26:
it is a pity...... but uhmm its strange that i can reach GitHub website BUT i cannot pulling scripts from raw.githubusercontent.com, and ive asked all my friends and they cannot either

PD-Kerman, 2022-02-26:
it is a pity...... but uhmm its strange that i can reach GitHub website BUT i cannot pulling scripts from raw.githubusercontent.com, and ive asked all my friends here and they cannot either

PD-Kerman, 2022-02-26:
WOW so uhm dietpi has a plan on building an image contains desktop ? Thats sooooo GOOD

PD-Kerman, 2022-02-26:
WOW so uhm dietpi is planning to build an image contains desktop ? Thats sooooo GOOD

MichaIng, 2022-02-26:
> it is a pity...... but uhmm its strange that i can reach GitHub website BUT i cannot pulling scripts from raw.githubusercontent.com, and ive asked all my friends here and they cannot either

That is a pity indeed, and if it was actively wanted, it doesn't make any sense either as any content can be seen here on GitHub the same way, only raw plain-text file downloads (which one can otherwise copy&paste) is done via `raw.githubusercontent.com`. I remember it was on some ad blocklist for e.g. Pi-hole a while back, but that seems to have been sorted.

At time of writing, we do not plan do host the DietPi repository at another place, or a mirror, where downloads could be done from when `raw.githubusercontent.com` is for whatever reason not available. Not sure whether using a different DNS nameserver is an option? This can be also done for the DietPi system only when assigning a static IP, or manually (and in case of DHCP temporarily), e.g. using Quad9 DNS:
```
echo 'nameserver 9.9.9.9' > /etc/resolv.conf
```

closed 2022-02-26 by MichaIng

eevan78, 2023-08-11:
Yesterday I have tried to install LXDE on a freshly installed latest image for Odroid C2 and it didn't want to start. Upon finding this page and looking into commit 572d1b3 I spotted that because of the bad condition value in line 2725 it didn't install libegl1 and libgles2, and created the config file. After I did that manually, LXDE could start.

For my Odroid C2, the value of `G_HW_MODEL` is 12, and the condition in line 2725 reads:
```
elif (( $G_HW_ARCH < 10 && $G_HW_MODEL < 10 ))
			then
...
```

So it seems that the 10 in the second expression should be changed to 13.
But I haven't tried it.