MichaIng/DietPi #5235: Odroid C2 - ODROID Sound MODULE not working

berni2a, 2022-02-03:
### Creating a bug report/issue

#### Required Information
- DietPi version | v8.0.2
- Distro version | 6
- Kernel version | 
- SBC model | 12
- Power supply used | 5V 2A
- SDcard used | (EG: SanDisk ultra 16 GB

#### Additional Information (if applicable)
- Software title | Sound Config
- Was the software title installed freshly
- Can this issue be replicated on a fresh installation of DietPi? --> yes

- Bug report ID | 381f75d9-0651-4b34-8354-ae6c16b50785

#### Steps to reproduce

1. ...Wanted to change sound card to: odroid-hifishiel-plus
2. 
2. ...

#### Expected behaviour

- ...should change

#### Actual behaviour

- ...ended with error message:
- Command: modprobe snd-soc-pcm5103 snd-soc-odroid-dac
- Exit code: 1
- ...
- modprobe: FATAL: Module snd-soc-pcm5102 not found in directory /lib/modules/5.10.81-meson64
- 
- 

#### Extra details

- ...

MichaIng, 2022-02-03:
Many thanks for your report.

I feared that with the new mainline kernel image the required modules have changed. The same should be true for the `odroid-hifishield-2`, i.e.
```sh
modprobe aml_i2c snd-soc-pcm512x snd-soc-pcm512x-i2c snd-soc-odroid-dac2
```
Fails the same way, probably not for `aml_i2c` but for the sound drivers themselves.

We need to check which ones are required, or whether it is actually required to manually load any kernel driver. Can you try to proceed by using the "Change command" option from the error prompt and replace everything with a single colon `:`? Then see whether it does work.

Also, is the DAC/shield probably listed in the "Auto detection" section already?

MichaIng, 2022-02-03:
Also, regarding another topic, can you please paste the output of:
```sh
cat /proc/interrupts
```

berni2a, 2022-02-03:
root@DietPi:/var/log# cat /proc/interrupts
           CPU0       CPU1       CPU2       CPU3
  9:          0          0          0          0     GICv2  25 Level     vgic
 11:      15465      17064      15490      18509     GICv2  30 Level     arch_timer
 12:          0          0          0          0     GICv2  27 Level     kvm guest vtimer
 14:          0          0          0          0     GICv2 169 Level     arm-pmu
 15:          0          0          0          0     GICv2 170 Level     arm-pmu
 16:          0          0          0          0     GICv2 185 Level     arm-pmu
 17:          0          0          0          0     GICv2 186 Level     arm-pmu
 20:          0          0          0          0     GICv2  53 Edge      c1108500.i2c
 21:          2          0          0          0     GICv2 105 Edge      c1108680.adc
 23:          3          0          0          0     GICv2 225 Edge      ttyAML0
 24:          4          0          0          0     GICv2 228 Edge      c8100580.ir
 25:          0          0          0          0     GICv2  76 Edge      vdec
 26:          0          0          0          0     GICv2  64 Edge      esparserirq
 28:       1777          0          0          0     GICv2 241 Edge      platform_mhu_link
 29:       2275          0          0          0     GICv2 242 Edge      platform_mhu_link
 30:       1286          0          0          0     GICv2  40 Level     eth0
 31:       6790          0          0          0     GICv2 249 Edge      d0072000.mmc
 32:         13          0          0          0     GICv2 250 Edge      d0074000.mmc
 33:          0          0          0          0     GICv2 192 Level     gp
 34:          0          0          0          0     GICv2 193 Level     gpmmu
 35:          0          0          0          0     GICv2 194 Level     pp_bcast
 37:          0          0          0          0     GICv2 196 Level     pp0
 38:          0          0          0          0     GICv2 197 Level     ppmmu0
 39:          0          0          0          0     GICv2 198 Level     pp1
 40:          0          0          0          0     GICv2 199 Level     ppmmu1
 41:          0          0          0          0     GICv2 200 Level     pp2
 42:          0          0          0          0     GICv2 201 Level     ppmmu2
 43:       1064          0          0          0     GICv2  35 Edge      meson
 44:        257          0          0          0     GICv2  89 Edge      dw_hdmi_top_irq, c883a000.hdmi-tx
 45:   14561864          0          0          0     GICv2  63 Level     c9100000.usb, dwc2_hsotg:usb1
 47:          2          0          0          0  meson-gpio-irqchip  29 Level     stmmac-0:00
IPI0:      3606       7700       7728       3268       Rescheduling interrupts
IPI1:      1219        432        449        329       Function call interrupts
IPI2:         0          0          0          0       CPU stop interrupts
IPI3:         0          0          0          0       CPU stop (for crash dump) interrupts
IPI4:         0          0          0          0       Timer broadcast interrupts
IPI5:       393        215        213        316       IRQ work interrupts
IPI6:         0          0          0          0       CPU wake-up interrupts
Err:          0

MichaIng, 2022-02-03:
Okay great, so another obsolete step we could remove on mainline kernel images when selecting "usb-dac" as sound card: https://github.com/MichaIng/DietPi/commit/ad6dbde

What about the [other questions](https://github.com/MichaIng/DietPi/issues/5235#issuecomment-1029077138)?

berni2a, 2022-02-03:
We need to check which ones are required, or whether it is actually required to manually load any kernel driver. Can you try to proceed by using the "Change command" option from the error prompt and replace everything with a single colon :? Then see whether it does work.
--> it changes the sounf card -> but there is still no sound

MichaIng, 2022-02-03:
Can you show:
```sh
aplay -L
```
Can check whether in `alsamixer` there is some volume bar which you can raise.

berni2a, 2022-02-03:
root@DietPi:~# aplay -l
**** Liste der Hardware-Geräte (PLAYBACK) ****
Karte 0: GXSOUNDCARD [GX-SOUND-CARD], Gerät 0: fe.dai-link-0 (*) []
  Sub-Geräte: 1/1
  Sub-Gerät #0: subdevice #0
root@DietPi:~#
-----------------------------------------------
alsa-mixer does not start with --> odroid-hifishield-plus

berni2a, 2022-02-03:
when i try an USB-DAC it works...

MichaIng, 2022-02-03:
Okay, so what is shown above seems to be the onboard sound card. Would be also interesting whether this works OOTB. On Odroid N2 it was quite complicated to configure it: https://github.com/MichaIng/DietPi/commit/b575b52#diff-e1f2e63b74209db8768bc28025d8727d20f2a2fa96ef44a0db7991724abeb507R2089

Btw, `aplay -l` output does not change, regardless whether you have the should plugged or not, right?

USB DACs usually work OOTB since the generic kernel driver works, at least with all basic functionality. The HiFi Shield Plus is connected via I2S header. So the header itself needs to work and probably a driver is required. The official instructions refer to the legacy kernel only (which is so old that it cannot be used with recent distro versions): https://wiki.odroid.com/accessory/sound/hifi_shield

The driver sources are [here](https://github.com/hardkernel/linux/blob/odroidc2-3.14.y/sound/soc/aml/m8/odroid_dac.c) but depend on the Amlogic I2C kernel module, but it surely won't work with mainline Linux. I will ask at the Armbian forum whether someone knows whether/how to get the Odroid HiFi shields working.

I would have tried to contact Hardkernel as well, but HiFi Shield Plus is for C1/C2 with are both EOL. A pain that SBC manufacturers (probably forced by SoC manufacturers, like Amlogic here) maintain their only, often single Linux version with drivers instead of working on support for mainline Linux. This single kernel version is then unusable after some years, so that the SBC cannot be maintained anymore with full features...

berni2a, 2022-02-03:
Btw, aplay -l output does not change, regardless whether you have the HiFi Shield plugged or not, right?
------------------------------
aplay -l shows build in sound card = hdmi & USB-dac but no Hifi Shield plus

By the way - the sound card works fine with VOLUMIO - but may have the former LINUX kernel used

MichaIng, 2022-02-03:
Jep, Volumio uses the legacy Hardkernel kernel. For audio only this is fine, for other network and server tasks, Docker and other things, it's not really usable anymore.

Okay, so I2S (the header/pins) in general and the Odroid HiFi Shields are something to check back on the Armbian forum. Until then, I'm afraid you either need to keep using onboard audio or USB DAC, or use Volumio.

berni2a, 2022-02-03:
the odroid-sound-card works also with the HARDKERNEL - Ubuntu MATE 20.04 desktop image 
but it's okay if it wont work with DIETPI
-----------------------------------------------------------------------------------
By the way is there a possibility to get the latest 7.x revision for ODROID C2

MichaIng, 2022-02-04:
> the odroid-sound-card works also with the HARDKERNEL - Ubuntu MATE 20.04 desktop image

All Hardkernel provided images use the legacy Hardkernel kernel, so yes it is supposed to work there.

> By the way is there a possibility to get the latest 7.x revision for ODROID C2

Do you mean a PCB (hardware) revision or DietPi version? The first of course cannot be upgraded, if you mean the second: The DietPi version is not tied to the kernel/firmware stack. We intentionally replaced the legacy kernel image since it does not work anymore with several software implementations and system parts which expect newer Linux features to be present, so we do not offer an image with the old kernel anymore. You can however create one yourself, starting with the Buster image provided by Meveric: https://oph.mdrjr.net/meveric/images/Buster/
Then running our conversion script on it: https://dietpi.com/docs/hardware/#make-your-own-distribution

MichaIng, 2022-02-05:
I disabled the HiFi Shield sound card selection for now, when the related kernel modules are not available: https://github.com/MichaIng/DietPi/commit/ac45a2d

On FriendlyARM SBCs (NanoPi), I2S indeed needs to be enabled via (custom) device tree overlay first. I can imagine that it is the same for Odroid C2: https://forum.armbian.com/topic/9009-info-friendlyarm-pcm5102a-hat-with-nanopi-neo-under-mainline-4xx-and-dev-5xx/

Need to ask at the Armbian forum: https://forum.armbian.com/topic/8761-odroid-c2-general/page/2/#comment-125553

As a first step, could you check whether an I2S device is generally available in the tree and whether its enabled or not?
```
for i in /proc/device-tree/soc/*i2s*
do
[[ -d $i ]] || continue
echo -n "${i}: "
cat "$i/status"
echo
done
```

berni2a, 2022-02-10:
Sorry I have some issues with the starting behavior of DIETPI after version 8.
Today I updated to version 8.1.
--------------------------------------------------------------------------------------------
When I start your code - there is NO echo.

berni2a, 2022-02-10:
I am sorry, could not create a working BUSTER image with the link you have provided. 
This might be to difficult for me. :-)

MichaIng, 2022-02-10:
What do you mean with "issue" and "string behaviour"? Our current Odroid C2 image did never support the HiFi Shields out of the box, but I2S needs to be enabled first. This is independent of the DietPi version but a question of the kernel version only.

Hmm, when there is no I2S at all in the device tree, then we indeed have an issue. Does the following return something?
```
for i in /proc/device-tree/__symbols__/*i2s*
do
[[ -e $i ]] || continue
echo -n "${i}: "
cat "$i"
echo
done
```

egon0, 2022-07-31:
> Hmm, when there is no I2S at all in the device tree, then we indeed have an issue. Does the following return something?

```/proc/device-tree/__symbols__/i2s_am_clk_pins: /soc/bus@c8100000/pinctrl@14/i2s_am_clk
/proc/device-tree/__symbols__/i2s_out_ao_clk_pins: /soc/bus@c8100000/pinctrl@14/i2s_out_ao_clk
/proc/device-tree/__symbols__/i2s_out_ch01_ao_pins: /soc/bus@c8100000/pinctrl@14/i2s_out_ch01_ao
/proc/device-tree/__symbols__/i2s_out_ch23_ao_pins: /soc/bus@c8100000/pinctrl@14/i2s_out_ch23_ao
/proc/device-tree/__symbols__/i2s_out_ch45_ao_pins: /soc/bus@c8100000/pinctrl@14/i2s_out_ch45_ao
/proc/device-tree/__symbols__/i2s_out_lr_clk_pins: /soc/bus@c8100000/pinctrl@14/i2s_out_lr_clk
/proc/device-tree/__symbols__/i2sout_ch23_y_pins: /soc/bus@c8834000/pinctrl@4b0/i2sout_ch23_y
/proc/device-tree/__symbols__/i2sout_ch45_y_pins: /soc/bus@c8834000/pinctrl@4b0/i2sout_ch45_y
/proc/device-tree/__symbols__/i2sout_ch67_y_pins: /soc/bus@c8834000/pinctrl@4b0/i2sout_ch67_y
```

MichaIng, 2022-09-18:
Sorry for the late replay. Do those exist?
```sh
cat /proc/device-tree/soc/bus@c8100000/pinctrl@14/status
cat /proc/device-tree/soc/bus@c8834000/pinctrl@4b0/status
```

nbenn, 2022-09-25:
I'm in the same situation as @egon0 (apart from running v8.9.2). For me, `status` files do not exist, all I have is

```sh
> ls /proc/device-tree/soc/bus@c8100000/pinctrl@14/
'#address-cells'   ee_cec	     i2s_out_ch23_ao   pwm_ao_a_12   remote_input_ao	 uart_ao_b
'#size-cells'	   i2c_ao	     i2s_out_ch45_ao   pwm_ao_a_3    spdif_out_ao_13	 uart_ao_b_cts_rts
 ao_cec		   i2s_am_clk	     i2s_out_lr_clk    pwm_ao_a_6    spdif_out_ao_6
 bank@14	   i2s_out_ao_clk    name	       pwm_ao_b      uart_ao_a
 compatible	   i2s_out_ch01_ao   phandle	       ranges	     uart_ao_a_cts_rts
> ls /proc/device-tree/soc/bus@c8834000/pinctrl@4b0/
'#address-cells'   eth-rgmii   i2sout_ch23_y   pwm_a_y	 sdcard		   spi-ss0
'#size-cells'	   eth-rmii    i2sout_ch45_y   pwm_b	 sdcard_clk_gate   uart_a
 bank@4b0	   hdmi_hpd    i2sout_ch67_y   pwm_d	 sdio		   uart_a_cts_rts
 compatible	   hdmi_i2c    name	       pwm_e	 sdio_clk_gate	   uart_b
 emmc		   i2c_a       nor	       pwm_f_x	 sdio_irq	   uart_b_cts_rts
 emmc-ds	   i2c_b       phandle	       pwm_f_y	 spdif_out_y	   uart_c
 emmc_clk_gate	   i2c_c       pwm_a_x	       ranges	 spi-pins	   uart_c_cts_rts
```

MichaIng, 2022-11-21:
I assume I2S is enabled by default then. Does the following kernel module exist?
```sh
modinfo snd-soc-pcm5102a
```
It does at least in edge kernel. With some luck it is the missing part for the HiFi Shield to work. To test:
```sh
apt install linux-{image,dtb}-edge-meson64 linux-u-boot-odroidc1-edge
reboot
```
But at best do a full system drive backup/clone, just in case edge kernel has serious issues. If it works:
```sh
modprobe snd-soc-pcm5102a
apt purge linux-{image,dtb}-current-meson64 linux-u-boot-odroidc1-current
```

MichaIng, 2022-12-04:
Same with HiFi Shield 2: https://dietpi.com/forum/t/fatal-module-aml-i2c-not-found-in-directory-lib-modules-5-19-17-meson64/15154

Gialt, 2022-12-10:
> Sorry for the late replay. Do those exist?
> 
> ```shell
> cat /proc/device-tree/soc/bus@c8100000/pinctrl@14/status
> cat /proc/device-tree/soc/bus@c8834000/pinctrl@4b0/status
> ```

2x No such file or directory.

Gialt, 2022-12-10:
I did everything listed above.
`aplay -L`
now results in:
```
null
    Discard all samples (playback) or generate zero samples (capture)
lavrate
    Rate Converter Plugin Using Libav/FFmpeg Library
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
hw:CARD=ODROIDC2,DEV=0
    ODROID-C2, 
    Direct hardware device without any conversions
plughw:CARD=ODROIDC2,DEV=0
    ODROID-C2, 
    Hardware device with all software conversions
default:CARD=ODROIDC2
    ODROID-C2, 
    Default Audio Device
sysdefault:CARD=ODROIDC2
    ODROID-C2, 
    Default Audio Device
dmix:CARD=ODROIDC2,DEV=0
    ODROID-C2, 
    Direct sample mixing device
usbstream:CARD=ODROIDC2
    ODROID-C2
    USB Stream Output

```

Gialt, 2022-12-10:
Post @ Hardkernel Odroid forum:
https://forum.odroid.com/viewtopic.php?f=201&t=45886

MichaIng, 2022-12-10:
Please run and paste the output of the following commands:
```sh
modprobe snd-soc-pcm5102a
aplay -l
```

Gialt, 2022-12-10:
```
**** List of PLAYBACK Hardware Devices ****
card 0: ODROIDC2 [ODROID-C2], device 0: fe.dai-link-0 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

MichaIng, 2022-12-11:
Doubles the forum post, but this would be next test:
```sh
modprobe snd-soc-pcm512x snd-soc-pcm512x-i2c
echo pcm5242 0x4c > /sys/class/i2c-adapter/i2c-1/new_device
amixer -c 1 set Analogue 0
amixer -c 1 set Digital 201
amixer -c 1 set Deemphasis on
aplay -l
amixer -c 1 controls
```

MichaIng, 2026-11-09:
Mainline Linux compatible overlays added with https://github.com/MichaIng/build/commit/17b940c.
Build running: https://github.com/MichaIng/DietPi/actions/runs/42658823648
Once done, it can be tested like this:
```sh
G_DEV_TEST_FIRMWARE
G_CONFIG_INJECT 'overlays=' 'overlays=odroidc2-hifishield2' /boot/dietpiEnv.txt
```
then reboot, and check in `dietpi-config` audio options whether it is listed below "auto-detect". Kernel modules should load automatically.

MichaIng, 2026-11-10:
Closing as done, as OP is probably absent in the meantime.

closed 2026-11-10 by MichaIng