MichaIng/DietPi #5238: Dietpi-drive_manager doesn't work after upgrading to 8.1.2 master (RPi3B+)

vplanida, 2022-02-06:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.1.2, master, MichaIng
- Distro version | bullseye 0
- Kernel version | Linux MiniEidrw 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux
- SBC model | RPi 3 Model B (aarch64)
- Power supply used | (Official 12.5W 2.5A)
- SD card used | SanDisk Extreme 128GB

#### Additional Information (if applicable)
- Software title | dietpi-drive_manager
- Updated with distro

#### Steps to reproduce

1. type sudo dietpi-drive_manager in the console

#### Expected behaviour

- Drive manager script starts with drive manager menu showing

#### Actual behaviour

- And it doesn't. Terminal stays on bash screen, user@machine disappears. Terminal is unresponsive to ctrl+c or ctrl+z, but typing in htop does bring up the process manager and restores bash back to working conditions.

#### Extra details

- Drive_manager did work back on 8.0.2 version

Joulinar, 2022-02-06:
Do you have any USB disk attached or did you removed one? Can you show following

```
cat /etc/fstab
df -h
lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid
```

vplanida, 2022-02-06:
> Do you have any USB disk attached or did you removed one? Can you show following

Not at the time of trying it

> cat /etc/fstab

`# You can use "dietpi-drive_manager" to setup mounts.
# NB: It overwrites and re-creates physical drive mount entries on use.
#----------------------------------------------------------------
# NETWORK
#----------------------------------------------------------------

#----------------------------------------------------------------
# TMPFS
#----------------------------------------------------------------
tmpfs /tmp tmpfs size=1024M,noatime,lazytime,nodev,nosuid,mode=1777
tmpfs /var/log tmpfs size=50M,noatime,lazytime,nodev,nosuid,mode=1777

#----------------------------------------------------------------
# MISC: ecryptfs, vboxsf (VirtualBox shared folder), gluster, bind mounts
#----------------------------------------------------------------

#----------------------------------------------------------------
# SWAP SPACE
#----------------------------------------------------------------
/var/swap none swap sw

#----------------------------------------------------------------
# PHYSICAL DRIVES
#----------------------------------------------------------------
PARTUUID=0140a7f0-02 / ext4 noatime,lazytime,rw 0 1
PARTUUID=0140a7f0-01 /boot vfat noatime,lazytime,rw 0 2
UUID=86FE87BDFE87A451 /mnt/86FE87BDFE87A451 ntfs noatime,lazytime,rw,permissions,big_writes,nofail,noauto,x-systemd.automount`

> df -h
`Filesystem      Size  Used Avail Use% Mounted on
/dev/root       118G  6.9G  106G   7% /
devtmpfs        447M     0  447M   0% /dev
tmpfs           480M     0  480M   0% /dev/shm
tmpfs           192M  2.9M  190M   2% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           1.0G     0  1.0G   0% /tmp
tmpfs            50M     0   50M   0% /var/log
/dev/mmcblk0p1  127M   32M   95M  25% /boot`

> lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid
`NAME        FSTYPE LABEL   SIZE RO TYPE MOUNTPOINT PARTUUID                             UUID
mmcblk0                  119.1G  0 disk
├─mmcblk0p1 vfat           128M  0 part /boot      0140a7f0-01                          E547-094A
└─mmcblk0p2 ext4           119G  0 part /          0140a7f0-02                          196af0eb-3cdf-4d16-a819-6d064fbd60ff
`

Okay, inserted the USB flashdrive and now it has launched effortlessly.
It appear to me, that I forgot to unmount the USB drive last time I used it, witch resolved in this issue.

Joulinar, 2022-02-06:
Yes looks like. I guess issue can be closed?

vplanida, 2022-02-06:
I mean, if issue is well known and the behavior is expected, I do believe so, yes
Thank you for looking in to it!

closed 2022-02-06 by vplanida

Joulinar, 2022-02-06:
It's a little bit expected as the drive manager is trying to mount the disk. But as the disk is not present, it's going to hang until it timed out.

MichaIng, 2022-02-06:
That's the downside of systemd.automount: It blocks the command schedule for 90 seconds timeout on mount point access while a regular mount attempt would fail immediately when the drive is not present.

I'm actually wondering whether it causes more issues than it solves. E.g. I have a docking station with two HDDs attached via USB to an RPi and they take quite some seconds on boot to spin up. But I used just `auto` here and they mount successfully. For network drives, the network-online.target should be sufficiently reliable to not attempt mounting them before the network is really up.

Probably we should add a switch. Only problem that for this we'd need to practically store the choice in a dedicated file, at least when the current method of forcefully recreating the fstab entries is kept.

However, for now drive manager does not support temporary mounts gracefully, but they need to be unmounted explicitly before being removed. For a temporary mount, I use `mount /dev/sdX1 /mnt/external` here, just from console, since an fstab entry is overkill.