MichaIng/DietPi #5242: dietpi-update | DNS resolver test does not work with 1.1.1.1 blocked, use DNS set in dietpi-config instead?

GvY85, 2022-02-06:
I am using Pi-hole and to prevent devices from using hardcoded DNS servers I blocked `8.8.8.8, 8.8.4.4, 1.1.1.1 and 1.0.0.1` in my router only to find out that `dietpi-update` does a `ping `to `1.1.1.1` to check internet connectivity.

```
Checking network connectivity                                                                                        │
                                                          │  - Command: ping -nc 1 -W 5 1.1.1.1                                                                                  │
                                                          │  - Exit code: 1                                                                                                      │
                                                          │  - DietPi version: v8.1.2 (MichaIng/master) | HW_MODEL: 21 | HW_ARCH: 10 | DISTRO: 6                                 │
                                                          │  - Image creator: DietPi Core Team                                                                                   │
                                                          │  - Pre-image: Debian mini.iso                                                                                        │
                                                          │  - Error log:                                                                                                        │
                                                          │ PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.                                                                         │
                                                          │ From 192.168.178.1 icmp_seq=1 Destination Host Unreachable                                                           │
                                                          │                                                                                                                      │
                                                          │ --- 1.1.1.1 ping statistics ---                                                                                      │
                                                          │ 1 packets transmitted, 0 received, +1 errors, 100% packet loss, time 0ms  
```

Perhaps it is better to have it use the system DNS server set by `dietpi-config` instead of a hardcoded ip? 

If I select `Change command` and change `1.1.1.1` to the ip of my pi-hole, router or another public DNS that is not blocked it still does not work for some reason..

Joulinar, 2022-02-06:
This check is not hardcode. DNS as well as IP check can be configured via `/boot/dietpi.txt`

MichaIng, 2022-02-06:
Many thanks for your report.

It uses the IP from `dietpi.txt`: https://github.com/MichaIng/DietPi/blob/master/dietpi.txt#L180

Lol doubled post.

We need to add this to network settings dialogs, probably an additional entry to change this right from the G_CHECK_CON/G_CHECK_DNS error prompts.

Joulinar, 2022-02-06:
You mean some special error handle for the connection/DNS check?

GvY85, 2022-02-06:
Thanks for the quick reply :).

Learned something new. But this is never asked during installation or initial configuration right? So why not use the DNS from dietpi-config or make settings like these configurable via a menu (dietpi-config)?

Joulinar, 2022-02-06:
It's a value set in configuration file `dietpi.txt`. Aim is to verify real internet connection. Means using a local DNS check against the router, which might be assigned via DHCP, would undermine the purpose of this check.

GvY85, 2022-02-06:
I agree that setting the routers ip is a dirty hack to bypass the check and undermine the purpose but perhaps this parameter can be made editable via `dietpi-config` instead of "hidden" away in `dietpi.txt`? That way users can set there own real internet ip and address if they want to.

So not a bug but an improvement request :)

MichaIng, 2022-02-06:
> You mean some special error handle for the connection/DNS check?

Yes, generally G_EXEC allows this: https://github.com/MichaIng/DietPi/blob/d4dc959/dietpi/func/dietpi-globals#L931-L932
We just don't make use of it yet 😅.

> Learned something new. But this is never asked during installation or initial configuration right? So why not use the DNS from dietpi-config or make settings like these configurable via a menu (dietpi-config)?

No, indeed, it is not somewhere in the dialogs yet. It should be. The IP and hostname used for the checks are however not in any way related, nor should they be, to the systems nameserver IP. A DNS provider's IP and hostname is only used since those are usually anycast IPs with many servers around the global, the quickest/closed one responding. So those are the most reliable ones as defaults for doing Internet connection and DNS resolver functionality. With the second test, the system's/dietpi-config configured upstream DNS is tested, but by asking him to resolve another hostname (not its own), which is only for mentioned reason an upstream DNS hostname as well 😄. When adding those options to dietpi-config, we should make that point clear: The only thing which is important when choosing IP and hostname for those checks is that those are guaranteed to be responding, that they are remote/WAN IP/hostname (since we want to check Internet connectivity) and that they are at best responding quickly, of course.