MichaIng/DietPi #5245: Raspberry Pi 4 | dietpi-firstrun fails | "python-pip no install candidate"

eyduh, 2022-02-07:
Hoi,

trying to setup a fresh install on a raspberry pi using the automation, I selected a bunch of tools I would like on there - among others python 3. I guess the package naming has changed since bullseye as it won't work? It's weird cause I could install python and pip from dietpi-software on the previous install before wiping it.

Other stuff that fails to install during first run skips itself and waits for reboot (Noticed docker did this in the install due to kernel update)

#### Details:
- Date           | Mon Feb  7 12:32:31 CET 2022
- DietPi version | v8.1.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux adaprints0 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -qq --allow-change-held-packages install motion v4l-utils curl python-pip python-dev gcc libssl-dev libcurl4-openssl-dev libjpeg-dev zlib1g-dev`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Package 'python-pip' has no installation candidate
```
---------------------------------------------------------------------

Joulinar, 2022-02-07:
I guess you are trying to install MotionEye correct? This is not working on Bullseye, since it did not support Python 3 and probably will not, because the maintainer is not continue working on the project.

eyduh, 2022-02-07:
Oh, yeah I was also going to install motioneye - but only because mjpg-streamer wasnt working for some reason.

The end goal of this pi is to run octoprint+klipper with a CSI camera monitoring prints.

I'll try again without motioneye & report back

MichaIng, 2022-02-07:
How did you try to install MotionEye? It should be disabled and deselected automatically when trying to install it on Bullseye and shouldn't be visible in the list 🤔.

If mjpg-streamer isn't working, then it wouldn't work in OctoPrint either, since it uses external camera IP streams (and we offer mjpg-streamer as combination) as well. Let's debug why mjpg-streamer isn't working, then.

Joulinar, 2022-02-07:
indeed should be disabled already 🤔 

```
root@DietPi4:~# dietpi-software list | grep Moti
ID 136 | =0 | MotionEye: Web interface & surveillance for your camera | +FFmpeg DISABLED for Debian bullseye | https://dietpi.com/docs/software/camera/#motioneye
root@DietPi4:~#
```

MichaIng, 2022-02-07:
I just recognised that software installs not supported on arch, hardware model or Debian version are not skipped when installing via automated first run installs. This was checked only for CLI or menu installs: https://github.com/MichaIng/DietPi/commit/b6ade21

Interesting that this was never recognised before, was like that since a long time 😅. Fixed with: https://github.com/MichaIng/DietPi/commit/b6ade21

eyduh, 2022-02-08:
So yeah, I swapped the motioneye auto-install for mjpg-streamer and the install worked fine - apart from some other weird quirks but those are unrelated to this so closing this issue.

closed 2022-02-08 by eyduh

eyduh, 2022-02-08:
> Let's also debug why mjpg-streamer isn't working.

After the finished install & auto-setup it worked so must've been something weird with the previous install c:

MichaIng, 2022-02-08:
Okay great.