MichaIng/DietPi #5246: Image | RADXA Rock(Pi?) 5

docgalaxyblock, 2022-02-07:
#### Formal device information
- Device name | Rock 5
- Official product URL | https://wiki.radxa.com/Rock5
- GitHub resource URL (not yet avaible) | https://github.com/radxa/
- Image download URL | https://wiki.radxa.com/Rock5/downloads (not yet avaible)

#### Is the SBC officially supported by the Debian installer?
- Seems like not

#### If not, is a reliable 3rd party Debian image available for this SBC?
- Radxa should provide one at there [Wiki](https://wiki.radxa.com/Rock5/downloads)

#### My Wishes
 - Hope you can develop a good stable image like your rockpi images 😉
 - Hope you will get a dev sample eg from [here](https://shop.allnetchina.cn/collections/frontpage/products/radxa-zero), ask the shop and maybe you get one 🙏
 
 I missed the preorder start ... hope to get one to move some homelab services.
 
 I hope the Dev team has some time 🙃
 
 Greetings from Germany 😃

Joulinar, 2022-02-07:
Just out of curiosity, did you tried the Zero image?

https://dietpi.com/downloads/images/DietPi_RadxaZero-ARMv8-Bullseye.7z

docgalaxyblock, 2022-02-07:
@Joulinar
I guess it will not work due the Zero is Armlogic S905 based and the Rock5 has a RK3588 
And I do not have one yet 😉

MichaIng, 2022-02-07:
Jep, requires a new image, similar to Quartz64. I'll ask for a developer sample.