MichaIng/DietPi #5250: Issue with default graphics controller on VirtualBox image

Camry2731, 2022-02-07:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.1.2
- Distro version | Bullseye
- Kernel version | 5.10.0-9
- SBC model | Virtual Machine (x86_64)
- Power supply used | N/A
- SD card used | N/A

#### Steps to reproduce

1. Import VM into VirtualBox
2. Attempt to start

#### Expected behaviour

- It starts

#### Actual behaviour

- Fails to start due to an issue with ethernet adapter

#### Extra details

- A weird one. It will give this issue upon booting. Disabling the adapter in VirtualBox fixes the issue, however, the issue seems to be caused by the preselected graphics controller "VboxVGA" and changing to "VMSVGA" fixes the issue with the adapter and boot fine. I'm not sure if the issue is specific to me but wanted more input on this to verify.
[image]

MichaIng, 2022-02-07:
Many thanks for your report.

The issue on very first start with the network adapter on VirtualBox is known and cannot be changed. When storing the information that bridged mode should be used for network (so that the VM can be accessed from within the LAN), a network interface needs to be stored as well. But we cannot know how the network interface is called, and which host OS is used etc, so we simply use "eth0" which is at least on Linux systems a common default. When you select "Change Network Settings" and simply select "OK" directly, it should work already, since VirtualBox pre-selects the first found network interface automatically. Of course if you want to access the VM locally only or from host only, you can change settings. The bridged mode is only what seems to be most commonly used, so that the VM appears like any other machine in the local network.

We currently ship VirtualBox VMs with the classic VBoxVGA graphics controller indeed, which is not developed anymore and hence not the VBox default. But in case of a server machine it doesn't matter at all. This is only relevant when you run heavy GUI applications, like for video playback or games. A desktop has zero issues with this. Originally we switched away from the new default since VMSVGA broke boot when using custom console resolutions. This issue however does not exist anymore, so I'll switch for new images to VMSVGA now to mute the warning.

Joulinar, 2022-02-07:
@MichaIng 
While testing, on my VirtualBox, a VM got created with attached USB3 device. But this is blocking the boot and I need to deselect USB.

Joulinar, 2022-02-07:
Yes my host support USB3. But I missed to install the Oracle VM VirtualBox Extension Pack. Without, you can't use USB3. Not sure if we should use USB1.1 as default. Just in case there are more user who missed the Extension Pack.

MichaIng, 2022-02-07:
Ahh, this makes sense. Jep lets switch to USB1.1 then by default. Btw, graphics controller switched for new images: https://github.com/MichaIng/DietPi/commit/9d1823e

MichaIng, 2022-02-07:
Done: https://github.com/MichaIng/DietPi/commit/7177f71
Ignore wrong "v8.1" in commit message 😄.

MichaIng, 2022-02-08:
Okay, I'll mark this as closed. As said, there is sadly not something we can do about the network adapter error, as long as we do not want to have host-only network enabled. But since DietPi is mostly used as server system, host-only network is not suitable, IMO.

But we should add a hint about this to our docs: https://dietpi.com/docs/install/
It is really as simple as hitting "Change Network Settings" and "OK", as an actually present network adapter is selected automatically. Of course in general it makes sense to configure the VM according to needs and host system capabilities, what we ship is more a compatible/minimal default 😉.

closed 2022-02-08 by MichaIng